cetic/6lbr #144: Cannot reach NODE

hauseja3, 2016-03-25:
Dear 6lbr master, 

I decided to share my settings because I´m desperate to ping my nodes. Could you please check it and maybe even help me? 

I attached all the files from my router website running on BeagleBone with CC2538 slip radio. 
My nvm tool output looks like this:

```
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 26
PAN ID : abcd

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : aaaa::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 52
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay disabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : False

NAT 64 Enabled : True
NAT 64 DHCP Enabled : True
NAT 64 static port mapping Enabled : True
NAT 64 RFC-6052 prefix : True
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1

Local address rewrite : False
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80
```

And my ifconfig output is:

```
eth0      Link encap:Ethernet  HWaddr 54:4a:16:c1:3f:0c  
          inet addr:147.32.86.205  Bcast:147.32.86.255  Mask:255.255.255.0
          inet6 addr: 2001:718:2:1654:564a:16ff:fec1:3f0c/64 Scope:Global
          inet6 addr: fe80::564a:16ff:fec1:3f0c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:93553 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6294 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:9986008 (9.5 MiB)  TX bytes:1359067 (1.2 MiB)
          Interrupt:40 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:285 errors:0 dropped:0 overruns:0 frame:0
          TX packets:285 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:35008 (34.1 KiB)  TX bytes:35008 (34.1 KiB)

usb0      Link encap:Ethernet  HWaddr 1a:fb:e4:fc:25:1b  
          inet addr:192.168.7.2  Bcast:192.168.7.3  Mask:255.255.255.252
          inet6 addr: fe80::18fb:e4ff:fefc:251b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5983 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4337 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:447695 (437.2 KiB)  TX bytes:760683 (742.8 KiB)
```

I tried to ping my IPV4 address  147.32.86.205 and everything went gr8. When i tried to ping my Node  147.32.86.205:12345 I found out that it is not working. I hope it will be some obvious stupid mystake, but I´ve spend so much time with configuration and now i feel I´m close. 
  
[image]
[image]
[image]
[image]

laurentderu, 2016-03-25:
The static port mapping works only for CoAP and HTTP, not for ICMP (i.e. ping messages). The concept or port has no meaning for icmp.

If you have a coap server instanciated on your node, you can use the ping feature of Copper using the IPv4 address. Your configuration looks ok, so there should not be any issue.

hauseja3, 2016-03-30:
Ok, that makes sense :+1: 

But now I cant reach CoAP server. I have compiled 6lbr-demo (make TARGET=cc2538dk WITH_COAPSERVER=1) and programmed it to my cc2538. I have read #145 and allowed the route creation from the RA sent by 6lbr as described here ( https://github.com/cetic/6lbr/wiki/6LBR-Connection), I actually added connection manually. 

Now my eth0 ifconfig on my linux PC is:

````
eth0      Link encap:Ethernet  HWaddr 00:0c:29:8b:54:1a  
          inet addr:192.168.111.153  Bcast:192.168.111.255  Mask:255.255.255.0
          inet6 addr: bbbb::101/64 Scope:Global
          inet6 addr: fe80::20c:29ff:fe8b:541a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:13076 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5731 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:15750361 (15.7 MB)  TX bytes:584092 (584.0 KB)
          Interrupt:19 Base address:0x2000 
````

My 6lbr settings is as above and I still get "Server not responding".

[image]

@laurentderu do you have any hint plz? Cheers!

tiagodiogo, 2016-03-30:
in my case i got communication by setting up the aaaa::/64 route on the linux PC and access the IP of the node with the prefix aaaa and not bbbb

hauseja3, 2016-03-30:
@tiagodiogo thanks I will give it a shot. :)

closed 2016-04-18 by laurentderu