cetic/6lbr #145: no communication 6lbr <-> 6lbr-demo with default configurations

tiagodiogo, 2016-03-27:
hello all,
i'm currently having issues while trying to communicate between the border router and the network nodes. in order to facilitate the exposure of my situation, i will start by doing and overview of my setup and what i have already working, folllowing with an explanation of my problem and some attemps to understand what is going on.

the following image shows an overview of my setup.
the border router, slip-radio and coap client (6lbr-demo) have been built from the 6lbr develop branch.
the coap client is firefox browser with the copper plugin

[image]

so far i believe i have managed to setup the 6lbr.
my linux client network interfaces are properly configured:

[image]

i can access the web interface both from the border router and the linux client as seen in the following picture:

[image]

and my nvm.dat file looks like this:

```
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 26
PAN ID : abcd

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : aaaa::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 10
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay disabled : False
Noncoresec anti-replay workaround : False
[... 519 lines cut ...]
2016-03-27 21:22:22.805479: PACKET: TAP: read: 373
2016-03-27 21:22:22.820962: PACKET: ETH: read: 373
2016-03-27 21:22:22.831918: PACKET: PF: eth_input: Dropping packet type=0x0800
2016-03-27 21:22:24.873081: PACKET: PF: bridge_output: Sending packet to Broadcast
2016-03-27 21:22:24.886950: PACKET: PF: wireless_output: sending packet
2016-03-27 21:22:24.896591: PACKET: BR-RDC: write: 80 (sid: 6, cb: 1)
2016-03-27 21:22:24.911575: PACKET: SLIP: write: 111
2016-03-27 21:22:24.933687: PACKET: SLIP: read: 5
2016-03-27 21:22:24.935195: DEBUG: SCMD: Packet data report for sid:6 st:0 tx:1
2016-03-27 21:22:24.943022: PACKET: BR-RDC: sid ack: 6 (0, 1)
2016-03-27 21:22:51.763677: PACKET: PF: bridge_output: Sending packet to Broadcast
2016-03-27 21:22:51.765452: PACKET: PF: wireless_output: sending packet
2016-03-27 21:22:51.766946: PACKET: BR-RDC: write: 80 (sid: 7, cb: 1)
2016-03-27 21:22:51.768466: PACKET: SLIP: write: 111
2016-03-27 21:22:51.785656: PACKET: SLIP: read: 5
2016-03-27 21:22:51.787505: DEBUG: SCMD: Packet data report for sid:7 st:0 tx:1
2016-03-27 21:22:51.788859: PACKET: BR-RDC: sid ack: 7 (0, 1)
2016-03-27 21:22:51.927540: PACKET: TAP: read: 373
2016-03-27 21:22:51.929024: PACKET: ETH: read: 373
2016-03-27 21:22:51.930401: PACKET: PF: eth_input: Dropping packet type=0x0800
2016-03-27 21:22:55.250722: PACKET: TAP: read: 373
2016-03-27 21:22:55.252332: PACKET: ETH: read: 373
2016-03-27 21:22:55.253749: PACKET: PF: eth_input: Dropping packet type=0x0800

```

msolters, 2016-03-27:
You're not seeing any `wireless_input` lines in your `/var/log/6lbr.log`.  In my experience, that means your SLIP radio is not actually receiving any traffic from your node.

Are your nodes definitely being compiled with radio channel 26 (that's what 6LBR is expecting according to your .dat file) and PAN_ID of `abcd`?  Does that COAP example that does show up in the neighbors list have different values for those two parameters?

tiagodiogo, 2016-03-27:
they both are,
the one that does not even show up (6lbr-demo) has those settings already present under section radio in project-conf.h:

```
*---------------------------------------------------------------------------*/
/* Radio                                                                     */
/*---------------------------------------------------------------------------*/

#ifndef IEEE802154_CONF_PANID
#define IEEE802154_CONF_PANID 0xABCD
#endif

#ifndef RF_CHANNEL
#define RF_CHANNEL 26
#endif
```

the one that does show up, but doesn't really work (er-rest-example) has those settings, commented, in its project-conf.h:

```
/* Custom channel and PAN ID configuration for your project. */

   #undef RF_CHANNEL
   #define RF_CHANNEL                     26

   #undef IEEE802154_CONF_PANID
   #define IEEE802154_CONF_PANID          0xABCD

```

however i tried with and without the commented section and the result was the same

tiagodiogo, 2016-03-28:
i saw that in the Makefile of the 6lbr/examples/ipv6/slip-radio the are the options:

```
CONTIKI_WITH_RPL = 0
CONTIKI_WITH_IPV6 = 1
```

i don't know exactly what the CONTIKI_WITH_RPL does but i have already tried with 0 and 1

laurentderu, 2016-03-29:
If er-rest-example shows up in 6lbr and 6lbr-demo does not, I think @msolters is right, the channel and or the pan-id is wrong. Maybe in the platform-conf.h or contiki-conf.h of the ReMote the channel is forced the code in 6lbr-demo will not change it as it will only update it if it's not yet configured.

The second issue, not being able to access the CoAP server from your PC, looks like the aaaa::/64 route is not created in your PC. You should either create it manually or allow route creation from the RA sent by 6LBR.

tiagodiogo, 2016-03-29:
you are absolutely right, the contiki-conf.h of the ReMote had channel 25 hardcoded in there.
i wasn't expecting it to work after i changed it because one of my tests was to replace the channel and pan id definition of 6lbr-demo:

```
#ifndef IEEE802154_CONF_PANID
#define IEEE802154_CONF_PANID 0xABCD
#endif

#ifndef RF_CHANNEL
#define RF_CHANNEL 26
#endif

```

with:

```
#undef IEEE802154_CONF_PANID
#define IEEE802154_CONF_PANID 0xABCD

#undef RF_CHANNEL
#define RF_CHANNEL 26

```

thinking that any previous value would be replaced at this point.
guess this is what happens when you don't know what you are doing...

regarding the access to CoAP resources from my PC, it seems I miss understood the role of the border-router, I thought that I was supposed to have the bbbb::/64 route on my linux machine, and the border-router would translate the request into an aaaa:://64 ip
as soon as I allowed the route creation from the RA sent by 6lbr i could immediately access the server resources.

thank you @msolters  and @laurentderu for your help

closed 2016-03-30 by tiagodiogo