cetic/6lbr #146: enabling llsec stops wireless communication

tiagodiogo, 2016-03-30:
hello all,
after achieving communication using the following hardware setup:

[image]

my next step was to enable link layer security. there is not much documentation regarding this feature, but what I did was enable LLSEC in the 6lbr-demo Makefile:

`WITH_LLSEC?=1`

and make sure the definitions in the project-conf.h:

```
#if WITH_LLSEC

#undef NETSTACK_LLSEC
#define NETSTACK_LLSEC noncoresec_driver

#undef LLSEC802154_CONF_SECURITY
#define LLSEC802154_CONF_SECURITY 1
#ifndef LLSEC802154_CONF_SECURITY_LEVEL
#define LLSEC802154_CONF_SECURITY_LEVEL 7
#endif

#undef AES_128_CONF
#define AES_128_CONF aes_128_driver

#define LLSEC_ANTIREPLAY_ENABLED 1
#define LLSEC_REBOOT_WORKAROUND_ENABLED 1

#else

#undef NETSTACK_LLSEC
#define NETSTACK_LLSEC nullsec_driver

#endif
```

are compatible with the definitions in the 6lbr:

[image]

however, after this installation I stopped seeing any wireless_input in my 6lbr.log file.
i noticed that @msolters had a similar problem in issue #115 but it was fixed by adding the correct framer:

```
#undef NETSTACK_CONF_FRAMER
#define NETSTACK_CONF_FRAMER noncoresec_framer
```

but that was for building the node from the contiki tree and not the 6lbr-demo

is there something else that needs to be done in order to enable LLSec in the 6lbr-demo?

i did not define the network key because I believe one is selected if none is specified as seen in noncoresec.c:

```
#ifdef NONCORESEC_CONF_KEY
#define NONCORESEC_KEY NONCORESEC_CONF_KEY
#else /* NONCORESEC_CONF_KEY */
#define NONCORESEC_KEY { 0x00 , 0x01 , 0x02 , 0x03 , \
                         0x04 , 0x05 , 0x06 , 0x07 , \
                         0x08 , 0x09 , 0x0A , 0x0B , \
                         0x0C , 0x0D , 0x0E , 0x0F }
#endif /* NONCORESEC_CONF_KEY */
```

laurentderu, 2016-03-31:
At first sight everything looks ok. Are you using the latest snapshot or develop ? The older version of 6LBR did not remove the security attribute when sending the encrypted packet to the slip-radio and the slip-radio dropped the packet as it did not understand them.

You could also disable the anti-replay on 6LBR and see if it works better.

If it still fails, enable the traces in ip/llsec/noncoresec.c to see if the packets are dropped there  and for which reason.

tiagodiogo, 2016-04-03:
I've been using the develop branch, but to clear any doubts I did a fresh checkout, reinstalled 6LBR on the raspberry pi, and updated the slip-radio and server nodes.
Disabling the anti-replay on 6LBR had no apparent effect as I'm still not seeing any wireless_inputs on the 6LBR log file.
What do you mean by "enable the traces" ? Is there some logger method I should use?

laurentderu, 2016-04-04:
The LLSEC non regression tests are ok, so it's either a ReMote specific problem or a setup issue. 

About the traces : the debug traces in Contiki must be manually activated in the code itself, usually it's by changing the line 

    #define DEBUG DEBUG_NONE

to

    #define DEBUG 1

tiagodiogo, 2016-04-04:
I believe I found the DEBUG line you were referring to in noncoresec.c and changed it to 1:

```
#define DEBUG 1
#if DEBUG
#include <stdio.h>
#define PRINTF(...) printf(__VA_ARGS__)
#else /* DEBUG */
#define PRINTF(...)
#endif /* DEBUG */
```

but then I started having compilation issues with some of the PRINTFs in that file:

../../core/net/llsec/noncoresec/noncoresec.c:85:28: note: in definition of macro 'PRINTF'
 #define PRINTF(...) printf(__VA_ARGS__)
                            ^
../../core/net/llsec/noncoresec/noncoresec.c:173:55: warning: spurious trailing '%' in format [-Wformat=]
     PRINTF("noncoresec: received nonauthentic frame %"PRIu32"\n",

../../core/net/llsec/noncoresec/noncoresec.c:213:54: error: expected ')' before 'PRIu32'
        PRINTF("noncoresec: received replayed frame %"PRIu32"\n",

laurentderu, 2016-04-05:
I guess the right include is missing or the compiler needs a specific flag to enable the PRI macros. In the mean time you can replace %"PRIu32" with %u.

tiagodiogo, 2016-04-06:
enabling the DEBUG has brought additional information.
I can see many **"noncoresec: received nonauthentic frame"** in the border-router.
which I believe is coming from this area of the noncoresec.c

```
 received_mic = dataptr + data_len;
  if(memcmp(generated_mic, received_mic, LLSEC802154_MIC_LENGTH) != 0) {
    PRINTF("noncoresec: received nonauthentic frame\n",
        anti_replay_get_counter());
    noncoresec_nonauthentic++;
    return;
  }
```

what can be the cause for the nonauthentic frame?
I've already been searching for some lost define statements in the contiki-conf.h of the RE-Mote platform but found nothing relevant.
my configurations remain the same as posted above

laurentderu, 2016-04-07:
That error means the LLSEC layer could not decrypt the incoming frame, either because the key used to encrypt the message is different or the encryption on the ReMote introduce a platform dependent bug (which is hard to debug). It could be the aes128 driver or the ccm driver.

tiagodiogo, 2016-04-07:
what can I do now @laurentderu ?
i have very little experience with the contiki-os, how would you proceed?

closed 2017-12-14 by laurentderu