cetic/6lbr #147: make: *** [install] Error 1 during the 6lbr installation

suspirik, 2016-04-02:
Hi all,

I followed that steps below and received the error. I checked the bin dir and its empty.
steps:
sudo apt-get update 
sudo apt-get install bridge-utils libncurses5-dev
git clone --recursive https://github.com/cetic/6lbr
cd 6lbr/examples/6lbr
make all_native plugins tools
sudo make install

I did not receive any error for make all native part but make install.

install -d /etc/6lbr
cp -r package/etc/6lbr /etc
install -d /etc/init.d
install package/etc/init.d/* /etc/init.d
install -d /usr/bin
install package/usr/bin/* /usr/bin
install -d /usr/lib/6lbr
cp -r package/usr/lib/6lbr /usr/lib
install -d /usr/lib/6lbr/bin
install bin/* /usr/lib/6lbr/bin
install: cannot stat ‘bin/*’: No such file or directory
make: *** [install] Error 1

Do you have any suggestion or solution for that ?

shantanoo-desai, 2016-04-02:
yes, I suggest you go to `bash` script and find the line `install bin/*` and replace it with `/bin/` maybe that might be the problem.

suspirik, 2016-04-03:
/usr/share/menu/bash I checked this file and could not find install bin line. can you please give me the path if you mean another script? 

?package(bash):needs="text" section="Applications/Shells" title="Bash" command="/bin/bash --login"
?package(bash):needs="text" section="Applications/Shells" title="Sh" command="/bin/sh --login"

Thanks.

laurentderu, 2016-04-04:
You should read more closely the build log :)

The target 'all_native' is deprecated in the develop branch (and soon 1.4.0), instead you must use 

    make TARGET=native all

The documentation still describes the 1.3.x way of building.

suspirik, 2016-04-04:
as a friend's suggestion 'make all plugins tools'  instead of make all_native plugins tools and then the next command worked.

closed 2016-04-18 by laurentderu