cetic/6lbr #148: Error writing nvm file: Permission denied

suspirik, 2016-04-04:
Hej,
After this step below I receive the error:

laurentderu, 2016-04-04:
/etc/6lbr/nvm.dat is usually owned by `root` so a normal user can not modify it. Either run nvm_tool as root or use another file, e.g. : 

    /usr/lib/6lbr/bin/nvm_tool --update --channel 25 my_nvm.dat

suspirik, 2016-04-06:
hi again,
 I do not know some how I lost the nvm.conf file and now I can not acces the web. can you please help about the matter.

suspirik, 2016-04-06:
Today I was able to access to the web but somehow I lost access and I receive this log 

user@instant-contiki:/dev$ sudo /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3 -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip -m /usr/lib/6lbr/plugins 
Contiki-develop-20160323-13-ga47daf1 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-04-06 20:24:52.190872: NOTICE: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20160323-13-ga47daf1)
2016-04-06 20:24:52.191161: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-04-06 20:24:52.191239: INFO: NVM: NVM Magic : 2009
2016-04-06 20:24:52.193079: INFO: NVM: NVM Version : 5
2016-04-06 20:24:52.193137: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-04-06 20:24:52.193166: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-04-06 20:24:52.193198: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-04-06 20:24:52.193210: INFO: MAC: Using 'csma' mac driver
2016-04-06 20:24:52.193291: INFO: SLIP: SLIP started on /dev/ttyACM0
2016-04-06 20:24:52.205755: INFO: SCMD: Started br-cmd process
2016-04-06 20:24:52.205772: INFO: BR-RDC: Reset SLIP Radio
2016-04-06 20:24:52.205777: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:53.205200: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:54.205314: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:55.205140: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:56.205136: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:57.205229: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:58.205246: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:24:59.205102: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:00.206448: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:01.298786: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:02.298278: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:03.299563: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:04.299150: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:05.299239: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:06.299169: INFO: BR-RDC: Fetching MAC address
2016-04-06 20:25:06.302313: INFO: BR-RDC: Got MAC: 00:12:4b:00:01:dc:73:61
2016-04-06 20:25:07.299165: INFO: ETH: RAW Ethernet interface init
2016-04-06 20:25:07.505033: INFO: TAP: opened device /dev/eth0
2016-04-06 20:25:07.505168: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
br0: error fetching interface information: Device not found
eth0      Link encap:Ethernet  HWaddr 00:0c:29:83:31:76  
          inet addr:192.168.16.133  Bcast:192.168.16.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fe83:3176/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:111 errors:0 dropped:0 overruns:0 frame:0
          TX packets:90 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:31239 (31.2 KB)  TX bytes:14025 (14.0 KB)
          Interrupt:19 Base address:0x2000 

2016-04-06 20:25:07.603263: INFO: TAP: Eth MAC address : 00:0c:29:83:31:76
2016-04-06 20:25:07.603453: INFO: ETH: Eth MAC address : 00:0c:29:83:31:76
2016-04-06 20:25:07.603617: INFO: LLSEC: Using 'nullsec' llsec driver
2016-04-06 20:25:07.603761: INFO: 6LBR: Security layer initialized
2016-04-06 20:25:07.603973: INFO: SLIP: cc2531: Setting RF channel 0x19
2016-04-06 20:25:07.604118: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:1dc:7361
2016-04-06 20:25:07.604323: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:1dc:7361
2016-04-06 20:25:07.604478: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-04-06 20:25:07.604608: INFO: 6LBR: RA Daemon enabled
2016-04-06 20:25:07.604749: INFO: 6LBR: Checking addresses duplication
2016-04-06 20:25:18.608564: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:1dc:7361
2016-04-06 20:25:18.611323: INFO: 6LBR: Starting as RPL ROUTER
2016-04-06 20:25:18.613786: INFO: NODECFG: Node Config init
2016-04-06 20:25:18.615685: INFO: NODECFG: No node_config.conf file specified
2016-04-06 20:25:18.617127: INFO: HTTP: Starting webserver on port 80
2016-04-06 20:25:18.617859: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2016-04-06 20:25:18.621161: INFO: DNS: DNS proxy started
2016-04-06 20:25:18.621355: INFO: 6LBR: CETIC 6LBR Started
RD client started

It looks everything ok cam you please help me ?

my 6lbr.conf:

MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh
LOG_LEVEL=3

thanks in advance...

laurentderu, 2016-04-06:
Please provide more contextual information to your issues. I don't see anything wrong in your setup, so I guess the issue is on your PC, please check https://github.com/cetic/6lbr/wiki/6LBR-Connection to configure it properly.

suspirik, 2016-04-07:
MODE=ROUTER
RAW_ETH=0
I changed the conf file as 
BRIDGE=1
CREATE_BRIDGE=0
DEV_ETH=eth0
DEV_BRIDGE=br0
DEV_TAP=tap0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0 
BAUDRATE=115200
SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh
LOG_LEVEL=3

Now I am able to access to the web and connected to sensor but I receive errors in the log below;

user@instant-contiki:/etc/6lbr$ sudo /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3 -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip -m /usr/lib/6lbr/plugins
Contiki-develop-20160323-13-ga47daf1 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-04-06 22:22:42.519031: NOTICE: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20160323-13-ga47daf1)
2016-04-06 22:22:42.519985: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-04-06 22:22:42.520052: INFO: NVM: NVM Magic : 2009
2016-04-06 22:22:42.520069: INFO: NVM: NVM Version : 5
2016-04-06 22:22:42.520117: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-04-06 22:22:42.520144: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-04-06 22:22:42.520187: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-04-06 22:22:42.520200: INFO: MAC: Using 'csma' mac driver
2016-04-06 22:22:42.520223: INFO: SLIP: SLIP started on /dev/ttyACM0
2016-04-06 22:22:42.530641: INFO: SCMD: Started br-cmd process
2016-04-06 22:22:42.530657: INFO: BR-RDC: Reset SLIP Radio
2016-04-06 22:22:42.530662: INFO: BR-RDC: Fetching MAC address
2016-04-06 22:22:43.530140: INFO: BR-RDC: Fetching MAC address
2016-04-06 22:22:43.532150: ERROR: SLIP: Unknown command
2016-04-06 22:22:44.530147: INFO: BR-RDC: Fetching MAC address
2016-04-06 22:22:45.530158: INFO: BR-RDC: Fetching MAC address
2016-04-06 22:22:45.531802: INFO: BR-RDC: Got MAC: 00:12:4b:00:01:dc:73:61
2016-04-06 22:22:46.530130: INFO: ETH: RAW Ethernet interface init
2016-04-06 22:22:46.530272: INFO: TAP: opened device /dev/eth0
2016-04-06 22:22:46.530314: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
br0       Link encap:Ethernet  HWaddr 00:0c:29:83:31:76  
          inet6 addr: bbbb::20c:29ff:fe83:3176/64 Scope:Global
          inet6 addr: bbbb::cac:9a2:de56:ed7c/64 Scope:Global
          inet6 addr: fe80::20c:29ff:fe83:3176/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:403 errors:0 dropped:0 overruns:0 frame:0
          TX packets:50 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:47758 (47.7 KB)  TX bytes:8679 (8.6 KB)

eth0      Link encap:Ethernet  HWaddr 00:0c:29:83:31:76  
          inet addr:192.168.16.133  Bcast:192.168.16.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fe83:3176/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1004 errors:0 dropped:0 overruns:0 frame:0
          TX packets:639 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:125663 (125.6 KB)  TX bytes:143689 (143.6 KB)
          Interrupt:19 Base address:0x2000 

2016-04-06 22:22:46.576683: INFO: TAP: Eth MAC address : 00:0c:29:83:31:76
2016-04-06 22:22:46.576899: INFO: ETH: Eth MAC address : 00:0c:29:83:31:76
2016-04-06 22:22:46.577046: INFO: LLSEC: Using 'nullsec' llsec driver
2016-04-06 22:22:46.577150: INFO: 6LBR: Security layer initialized
2016-04-06 22:22:46.577488: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:1dc:7361
2016-04-06 22:22:46.577649: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:1dc:7361
2016-04-06 22:22:46.578985: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-04-06 22:22:46.579105: INFO: 6LBR: RA Daemon enabled
2016-04-06 22:22:46.579196: INFO: 6LBR: Checking addresses duplication
2016-04-06 22:22:48.978723: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
2016-04-06 22:22:51.578498: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
2016-04-06 22:22:55.147697: ERROR: BR-RDC: br-rdc: ack received for unknown packet (0)
2016-04-06 22:22:57.585115: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:1dc:7361
2016-04-06 22:22:57.589360: INFO: 6LBR: Starting as RPL ROUTER
2016-04-06 22:22:57.592127: INFO: NODECFG: Node Config init
2016-04-06 22:22:57.593383: INFO: NODECFG: No node_config.conf file specified
2016-04-06 22:22:57.593705: INFO: HTTP: Starting webserver on port 80
2016-04-06 22:22:57.594578: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2016-04-06 22:22:57.596703: INFO: DNS: DNS proxy started
2016-04-06 22:22:57.596876: INFO: 6LBR: CETIC 6LBR Started
RD client started

laurentderu, 2016-04-07:
A few ack-timeout or unknown command errors can happen at 6LBR start up as the slip-radio is maybe not properly synced. It happens more often with CC2530. If it happens again, it may be due to some other process trying to read from the slip-radio device.

closed 2016-04-18 by laurentderu