cetic/6lbr #149: No address assigned on Beaglebone Black

joeythesaint, 2016-04-05:
I apologize if this is a FAQ, I'm searching through the mail archives now, too, but so far I haven't found a solution to my problem.

I'm trying to bring up 6lbr on my BeagleBone Black with a TI CC2531EMK.  I've flashed the slip radio firmware on it as described [here](http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples#2.4GHz) then started working through the howto described [here](http://processors.wiki.ti.com/index.php/Contiki-6LOWPAN-BBB) with no success.

So now I've taken a step back and all I want to see first is the interface come up in some expected configuration such that I might be able to ping a remote 6lowPAN device.

My BBB is currently running the 2016-01-24 image of Debian Jessie from [BeagleBoard.org](http://beagleboard.org/latest-images) (I changed the default /bin/sh->bash from dash because dash would throw errors on `<6lbr-functions>` when it was sourced) and I've cloned 6lbr and built it directly on the board with the following steps:

```
cd 6lbr/examples/6lbr
make TARGET=native all plugins tools
suod make install
```

Then I copied the `<6lbr.conf.beaglebone>` to `</etc/6lbr/6lbr>` and changed `<DEV_RADIO=/dev/ttyUSB0>` to `<DEV_RADIO=/dev/ttyACM0>` since that's how my 2531 appears on the system.  And I bumped up the verbosity of the debug messages hoping it'd help me figure out what was going wrong.  So my config file looks like this:

```
root@beaglebone:~# cat /etc/6lbr/6lbr.conf
#This file contains a default configuration for BeagleBone platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=1

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=4 #INFO and above only
```

Then I ensured I had my system configured to accept RA messages as described [here](https://github.com/cetic/6lbr/wiki/6LBR-Connection) and enabled forwarding for both IPv4 and IPv6:

```
root@beaglebone:~# sysctl net.ipv6.conf.eth0 | grep accept_ra
net.ipv6.conf.eth0.accept_ra = 1
net.ipv6.conf.eth0.accept_ra_defrtr = 1
net.ipv6.conf.eth0.accept_ra_from_local = 0
net.ipv6.conf.eth0.accept_ra_mtu = 1
net.ipv6.conf.eth0.accept_ra_pinfo = 1
net.ipv6.conf.eth0.accept_ra_rt_info_max_plen = 64
net.ipv6.conf.eth0.accept_ra_rtr_pref = 1
sysctl: reading key "net.ipv6.conf.eth0.stable_secret"
root@beaglebone:~# sysctl net.ipv4.ip_forward
net.ipv4.ip_forward = 1
root@beaglebone:~# sysctl net.ipv6.conf.all.forwarding
net.ipv6.conf.all.forwarding = 1
root@beaglebone:~#
```

Then I started 6lbr:

```
root@beaglebone:~# service 6lbr start
root@beaglebone:~# service 6lbr status
[... 174 lines cut ...]
          RX packets:3796 errors:0 dropped:3 overruns:0 frame:0
          TX packets:412 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:496135 (484.5 KiB)  TX bytes:82504 (80.5 KiB)
          Interrupt:177

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2664 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2664 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:208351 (203.4 KiB)  TX bytes:208351 (203.4 KiB)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5 errors:0 dropped:0 overruns:0 frame:0
          TX packets:531 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:614 (614.0 B)  TX bytes:73050 (71.3 KiB)

```

Any idea what I'm missing or doing wrong here?

laurentderu, 2016-04-05:
At first sight everything looks good, except that you have enabled accept_ra and accept_ra_rt_info_max_plen on **eth0** and not **br0** As eth0 is part of a bridge, it's IP stack is disabled. You have to use br0 instead.

    sysctl -w net.ipv6.conf.br0.accept_ra=1
    sysctl -w net.ipv6.conf.br0.accept_ra_rt_info_max_plen=64

If that does not solve your problem, maybe you have a network manager, like wicd or NetworkManager that override your configuration.

joeythesaint, 2016-04-06:
> If that does not solve your problem, maybe you have a network manager, like wicd or NetworkManager that override your configuration.

That's a very good point, I realized after I read your comment that the Debian image runs `connman` by default and thought it might be getting in the way.  First I went back and disabled it with `service connman disable` and that didn't seem to help, so I swapped out my whole image in favour of the Debian console image from [http://elinux.org/Beagleboard:BeagleBoneBlack_Debian](http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#microSD.2FStandalone:_.28console.29_.28BeagleBone.2FBeagleBone_Black.2FBeagleBone_Green.29).  I corrected the `/bin/sh` link again then grabbed the 1.3.3 6lbr release from https://github.com/cetic/6lbr/wiki/Releases and installed the .deb rather than going through the build process again.  The console image is *very* limited.  I had to install `bridge-utils` but after that 6lbr seemed happy enough to start.

I'm still not seeing any routes set up and the configuration applied.

Just to re-cap what I've done since re-imaging my SD card:

```
root@beaglebone:~# cat /etc/6lbr/6lbr.conf
#This file contains a default configuration for BeagleBone platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=1

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=4 #INFO and above only
```

Which is the `6lbr.conf.beaglebone` file tweaked with a higher log-level and `DEV_RADIO` pointing at `ttyACM0` rather than `ttyUSB0` since that's where my device appears.

```
root@beaglebone:~# cat /etc/network/interfaces
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
#auto eth0
#iface eth0 inet dhcp
# Example to keep MAC address between reboots
#hwaddress ether DE:AD:BE:EF:CA:FE

# The secondary network interface
#auto eth1
#iface eth1 inet dhcp

iface eth0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
    bridge_ports eth0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show eth0 | grep ether | awk '{print $2}'`

# WiFi Example
[... 215 lines cut ...]
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL Preference : 0
RPL version ID :  : 246
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL minimum rank increment : 256
RPL lifetime unit : 256

Webserver configuration page disabled : False
```

Obviously I've cycled my logs on my system while debugging so we're only looking at the latest run of logs above.  Is there something else I could provide or look at that might help me understand what's going on here?

Right now I'm having my BBB build `master` from git to see if that helps, but that's kind of a lengthy process, as you probably know.  :-)

Somewhat related, in the future if I'm providing longer blocks of information like this, would it be better for me to put it all in a gist and link to it from the comments or dump the info right in the comments?  Different projects / maintainers / devs have different preferences and I want to stick to whatever is easiest.

laurentderu, 2016-04-06:
Thanks to you extensive blocks of information ;) I think I know the root cause of your problem :

    net.ipv6.conf.all.forwarding = 1

This setup configure your system as an IP Router, and, by definition a router never listen to RA messages. So, the RA sent by 6LBR is ignored by the beaglebone. You can either reverse it to 0 or use a magic mode provided by the linux kernel :

    net.ipv6.conf.br0.accept_ra=2

With this value, the system will behave as a router, but your interface will still listen to RA and use them for the address autoconfiguration

joeythesaint, 2016-04-06:
Awesome, that was it.  I set `net.ipv6.conf.br0.accept_ra=2` and now:

```
root@beaglebone:~# ifconfig
br0       Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          inet6 addr: bbbb::a:bff:fe0c:d0e/64 Scope:Global
          inet6 addr: fe80::a:bff:fe0c:d0e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:616 (616.0 B)  TX bytes:1026 (1.0 KiB)

eth0      Link encap:Ethernet  HWaddr 78:a5:04:bc:d0:04
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:175

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:714 (714.0 B)  TX bytes:1336 (1.3 KiB)

usb0      Link encap:Ethernet  HWaddr 78:a5:04:bc:d0:00
          inet addr:192.168.7.2  Bcast:192.168.7.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@beaglebone:~# route -6 -n
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
aaaa::/64                      fe80::7ff:ff68:d0f5        UGAe 1024 0     0 br0
bbbb::/64                      ::                         UAe  256 0     0 br0
fe80::/64                      ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1     1 lo
::1/128                        ::                         Un   0   1     1 lo
bbbb::/128                     ::                         Un   0   1     0 lo
bbbb::a:bff:fe0c:d0e/128       ::                         Un   0   1     0 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::a:bff:fe0c:d0e/128       ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 tap0
ff00::/8                       ::                         U    256 3     0 br0
::/0                           ::                         !n   -1  1     1 lo
```

Thanks so much!

closed 2016-04-06 by joeythesaint

RaghavendraSurpur, 2017-11-23:
Hi All,

We are working on BBB using 04.00.00.04  TI SDK (arago on linux) , After using this command "make TARGET=native all plugins tools" we are getting these errors :+1: 
make TARGET=native all plugins tools    
rm -rf bin
mkdir -p bin
make TARGET=native clean
make[1]: Entering directory '/etc/6lbr/examples/6lbr'
rm -f *~ *core core *.srec \
*.lst *.map \
*.cprg *.bin *.data contiki*.a *.firmware core-labels.S *.ihex *.ini \
*.ce *.co
rm -rf *.so obj_native/*.so
rm -rf obj_native
make[1]: Leaving directory '/etc/6lbr/examples/6lbr'
make TARGET=native bin/cetic_6lbr_smart_bridge
make[1]: Entering directory '/etc/6lbr/examples/6lbr'
make TARGET=native CETIC_6LBR_SMARTBRIDGE=1 WITH_UDPCLIENT=0
make[2]: Entering directory '/etc/6lbr/examples/6lbr'
make[2]: Nothing to be done for 'cetic-6lbr'.
make[2]: Leaving directory '/etc/6lbr/examples/6lbr'
mkdir -p bin
mv cetic-6lbr.native bin/cetic_6lbr_smart_bridge
mv: can't rename 'cetic-6lbr.native': No such file or directory
make[1]: *** [platform/native/Makefile.native:40: bin/cetic_6lbr_smart_bridge] E1
make[1]: Leaving directory '/etc/6lbr/examples/6lbr'
make: *** [platform/native/Makefile.native:76: all] Error 2

So please give any solutions for this,

Thanks,
Raghavendra