cetic/6lbr #150: Could not ping6 to nodes and Coap server not responding error

nd15, 2016-04-06:
Dear all,

I setup 6lbr for RPi 2 and slip radio and nodes follow the TI example. The sensor web server in below picture:
[image]

The issuse is that I can ping from PC to bbbb::100 and slip-radio but cannot ping from PC to nodes and so I meet the error Coap server not responding:
`user@ubuntu:~$ ping6 bbbb::100
PING bbbb::100(bbbb::100) 56 data bytes
64 bytes from bbbb::100: icmp_seq=1 ttl=64 time=0.654 ms
64 bytes from bbbb::100: icmp_seq=2 ttl=64 time=0.656 ms
^C
--- bbbb::100 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 999ms
rtt min/avg/max/mdev = 0.654/0.655/0.656/0.001 ms
user@ubuntu:~$ ping6 aaaa::212:4b00:7ba:c507
PING aaaa::212:4b00:7ba:c507(aaaa::212:4b00:7ba:c507) 56 data bytes
64 bytes from aaaa::212:4b00:7ba:c507: icmp_seq=1 ttl=64 time=0.651 ms
64 bytes from aaaa::212:4b00:7ba:c507: icmp_seq=2 ttl=64 time=0.586 ms
64 bytes from aaaa::212:4b00:7ba:c507: icmp_seq=3 ttl=64 time=0.606 ms
64 bytes from aaaa::212:4b00:7ba:c507: icmp_seq=4 ttl=64 time=0.605 ms
^C
--- aaaa::212:4b00:7ba:c507 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 2998ms
rtt min/avg/max/mdev = 0.586/0.612/0.651/0.023 ms`

[image]

I monitor the network through Wireshark when ping from linux PC to node but dont see any reply from node. 
[image]
 
In line 11693 and 11694, they can ping with local address between node (aaaa::212:4b00:7ba:d001) and DODAG root (slip-radio: aaaa::212:4b00:7ba:c507). But when ping from PC (bbbb::a1d1:fd7a:de63:6435) to node (aaaa::212:4b00:7ba:d001), the node addr become to ::212:4b00:7ba:d001. 

Do we have any solution for this?
Thanks for your time!

Here are my configurations and debug outputs:

6lbr.conf
`MODE=ROUTER
#MODE=SMART-BRIDGE

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyUSB1 
BAUDRATE=115200
LOG_LEVEL=3`

6lbr.log
`********************************************************************************
Wed 6 Apr 03:20:02 UTC 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyUSB1 -t eth0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
Contiki-6lbr-1.3.3 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2016-03-06 3:20:02.258757: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-03-06 3:20:02.259534: INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.3)
2016-03-06 3:20:02.259564: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2016-03-06 3:20:02.259635: INFO: NVM: NVM Magic : 2009
2016-03-06 3:20:02.259648: INFO: NVM: NVM Version : 1
2016-03-06 3:20:02.259663: INFO: SCMD: Started br-cmd process
2016-03-06 3:20:02.259676: INFO: NODECFG: No node_config.conf file specified
2016-03-06 3:20:02.259689: INFO: ETH: RAW/TAP init
2016-03-06 3:20:02.259922: INFO: SLIP: SLIP started on /dev/ttyUSB1
2016-03-06 3:20:02.304663: INFO: TAP: opened device /dev/eth0
2016-03-06 3:20:02.304985: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b8:27:eb:31:97:d0
          inet addr:192.168.1.101  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::cbc7:30be:75ce:d2dd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6937 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4656 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:937483 (915.5 KiB)  TX bytes:905374 (884.1 KiB)

2016-03-06 3:20:02.455255: INFO: TAP: Eth MAC address : b8:27:eb:31:97:d0
2016-03-06 3:20:02.455386: INFO: BR-RDC: Reset SLIP Radio
2016-03-06 3:20:02.455442: INFO: BR-RDC: Fetching MAC address
2016-03-06 3:20:02.455611: ERROR: SLIP: Unknown command
2016-03-06 3:20:02.489176: ERROR: SLIP: Unknown command
2016-03-06 3:20:02.519192: INFO: BR-RDC: Got MAC: 00:12:4b:00:07:ba:c5:07
2016-03-06 3:20:03.455415: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:7ba:c507
2016-03-06 3:20:03.455700: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:7ba:c507
2016-03-06 3:20:03.455918: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-03-06 3:20:03.456048: INFO: 6LBR: RA Daemon enabled
2016-03-06 3:20:03.456121: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2016-03-06 3:20:03.456652: INFO: 6LBR: Configured as DODAG Root
2016-03-06 3:20:03.456732: INFO: 6LBR: Starting as RPL ROUTER
2016-03-06 3:20:03.457471: INFO: UDPS: UDP server started
2016-03-06 3:20:03.457714: INFO: 6LBR: CETIC 6LBR Started
2016-03-06 3:20:03.458160: ERROR: SLIP: Unknown command`

6lbr.err
`********************************************************************************
Wed 6 Apr 03:20:02 UTC 2016 : Starting 6LBR`

laurentderu, 2016-04-06:
The node addr ::212:4b00:7ba:d001 is an artefact of Wireshark, as 6LoWPAN uses prefix compression, Wireshark must be aware of the compressed prefixes. In Preferences -> 6LoWPAN you must set aaaa:: as context 0.

Now if you can ping the node you should be able to access the CoAP server on the node. Which firmware are you using on the node ?

nd15, 2016-04-06:
Hi @laurentderu ,
I cannot ping to node from PC, I just ping to slip-radio add (aaaa::212:4b00:7ba:c507). I use 6lbr v1.3.3  and develop but it has the same problem. The node and slip-radio firmwares I built from contiki master.
I Do I miss configure somthing in 6lbr?

laurentderu, 2016-04-06:
Is your contiki master from before or after mid-march ? If after, that would explain the problem. Contiki has changed the value for the prefix compression.
To solve the issue you need to use the latest snapshot of 6LBR which allow to modify the 6lopwan context. You need to set 00fd:: instead of aaaa:: in context 0. Another possibility is to change in your firmware the context used back to aaaa::

nd15, 2016-04-06:
I already tested in some cased but it still had same problem:
- change default context from 0xfd00 to 0xaaaa on contiki master
- checkout contiki to Jan 2016
- change 6lbr between version 1.3.3 & 1.4.x

Do you have others solution for this?

laurentderu, 2016-04-06:
Which firmware are you running on the (I believe) Sensortag, 6lbr-demo, cc26xx-demo, cc26xx-web-demo ?

Could you send me or attach a sensniff pcap which show the failing ping or coap request but also the udp messages between the Sensortag and 6LBR (it's the message used to update the parent info in the sensors page)

nd15, 2016-04-07:
Now I am running cc26xx-web-demo app and 6lbr v1.3.3 (for stability). I use 2 CC2650EM(on SRF06EB) for node and slip-radio. Plz find the pcap file and all 6lbr configuration in attached file.
[6lbr.zip](https://github.com/cetic/6lbr/files/207715/6lbr.zip). 

Thanks.

pepeee, 2016-05-02:
Hi all,
I have the same problem of @nd15 !
Setup a 6lbr router with develop version on a cc2538dk+smartRF06 and a sensortag as a node.
After having problem to allow sensortag to join the router network (so I have change PANID, channel, RDC and MAC protocol to match on the following value: 0xabcd, 25, nullrdc, csma), I succeed on the intent.
Now I see the sensortag on the node panel of the 6lbr interface, but also if I have route to the sensor:
`aaaa::/64 via fe80::212:4b00:3a6:515b dev tap0  proto ra  metric 1024  expires 1765sec
bbbb::/64 dev tap0  proto kernel  metric 256  expires 86365sec
fe80::/64 dev eth0  proto kernel  metric 256 
fe80::/64 dev tap0  proto kernel  metric 256 
fe80::/64 dev wlan0  proto kernel  metric 256
`
I cannot ping the node. (515b is the cc2538dk module).
Do you have fix this problem?
@laurentderu : what do you mean with "set 00fd:: instead of aaaa:: in context 0" ?

Thanks in advice for help!

My 6lbr.conf file:

`ODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

#DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200

LOG_LEVEL=3
`

pepeee, 2016-05-02:
Just to add a more clear vision of the problem, it seems that the up PRR remain fixed on 100%, while the down PRR decrease with time, starting from 100%.
[image]

laurentderu, 2016-05-03:
@pepeee I believe yo are using a Sensortag build with contiki after mid march, so the default 6lowpan context is not the same as 6lbr.

You need to modify it in the configuration page of 6lbr or rebuild your sensortag with aaa:: default prefix.

laurentderu, 2016-05-03:
@nd15 Sorry, I overlooked your last message, do you still have the issue ? I had a quick look at the pcap and the content is a bit strange, the communication between the 6LBR and the node seems ok, one can see the RPL, UDP and link local pings communications, but when the global address of the node is used, it does not reply. Did you change something in the code of the node ?

pepeee, 2016-05-03:
Thanks for reply @laurentderu !
I try to do what you suggest but It still doesn't work (I can't ping node!). 
What I do:
slip-radio from latest Contiki
cc26xx-web-demo from latest Contiki
6lbr service from latest 6lbr repo (develop)
same PANID, channel, RDC(nullrdc), MAC(csma).
Change context on 6lbr configurator to 0x00fd as follow:
[image]
Maybe I miss something? 
Actually I succeed only follow the suggestion of this topic [https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/p/495080/1790746#pi239031350=7](url) and using a Contiki old version (7dc8ace) for cc26-web-demo.

suspirik, 2016-05-03:
Hi, I have similar problem, On 6lbr page coap button is not working as web button does. So I got the coap  adress with right click and open new page.Copper looks connected but I am receiving error for post method as below:
[image]

nd15, 2016-05-04:
Hi @laurentderu ,
I can run only for nullRDC, not for contikimac RDC.

Hi @pepeee,
You must change all: Prefix, 6LoWPAN contex 0, Manual address from aaaa:: to fd00::
I had succeed with this.

closed 2016-05-04 by nd15

pepeee, 2016-05-04:
@nd15 It works!thanks!
@suspirik Follow my configuration everything works (I also have cleared this define in cc26xx-web-demo: #define BLE_CONF_ENABLED 0)!..It seems you are doing a method (POST) not allowed on the resource selected (.well-know/core); in this resource you can only use a GET. POST method is allowed on, for example, lt/r, switching the state of red led. Have you try to do this?

suspirik, 2016-05-04:
@pepeee Thanks! I tried, the led example and its working. Do you know how to go on with er-rest example on copper?

nd15, 2016-05-05:
@pepeee 
Do you run on contikimacrdc or not (with comment out #define BLE_CONF_ENABLED 0)?

pepeee, 2016-05-05:
@suspirik  I never use the er-rest example, sorry :)
@nd15 With the test I did, the system works only  with nullrdc on both side...It is a very sick story since we cannot use the sensortag as battery powered for a long time..
For all: I still have some problem in using mqtt..Anyone succeed with mqtt?The only thing I be able to see on serial monitor is "CC26XX MQTT Client Process", but no other..it seems it exit after the first init_config() in code..I wll go more in detail later..

alterschwede, 2016-08-25:
Hey @pepeee !
You made my day!
Thank you very very much!
I have been struggling with this for some days now.
A hint to all the others, still struggling:
You have to add the appropriate route on the computer from where you interact with the node too!
On a Mac: sudo route add -inet6 -prefixlen 64 fd00:: bbbb::100

...as the Sensortag shows up with the address starting with fd00:: 

[image]

In the FireFox CoAp Copper-Plugin wou will see a full range CoAp API for all valaues:

[image]