cetic/6lbr #151: Unable to access the web server thus data on the sensortag mote

drako65, 2016-04-07:
Hi Guys,
I am setting up a 6lowpan WSN, with three sensor tags.
Two have cc26xx-web-demo firmware from contiki source.
The other sensortag acts as a slip radio on raspberry pi.
I followed the guide here: http://blog.resens.net//post/Contiki-6LowPAN-Wireless-Sensor-Network-using-SensorTag
but for my case I was unable to get data from sensortags through web.
How to I setup routing in 6lbr to I can ping or http to the nodes?
The guide https://github.com/cetic/6lbr/wiki/6LBR-Connection
doesn't seem to work.
Thanks for your help.

laurentderu, 2016-04-07:
Did your contiki source is from before of after mid-march ? If after, you need to change the default 6LoWPAN context from aaaa:: to FD00:: in 6LBR (you need the latest development snapshot to do that)

drako65, 2016-04-07:
after mid march, YES IT WORKS! 
Thank you very much.! I've been working for this for two weeks.

closed 2016-04-07 by drako65