cetic/6lbr #153: Connecting two devices with 6lbr

joeythesaint, 2016-04-07:
I'm still using the configuration I detailed in #149 and I'm having issues with my CC2650STK SensorTag, so I thought I would try another approach and apply the successful configuration from #149 to both my BeagleBone Black and an i.mx6 board I had sitting around.  So the current configuration is my BBB has a CC2531EMK on the USB port and wired ethernet connected on the other side just as described in #149.  My i.mx6 has a CC2531EMK connected to one of the USB ports and I'm able to bring up 6lbr on it in the same way with wired ethernet connected.

I can ping6 between the two machines using their respective `bbbb::` addresses, but if I disconnect the ethernet from one and assign `br0` an `aaaa::` address, I cannot ping6 that address from the other machine.  What I'm wondering, though, is if this even makes sense to try, I was approaching this thinking it would be very similar to the procedure described [here](https://bergzand.net/6lowpan-on-linux-with-raspberry-pis.html) but I was hoping to not have to bring in `wpan-tools` since it seems to rely on a newer kernel than what I currently have for my setup (4.1 on the BBB and 3.14 on the i.mx6) and 6lbr seems to be working quite well so far.  Obviously I'm rather new to a lot of this, though, so it may be I'm just asking a nonsense question, if so I apologize.

laurentderu, 2016-04-18:
Sorry, I'm a bit late to respond here...

It's possible to use two BR as a 6LoWPAN link, however it requires some configuration to do so as by default, 6LBR is meant to manage sensors and not networks.
To do so, you have to use the NDP Router mode (It's a pure 6LoPWAN Router, without RPL support) and have 3 different subnets :

      LAN-1 aaaa::/64 <-> BR-1 <------ WSN : bbbb::/64 -----> BR-2 <-> cccc::/64LAN-2

Also, in each BR you need to have a static route for the opposite subnet, i.e, on BR 1 you need a route towards cccc::/64 via BR-2 and on BR-2 you need a route towards aaaa::/64 via BR-1

If that's more or less what you're looking for, I will document better how to make the configuration.

limotkl, 2016-05-09:
Hello, I want to try 6lbr with two BRs, and I'm looking forward to your configuration document. Thank you very much

laurentderu, 2016-05-10:
@limotkl Theoretically 6LR mode should be able to do that, however the node should send DAO message on behalf of all the hosts on the subnet routed by the 6LR or having DAO messages for whole subnet, which would need modification in the code and testing.
On the other hand, if you only need to have access to services running on the 6LR then it should be enough.

joeythesaint, 2016-05-11:
@laurentderu That's exactly what I was trying to set up, yes, so definitely the documentation updates would be hugely helpful.  Based on what you provided, though, I was able to get my configuration working as I needed, so thanks for that too.

phsontung, 2016-10-02:
@joeythesaint 

Did you successfully with above configuration for connect between two BRs? Can you share me the sample configuration?

Thanks for that.

michelesponchiado, 2017-08-09:
Hi,
I am currently stuck with a similar problem and a little help would be a real godsend!
I am using a couple of SmartRF modules plugged via USB on two different Linux Ubuntu PCs, and I wanted to use the radio modules to send radio messages from a PC to another using 6lowPAN.
The final goal would be to use 6lowPAN to give to an Android hand held unit (HHU) -equipped with a CC2650- _seamless_ network connectivity through an OpenWRT board -also equipped with a CC2650-working as the _real_ border router.

I have configured both the 6lbr daemons as a NDP-ROUTER, here the 6lbr.conf file:
on both PCs
```
CREATE_BRIDGE=6LBR
MODE=NDP-ROUTER
DEV_TAP=tap0
DEV_ETH=wlan0
DEV_BRIDGE=my6b
RAW_ETH=0
BRIDGE=1
ETH_JOIN_BRIDGE=0
DEV_RADIO=/dev/ttyUSB1
BAUDRATE=115200
LOG_LEVEL=3
LOG_SERVICES=-1
```
I set the WSN/Ethernet addresses to
- aaaa::1/bbbb::100 on PC1
- aaaa::2/cccc::100 on PC2

I added the routing on nvm.conf files:
on PC1
_[network.route]
dest=cccc::/64
via=aaaa::2_

on PC2
_[network.route]
dest=bbbb::/64
via=aaaa::1_

the nvm file is:
```
/usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::2
WSN accept RA : True
WSN IP address autoconf : False
WSN 6LoWPAN context 0 : aaaa::
DNS server : ::

Eth network prefix : cccc::
Eth network prefix length : 64
Eth IP address : cccc::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 1000
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : aaaa::2
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 45
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

RPL Non Storing : False
RPL DAO ACK : True
RPL DAO ACK local repair : False
DIO triggers route refresh : False
MAC layer : 1
Security layer : 0
Security level : 6
Network key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay enabled (Legacy) : False
Noncoresec anti-replay workaround (Legacy) : False
Filter unknown nodes : False
Disable NUD : True

NAT 64 Enabled : False
NAT 64 DHCP Enabled : False
NAT 64 static port mapping Enabled : False
NAT 64 RFC-6052 prefix : False
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80
```
and this is the route setting on PC2 (PC1 has mirrored settings):
```
	sudo route -A inet6 add aaaa::/64 gw cccc::100
	sudo ip -6 addr add cccc::1/64 dev my6b 
	sudo route -A inet6 add bbbb::/64 gw cccc::100 
```
from the 6lbr.log I see the boot works OK and I can browse the webserver on both PCs

But still, as you can imagine, I am not able to ping6 cccc::100 from PC1/bbbb::100 from PC2!
I tried using 6LR mode too with no luck.

Have you any suggestions?
Many thanks for your help!