cetic/6lbr #154: running 6lbr and webserver on the same RPi

janeksz, 2016-04-08:
Hi,
Q1. Is it possible to run 6lbr on Raspberry Pi (in router mode) in GUI mode?
I connected a screen to RPi and was trying to run GUI on the same RPi, (that has 6lbr router running correctly in headless mode), but when I started "service 6lbr start" and checked top, I can not see 6lbr in there. What is the trick?
Q2. My aim is to run 6lbr and nodejs on the same RPi, is it possible and if yes how? 
I started nodejs and it is running together with 6lbr, but I am unable to read the 6lbr webpage from webserver.

msolters, 2016-04-08:
Q1.  I'm afraid I know nothing about the GUI mode.  Regarding the 6LBR service and it's health though:

1.  Try running it as sudo?   e.g. `sudo service 6lbr start`
2.  It may be running but then (very swiftly) crashing.  Check the contents of `/var/log/6lbr.log`.  Does such a file exist?  If yes, 6lbr is starting, and the contents will tell you where you're running into problems.  If no, then it's probably not starting at all.
3.  Also, try `sudo service 6lbr status` to see what the status of the module is and if it's successfully registered in your system.

Q2.  Yes, to run 6LBR and node.js processes on one raspberry pi is completely possible.  Just install each, they work out of the box.  Regarding the web page stuff, it sounds like basically you just don't have the appropriate routes established for the server to be reached (perhaps the problem is only localhost -> localhost).  What's your `ifconfig` look like once 6lbr is up?  That will tell you a lot.  If you want to handle network traffic from your mesh to a node.js server, or use your nodes as servers, that's all entirely possible as well.  You just have to make sure you establish the right routes!

For example, let's say in 6lbr config page: your gateway is like `bbbb::100`, and we're using a bridge `br0`.  Let's say the nodes have a default subnet like `aaaa::/64`

If you want your UDP client in Contiki to reach a node.js server on the RPi, you need to give your `br0` interface an IP in that subnet too, like `bbbb::1`.  On the pi's terminal:

```bash
sudo ip -6 addr add bbbb::1/64 dev br0
```

Then, in Contiki, UDP connections to `bbbb::1` will be resolvable through the gateway `bbbb::100`, so now motes can reach the node.js server on the RPi, if that server were listening on `bbbb::1`!

Example C code:
```c
// Standard Contiki Functionality
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <string.h>
#include "contiki.h"
#include "contiki-lib.h"

// Networking
#include "contiki-net.h"
#include "ip64-addr.h"

uip_ipaddr_t server_ipaddr;
struct uip_udp_conn *udp_client_conn;
char udp_buf[256]; // 256 will be arbitrary maximum UDP msg size

// Let's setup a UDP connection with a (implied nodejs) server on the RPi!
void
connect_to_server()
{
  // (1) Define RPi IP as destination: bbbb::1/64
  uip_ip6addr(&server_ipaddr, 0xbbbb, 0, 0, 0, 0, 0, 0, 0x1);

  // (2) Establish UDP connection with server (implied listening on port 5000)
  udp_client_conn = udp_new(&server_ipaddr, UIP_HTONS(5000), NULL);
  udp_bind(udp_client_conn, UIP_HTONS(5000));
}

void
send_udp_to_hub(void *msg)
{
  memcpy(udp_buf, msg, strlen(msg));
  sprintf(udp_buf, "%.*s", strlen(msg), udp_buf);
  uip_udp_packet_send(udp_client_conn, udp_buf, strlen(udp_buf));
}
```

What about the other way around?  How can a node.js app on the RPi reach a node, if I know it's IP address?  For node.js server to reach the nodes, you can add a route on the pi: 

```bash
sudo route -A inet6 add aaaa::/64 gw bbbb::100
```

Now, the node.js server can successfully transmit messages to any node with an IP in `aaaa::/64`

These steps should be taken after 6lbr is booted up, but before you initialize your node.js processes.  6lbr can overwrite or destroy these configurations while booting up, but node.js servers require the network to be properly configured to listen on the right interface, so the order matters.

msolters, 2016-04-08:
I should emphasize that the big picture here is sort of like the RPi has 2 IP addresses.  One is the gateway, `bbbb::100` which belongs to the BR.  The other is the IP Node.JS process will use, `bbbb::1`.

janeksz, 2016-04-09:
Thanks for your comments.
I am using router mode on wlan0 for 6lbr setup. It works, I can access router's website and motes 
from Win7 Firefox/Copper and can access router's website on Mac, but not able to access motes on Mac - but it is a different subject.
For the web server project I started with 2 RPis, one working as 6lbr router and another one as webserver and would like to move later to one RPi combining both functions.
My webserver is running ok on Windows and iOS and talking OK to 6lbr router, but on RPi I have issues
First issue I have at the moment is IPv6 setup on RPi, when I do:

root@raspberrypi:/www/WSNwebserverA2016# ping6 bbbb::100
PING bbbb::100(bbbb::100) 56 data bytes
From bbbb::1 icmp_seq=1 Destination unreachable: Address unreachable
From bbbb::1 icmp_seq=2 Destination unreachable: Address unreachable
...
and my ifconfig is:
...
wlan0     Link encap:Ethernet  HWaddr 00:c1:41:34:0f:5f
          inet addr:10.0.0.5  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::2d5f:293b:320b:f8a9/64 Scope:Link
          inet6 addr: bbbb::ae94:ace3:f999:6b42/64 Scope:Global
          inet6 addr: bbbb::1/64 Scope:Global
          inet6 addr: bbbb::101/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1354 errors:0 dropped:0 overruns:0 frame:0
          TX packets:597 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:375257 (366.4 KiB)  TX bytes:154279 (150.6 KiB)

Much appreciate your help.
Jan

janeksz, 2016-04-10:
Hello again,
more generic question is:
How to run 6lbr and access it on the same Linux machine?
Let's say that 6lbr is running on Linux PC and the user can open Firefox browser on the same machine and interact Not able to do it now.

root@raspberrypi:~# service 6lbr start
root@raspberrypi:~# service 6lbr status
â 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr)
   Active: active (running) since Sun 2016-04-10 17:09:13 AEST; 10s ago
  Process: 759 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/6lbr.service
           ââ764 /bin/sh /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
           ââ768 /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o ...
           ââ769 /bin/sh /usr/lib/6lbr/6lbr-watchdog 768 60 3 /var/log/6lbr.t...
           ââ836 sleep 1

Apr 10 17:09:13 raspberrypi 6lbr[759]: Starting 6LoWPAN Border Router:.
Apr 10 17:09:13 raspberrypi systemd[1]: Started LSB: 6LoWPAN Border Router.
root@raspberrypi:~# ping6 bbbb::100
connect: Network is unreachable
root@raspberrypi:~#

laurentderu, 2016-04-13:
According to your configuration you have the bbbb::/64 network set up on your wlan0 interface. With that configuration you will not be able to access 6LBR.
You need to create a bridge and assign to it both wlan0 and tap0 and set up the right bbbb::/64 address on it
If you don't need to access 6LBR or the WSN from your lan, you could otherwise setup the bbbb::/64 address only on the tap interface.
see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

janeksz, 2016-05-03:
After trying all options with no luck I need to ask you for help step by step.
My interface is wlan0
Firstly:
my /etc/network/interfaces is - Is it OK or not?
root@raspberrypi:~# cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0

iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

#allow-hotplug wlan1
#iface wlan1 inet manual
#    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

iface wlan0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
    bridge_ports wlan0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show wlan0 | grep ether | awk '{print $2}'`

root@raspberrypi:~#

janeksz, 2016-05-03:
my apologies for strange formatting. Again my /etc/network/interfaces is:

root@raspberrypi:~# cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0

iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

#allow-hotplug wlan1
#iface wlan1 inet manual
#    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

iface wlan0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
    bridge_ports wlan0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show wlan0 | grep ether | awk '{print $2}'`

root@raspberrypi:~#

laurentderu, 2016-05-03:
Your interfaces files looks ok (apart from the formatting :) ), but you probably need to either manually configure a bbbb::/64 address on it manually and add a route towards aaaa::/64 or activate RA message processing (see https://github.com/cetic/6lbr/wiki/6LBR-Connection).

Beware that several WifI controllers does not support promiscuous mode and so can not join a bridge.

janeksz, 2016-05-03:
Thanks a lot.
Now my 6lbr.conf is:

root@raspberrypi:~# cat /etc/6lbr/6lbr.conf
#This file contains a default configuration for Raspberry PI platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

#MODE=ROUTER
MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
CREATE_BRIDGE=0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

NODE_CONFIG=/etc/6lbr/node_config.conf

LOG_LEVEL=3 #INFO and above onlyroot@raspberrypi:~#

//-----------------
Now when I follow your link for connection I have
root@raspberrypi:~# sysctl -w net.ipv6.conf.wlan0.accept_ra=1
net.ipv6.conf.wlan0.accept_ra = 1
root@raspberrypi:~# sysctl -w net.ipv6.conf.wlan0.accept_ra_rt_info_max_plen=64
sysctl: cannot stat /proc/sys/net/ipv6/conf/wlan0/accept_ra_rt_info_max_plen: No such file or directory
root@raspberrypi:~#

or 

root@raspberrypi:~# sudo ip -6 addr add bbbb::101/64 dev wlan0
RTNETLINK answers: Permission denied
root@raspberrypi:~# route -A inet6 add aaaa::/64 gw bbbb::100
SIOCADDRT: File exists
root@raspberrypi:~#

Why is 'Permission denied'

//--------------------------------
The resulting ifconfig before starting 6lbr service is:
root@raspberrypi:~# ifconfig
br0       Link encap:Ethernet  HWaddr 5c:d9:98:ae:09:03
          inet6 addr: fe80::7574:d96e:7f1c:b36d/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr b8:27:eb:0e:18:c9
          inet6 addr: fe80::8fff:4653:7046:1e08/64 Scope:Link
          inet6 addr: bbbb::101/64 Scope:Global
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4416 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4416 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:357696 (349.3 KiB)  TX bytes:357696 (349.3 KiB)

wlan0     Link encap:Ethernet  HWaddr 5c:d9:98:ae:09:03
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:794 errors:0 dropped:6667 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:107190 (104.6 KiB)  TX bytes:0 (0.0 B)

root@raspberrypi:~#

It doesn't look good to me

//------------------------------
After stating 6lbr service
root@raspberrypi:~# service 6lbr start
root@raspberrypi:~# ifconfig
br0       Link encap:Ethernet  HWaddr 5c:d9:98:ae:09:03
          inet addr:169.254.77.32  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::7574:d96e:7f1c:b36d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3 errors:0 dropped:0 overruns:0 frame:0
          TX packets:48 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:168 (168.0 B)  TX bytes:10566 (10.3 KiB)

eth0      Link encap:Ethernet  HWaddr b8:27:eb:0e:18:c9
          inet6 addr: fe80::8fff:4653:7046:1e08/64 Scope:Link
          inet6 addr: bbbb::101/64 Scope:Global
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4544 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4544 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:368064 (359.4 KiB)  TX bytes:368064 (359.4 KiB)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          inet6 addr: fe80::b0fa:866e:73cd:21ca/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3 errors:0 dropped:0 overruns:0 frame:0
          TX packets:89 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:210 (210.0 B)  TX bytes:18722 (18.2 KiB)

wlan0     Link encap:Ethernet  HWaddr 5c:d9:98:ae:09:03
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:794 errors:0 dropped:6667 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:107190 (104.6 KiB)  TX bytes:0 (0.0 B)

root@raspberrypi:~#
//------------------------------------
What is the webpage address of 6lbr now? Is it bbbb::100 or 169.254.77.32?
//------------------------------
// starting nodejs webserver - still doesn't work

janeksz, 2016-05-04:
Another question:
If i wish to use routing mode as described here https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration 
RAW_ETH=0
BRIDGE=0
DEV_ETH=eth0
DEV_TAP=tap0 

what is the coresponding MODE? Is it MODE=ROUTER?

janeksz, 2016-05-04:
Finally got it working - both 6lbr and a nodejs webserver on the same RPI using Routing mode - thanks Laurent. The only caveat is that I am not able to access 6lbr website from another computer on the network, so the last question is: Is there a fix for it?
[image]

laurentderu, 2016-05-04:
About your first question, there is no correspondance between the operational mode, Router, SmartBridge, ... and the interface configuration, raw, bridged, routed. The interface configuration is the way 6LBR is connected to your lan, not how it works.

From another computer you have to do something similar to what you did on the RPi, either manually add an address or activate the RA auto configuration and get an address from 6LBR.

janeksz, 2016-05-04:
@laurentderu 
Can you be more specific what commands I need to issue manually or how to activate RA auto configuration on my Macbook or desktop PC (Win7) in order to be able to view 6lbr webpage?

My settings again (6lbr.conf):
MODE=ROUTER
RAW_ETH=0
BRIDGE=0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0

after starting 6lbr service on RPi I issue the following commands on RPi:
// Manual connection https://github.com/cetic/6lbr/wiki/6LBR-Connection 
sudo ip -6 addr add bbbb::101/64 dev wlan0
sudo ip -6 addr add bbbb::101/64 dev tap0
route -A inet6 add aaaa::/64 gw bbbb::100

my RPi based webserver works from there, being able to read the webpage of 6lbr (programmatically), but the 6lbr webpage is not accessible to other computers on the network.

Thanks again for your help.

laurentderu, 2016-05-10:
Your interfaces are configured in routed configuration, it means that you need explicit routing between you LAN, 6LBR and the WSN. This is not the easiest configuration to set up, especially if 6LBR is running Router mode as this will require three subnet.
An easier configuration would be to switch to SmartBridge Mode and have a RADVD running on the tap0 interface, advertising to 6LBR that the RPi is the router to be used, with something like :

<pre>
interface tap0
{
        AdvSendAdvert on;
        prefix aaaa::0/64
	{
                AdvOnLink on;
                AdvAutonomous on;
        };
        AdvDefaultLifetime 1800;
};
</pre>

The wlan0 can keep it's bbbb::/64 address

I will update the documentation for the routed configuration to make this much clearer I hope.

On MacOS-X, you may to enable IPv6 or IPv6 auto configuration, it's in the advanced taboo the network settings. On Windows it's enabled by default.

INemesisI, 2017-01-19:
Hello,
first of all, thank you for those elaborate answers. It actually helped me a lot with my very similar setup. I have a RPI running 6lrb and a MQTT server. 
At first, the server could not reach the subnet of my nodes, so no connection could be established. Adding the route manually ("sudo route -A inet6 add aaaa::/64 gw bbbb::100") fixed the problem for me. The problem is that I need to add the route after every restart. Is there a way to add it automatically or make it persistent?

laurentderu, 2017-01-20:
There are several way to do it, you can configure Linux to configure the interface from the RA messages send by 6LBR, either by adding the following lines in /etc/sysctl.conf or /etc/sysctl.d/ :

    sysctl -w net.ipv6.conf.br0.accept_ra=1
    sysctl -w net.ipv6.conf.br0.accept_ra_rt_info_max_plen=64

Or you can set in your 6lbr.conf file the following line :

    ITF_AUTOCONF=1

Another possibility is to add the route creation in your network manager or in /etc/network/interfaces

stnitomy, 2017-03-22:
HI guys,
Am facing some issues in starting my 6lbr:

I was going through the installation of 6lbr integerated WSN from 
https://github.com/cetic/6lbr/wiki/Other-Linux-Software-Configuration

My current JAVA version is:
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)

Also i have edited /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait dwc_otg.speed=1

Have disabled getty on Raspberry Pi serial line
sudo systemctl stop serial-getty@ttyAMA0.service 

When i am trying to run make cooja-small, I am encountering this error:

BUILD SUCCESSFUL
Total time: 2 seconds
~/6lbr/examples/6lbr/test
if [ ! -f ~/.cooja.user.properties ]; then cp ./conf-templates/cooja.user.properties.example ~/.cooja.user.properties; fi
if [ `whoami` = root ] && [ ! -f /root/.cooja.user.properties ] ; then cp ./conf-templates/cooja.user.properties.example /root/.cooja.user.properties; fi
if ! grep -Fq "serial2pty" ~/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/serial2pty/" ~/.cooja.user.properties; fi
if ! grep -Fq "radiologger-headless" ~/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/radiologger-headless/" ~/.cooja.user.properties; fi
if [ `whoami` = root ] && ! grep -Fq serial2pty /root/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/serial2pty/" /root/.cooja.user.properties; fi
if [ `whoami` = root ] && ! grep -Fq radiologger-headless /root/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/radiologger-headless/" /root/.cooja.user.properties; fi
java -jar ../../../tools/cooja/dist/cooja.jar -quickstart=`cat ./coojagen/output/LASTFILE`
 INFO [main] (Cooja.java:1315) - > Starting Cooja
 INFO [main] (Cooja.java:2856) - External tools default settings: /external_tools_linux.config
 INFO [main] (Cooja.java:2886) - External tools user settings: /home/pi/.cooja.user.properties
 INFO [main] (Simulation.java:431) - Simulation random seed: 9194033042561454866
 INFO [main] (CompileContiki.java:140) - > make TARGET=cooja clean 
 INFO [main] (CompileContiki.java:140) - > make slip-radio.cooja TARGET=cooja 
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:171: Error: bad instruction `popl %esi'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:172: Error: bad instruction `popl %edx'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:173: Error: bad instruction `popl %ecx'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:174: Error: bad instruction `popl %ebx'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:175: Error: bad instruction `popl %eax'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:176: Error: bad instruction `leave'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:177: Error: bad instruction `ret'
FATAL [main] (ContikiMoteType.java:350) - >> make: *** [obj_cooja/cooja_mtarch.o] Error 1
FATAL [main] (ContikiMoteType.java:350) - >> ../../../Makefile.include:240: recipe for target 'obj_cooja/cooja_mtarch.o' failed
FATAL [main] (ContikiMoteType.java:350) - >> Process returned error code 2
FATAL [main] (ContikiMoteType.java:353) - Compilation error: Bad return value

What might be the reason for this error? Did I miss something in configuration????

closed 2017-12-14 by laurentderu

janeksz, 2018-04-06:
thanks, works well