cetic/6lbr #156: Cannot open web page on CC2650 cc26xx-web-demo from 6lbr

chenek, 2016-04-13:
I git latest Contiki and build cc26xx-web-demo for my CC2650STK. The following is the UART output

Starting Contiki-3.x-2278-g8f06451
With DriverLib v0.44336
TI CC2650 SensorTag
Net: sicslowpan
MAC: CSMA
RDC: ContikiMAC, Channel Check Interval: 16 ticks
RF: Channel 25
CC26XX Web Demo Process
CC26XX Web Server
CC26XX CoAP Server
6LBR Client Process
CC26XX MQTT Client Process
CC26XX Net UART Process
Could not open flash to load config

I also git latest 6lbr and join cc2650STK to 6lbr. I can see CC2650STK list on 6lbr sensor list.
[image]
[image]

However, I cannot open the web page when I access it. What would be the problem?
timeout
[image]

janeksz, 2016-04-13:
maybe reading this one:
https://github.com/cetic/6lbr/wiki/6LBR-Connection 
will help you

chenek, 2016-04-14:
Finally, I find out the problem is caused by setting "#define RF_BLE_CONF_ENABLED 1" in Contiki cc26xx-web-demo. After I revise it to "#define RF_BLE_CONF_ENABLED 0", it works now.

closed 2016-04-14 by chenek

desword, 2018-10-30:
> Finally, I find out the problem is caused by setting "#define RF_BLE_CONF_ENABLED 1" in Contiki cc26xx-web-demo. After I revise it to "#define RF_BLE_CONF_ENABLED 0", it works now.

Hi, I encountered the same problem, however, I am unable to access the web page of the sensor, even after I have set the "#define RF_BLE_CONF_ENABLED 1"  to 0.

chenek, 2018-10-30:
Contik/6lbr are changed a lot since this thread was started. I would suggest you to refer to https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html?m=1 and test with Contiki-NG

desword, 2018-10-30:
Thanks I have fixed this

desword, 2018-10-30:
> Contik/6lbr are changed a lot since this thread was started. I would suggest you to refer to https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html?m=1 and test with Contiki-NG

Thanks again for the previous help.
And I found that you encountered a same problem, we may start a new issue session but I want to communicate with you.

The problem is that I got a bad request when I tried to set the IP broker of MQTT on my sensors' web page. I have opened a new issue session and can you give me some help? thanks.
https://github.com/contiki-ng/contiki-ng/issues/736