cetic/6lbr #158: Porting to IMx6 Alignment Trap

gregcjames, 2016-04-18:
Hi
I am trying to port the 6lbr to a IMx6 processor.

It generates a Alignment trap when it executes the  uip_ipaddr_copy in the function uip_nd6_ns_output that is found in core/net/ipv6/uip-nd6.c

uip_ipaddr_copy((uip_ipaddr_t *) &UIP_ND6_NS_BUF->tgtipaddr, tgt);

The compiler flags are

CFLAGSNO = -Wall -g -I/usr/local/include $(CFLAGSWERROR)
CFLAGS  += $(CFLAGSNO) -O0 -g
CFLAGS  +=  -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard 
CFLAGS  +=   -mfpu=neon -mtune=cortex-a9 

LDFLAGS += -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed

I am compiling it on the imx6 host  with gcc 4.9.2

Any help would be appreciated 

Regards 

Greg James

laurentderu, 2016-04-21:
Are you using the master branch or the develop branch ? On the develop branch I fixed some time ago a similar bug and there is another fix from Contiki that I will merge soon, see https://github.com/contiki-os/contiki/pull/1394

gregcjames, 2016-04-21:
> Contiki that I will merge soon, see contiki-os/contiki#1394

Hi Laurent
Thanks for your response. I am currently using the Master branch.
I will try the develop branch

Greg James

Wireless Motorsport
PO Box 1325
Mona Vale 2103
Australia

Ph +61 (0) 409566119

dbjorklu, 2016-05-27:
I also had alignment traps on an arm. The memory access addresses in the trap messages showed the problem was in accessing the uip_buf buffer, more specificly in the UIP_IP_BUF part. I suspected it was an alignment problem of the UIP_IP_BUF uip_ip_hdr struct into the uip_buf array, tried increasing `UIP_CONF_LLH_LEN` to 16 and the alignment traps went away. 
Then I found this in `6lbr-conf-zoul.h`
```
// IP packet must be 4 bytes aligned
#undef UIP_CONF_LLH_LEN
#define UIP_CONF_LLH_LEN 16
``` 
Seems they had the same problem.
Well just increasing `UIP_CONF_LLH_LEN` is not ok, as the ethernet header read is in fact 14 and not 16 bytes, so I padded the `uip_eth_hdr` part of `uip_buf` with two bytes in the `eth_drv_input` function.
It seems I still have something to fix though, but the alignment traps went away.

closed 2017-12-14 by laurentderu