cetic/6lbr #159: unable to access webserver on a node using 6lbr router

shatruddha, 2016-04-19:
Hi
I'm having a strange problem, I'm able to access the webpage at router [bbbb::100] but when a node is connected to it, I'm not able to open the webpage hosted there. following are the details
Target=srf06-cc26xx Board=srf06/cc13xx
one cc13xx module is programmed as slip radio and is connected to RPi. other node is programmed with cc26xx-web-demo
has anyone else faced the same issue?
further information -
as mentioned on http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
netstat -rn does not show me a route to aaaa::
I tried to ping the node, that also fails with "network is unreachable"

laurentderu, 2016-04-21:
You need to configure your local network to access the BR, see https://github.com/cetic/6lbr/wiki/6LBR-Connection

Praveen-Kumar-K, 2016-05-02:
Hi,
I have configure 6lbr on beagle bone black in Router mode, I am able to view [bbbb::100] web page from my PC. In web page how can i access webserver or coap page placed at sensor tab.
actually i want to connect device through coap, how it could be achieved, please any one help me out regarding this.

Thanks,
Praveen Kumar K

janeksz, 2016-05-03:
You need to have a Copper - Firefox extension for Coap. Right click on the Coap link for the mote and open in a new tab. That's it.
[image]
[image]

Praveen-Kumar-K, 2016-05-03:
Hi janeksz,
   I want to implement CoAP application in my node (cc2538dk) and want to control a led on it. I have configure 6LBR on Beagle bone black in Router mode, What code should i flash to cc2538dk to implement CoAP.
I am having Contiki 3.0 OS, can you help me in this regard.

Thanks,
Praveen Kumar K

laurentderu, 2016-05-03:
You can use either the er-rest-example from Contiki or 6lbr-demo with the CoAP server activated, but you have to code the resources controlling your LED.

Praveen-Kumar-K, 2016-05-04:
Thanks laurentderu,
i have flashed er-rest-example to cc2338dk and running 6LBR on Beagle bone black, but i am unable to open webserver or CoAP page from my PC. I am able to access up to basic webpage @bbbb::100 only but unable to move forward to open CoAP page.
can any one let me know, is there any flags or macros should be enable in code before compiling to enable coap services.

Thanks,
Praveen Kumar k

janeksz, 2016-05-04:
@Praveen-Kumar-K 
I don't know the cc2358dk platform, so my advice is to ask for help from TI E22 forum for some pointers and examples. You might check as well https://github.com/thingsquare/hardware , http://www.openmote.com/home.html

Praveen-Kumar-K, 2016-05-05:
Hi janeksz,
Thanks for links,
Actually i need to connect node to internet and implement CoAP on it. I am able to view the 6LBR server page from my PC, but iam unable to forward to open CoAP page from it.
Can any one suggest how to implement this.
Thanks,
Praveen Kumar Kamma

janeksz, 2016-05-08:
Your 6LoWPAN motes have to have a proper firmware. Either use the Contiki examples for your platform, modify examples from other platforms or create your own. Reading #11.3 from this book http://wireless.ictp.it/school_2015/book/book.pdf might help in understanding CoAP.

Praveen-Kumar-K, 2016-05-09:
Hi janeksz,
Thanks for book link,
in that they mention that Border router is RPL-border router, can i use Cetic-6lbr in place of that, can any one clarify this.

Thanks,
Praveen Kumar K

mario9876, 2016-12-23:
Hi, I have a problem. When I click on coap, It opens the page, but clicking on coap discover not find the resources. How can I do?

paishreeni, 2017-01-20:
[image]

Is it because of this error i'm not able to open webserver?

laurentderu, 2017-01-20:
@paishreeni that means your BR has lost the connection with the slip-radio. Although you should be still able to connect to the BR.

paishreeni, 2017-01-20:
How to proceed further if I need to open the webserver?

closed 2017-06-15 by laurentderu

KavinduS, 2019-09-11:
How can I access the node using COAP.  Copper add on is not supported now.