cetic/6lbr #161: [Question] Compiling 6lbr to run on Intel Quark

hoqhuuep, 2016-04-21:
Hello,

I would like to get 6lbr running on a gateway with an Intel Quark processor. The gateway is the MXE-101i from Adlink (http://www.adlinktech.com/PD/web/PD_detail.php?pid=1563). It is very similar to the Galileo. I see that Contiki supports the Galileo platform (https://github.com/contiki-os/contiki/tree/master/platform/galileo).

The gateway is running WindRiver linux. I have to cross-compile 6lbr from 32-bit Ubuntu. I am told that I have to "add “-m32 -march=i586” to C and C++ flags when building the applications". However I do not know to which makefiles I need to add these flags.

Any advice you can offer is greatly appreciated.

Cheers!

laurentderu, 2016-04-21:
The easiest way is to do something like :

    CFLAGS="-m32 -march=i586" make bin/cetic_6lbr_router

A more robust solution would be to create a ghost platform and add it in the Makefile, like :

examples/6lbr/platform/quark/Makefile.quark

    CFLAGS+=-m32 -march=i586
    PROJECTDIRS += platform/native
    include platform/native/Makefile. native

There might be some other minor modifications needed :)

hoqhuuep, 2016-04-22:
Hi Laurent,

Thank you very much for the quick reply!

By following your first suggestion, I have managed to get 6lbr to compile for the correct architecture. These are the exact commands I have used.

```bash
> sudo apt-get install libncurses5-dev bridge-utils
> cd ~
> git clone https://github.com/cetic/6lbr
> cd 6lbr
> git submodule update --init --recursive
> cd examples/6lbr
> CFLAGS="-m32 -march=i586" make bin/cetic_6lbr_router
```

However, the first time I ran these commands I was faced with the following error.

```
user@instant-contiki:~/6lbr/examples/6lbr$ CFLAGS="-m32 -march=i586" make bin/cetic_6lbr_router
make TARGET=native CETIC_6LBR_ROUTER=1 WITH_UDPCLIENT=0
make[1]: Entering directory `/home/user/6lbr/examples/6lbr'
  CC        cetic-6lbr.c
<command-line>:0:0: warning: "CETIC_6LBR_ROUTER" redefined [enabled by default]
<command-line>:0:0: note: this is the location of the previous definition
  LD        cetic-6lbr.native
cetic-6lbr.co: In function `process_thread_cetic_6lbr_process':
/home/user/6lbr/examples/6lbr/cetic-6lbr.c:545: undefined reference to `udp_client_process'
collect2: ld returned 1 exit status
make[1]: *** [cetic-6lbr.native] Error 1
rm cetic-6lbr.co
make[1]: Leaving directory `/home/user/6lbr/examples/6lbr'
make: *** [bin/cetic_6lbr_router] Error 2
user@instant-contiki:~/6lbr/examples/6lbr$ 
```

I have avoided this error by commenting out the following line from /home/user/6lbr/examples/6lbr/Makefile

```make
ifneq ($(WITH_UDPCLIENT),0)
# CFLAGS += -DUDPCLIENT=1 -DUDP_CLIENT_AUTOSTART
PROJECT_SOURCEFILES += udp-client.c
endif
```

Running the command to compile again, after commenting out this line, does result in a successful compilation. However I am unsure what the consequences of this modification are. Could you please explain to me what features I will not be able use without this UDPCLIENT and, if they are important, how I might be able to fix the error otherwise?

I also have some questions about the configuration. But before I ask, I think I should explain what my goal is. I need to have 6lbr running on the gateway and I need to be able to **programmatically get a list of the connected sensor nodes and send CoAP requests to the sensor nodes with software running on the gateway itself**. I believe I can get the list of sensor nodes by scraping them of http://[bbbb::100]/sensors.html (if there is a better way please let me know!). And, if the routing is setup correctly, I know can send the CoAP requests using the Californium Java library. **I do not need to be able to access the webui (http://[bbbb::100]) or send CoAP requests to the sensor nodes from any other device connected to the gateway via ethernet, though this would be useful for debugging**.

## Configuration attempt 1 - raw ethernet

I know from the wiki that this configuration will not allow me to do the requests I need from the gateway itself, and hence I know this is the wrong approach. I just tried it first as it is the simplest and I wanted to be sure the compilation had been successful.

```bash
MODE=ROUTER
DEV_ETH=eth0
RAW_ETH=1
BRIDGE=0
CREATE_BRIDGE=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
```

Using the above configuration I was able to see the webui (http://[bbbb::100]) from my laptop connected to the gateway via ethernet. From this I think the compilation was successful, this is further than I got without your instructions :)

However, I need to be able to access this webui from the gateway itself (or have another way to list the sensor nodes). And I need to be able to send CoAP requests to the sensor nodes from the gateway itself.

## Configuration attempt 2 - bridge mode

Here is my attempt at getting bridge mode going. If this is the wrong approach for my use-case please let me know!
[... 187 lines cut ...]
I notice that the bridge interface is not created... I think I do have bridge-utils installed.

```
root@WR-IntelligentDevice:~# brctl --help
BusyBox v1.19.4 (2015-05-27 16:22:46 CST) multi-call binary.

Usage: brctl COMMAND [BRIDGE [INTERFACE]]
```

And the result is that I cannot access or ping the router or the nodes.

```
root@WR-IntelligentDevice:~# ping6 bbbb::100
connect: Network is unreachable
```

```
root@WR-IntelligentDevice:~# wget http://[bbbb::100]
--2016-04-21 22:32:11--  http://[bbbb::100]/
Connecting to bbbb::100:80... failed: Network is unreachable.
```

Could you please give some advice as to what might be wrong with my configuration?

Thanks again for your support!

hoqhuuep, 2016-04-26:
Never mind. This was my mistake. I had not given execute permissions to the scripts in `/etc/6lbr/ifup.d` and `/etc/6lbr/ifdown.d`. After fixing this, things are looking much better! :)

I am still interested in reading your opinion on the consequences of commenting out the line in the makefile (`# CFLAGS += -DUDPCLIENT=1 -DUDP_CLIENT_AUTOSTART`). And also if you have any comments on my configuration for my use-case.

But other than that I think my issue is resolved. Thanks again!

laurentderu, 2016-04-26:
6LBR comes in several flavors, and one of them can be used to make it behave like an intermediate RPL node, a 6LR. In that case instead of the udp-server we enable the up-client so the 6LR is registered in the sensors list of the BR.

Here, for a reason I can't understand at first sight, it seems your make program thinks that WITH_UDPCLIENT is set to 1 even though it should not be when building a cetic_6lbr_router target. Maybe it's a side effect of forcing CFLAGS or a side effect of the cross-compilation toolchain ?

I should have a closer look at it but there are no consequences of commenting it out as a workaround, unless you want also the 6LR mode of course.

hoqhuuep, 2016-04-28:
Cheers!

closed 2016-04-28 by hoqhuuep

prasanthmaran, 2016-06-21:
hi bro,
 I am also using MXE-101i gateway but problem is i dont konw how to start.
please help me to get the solution.

prasanthmaran, 2016-06-21:
i want to know how to compile my c program from 64bit ubuntu 14 to gateway

hoqhuuep, 2016-06-26:
Without knowing anything more about what you are compiling, it should be as simple as adding "-m32 -march=i586" to your GCC flags.