cetic/6lbr #163: 6lbr in Bridge mode on RPI3 (wlan0)?

flygbengan, 2016-05-01:
What should be the content of the 6lbr.conf and /etc/network/interfaces files if I want to run 6lbr in Bridge mode on a Raspberry PI 3 via its wlan inferface (wlan0)?

flygbengan, 2016-05-01:
The result of running ifconfig on the RPI3 without any 6lbr started:

ifconfig
enxb827eb24c931 Link encap:Ethernet  HWaddr b8:27:eb:24:c9:31  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3909 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2486 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4482774 (4.4 MB)  TX bytes:270464 (270.4 KB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:951 errors:0 dropped:0 overruns:0 frame:0
          TX packets:951 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:100146 (100.1 KB)  TX bytes:100146 (100.1 KB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:71:9c:64  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:8658 errors:0 dropped:268 overruns:0 frame:0
          TX packets:5562 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7999913 (7.9 MB)  TX bytes:704820 (704.8 KB)

laurentderu, 2016-05-03:
The default 6lbr.conf and network.interfaces.dhcp examples should work provided you replace eth0 by wlan0.
However you wlan controller must be able to enter promiscuous mode in order to have a functional bridge.

flygbengan, 2016-05-03:
Thank you. I googled, and it seems that the WLAN chip on RPI3 unfortunately does not support promiscuous mode. 

Could there be a problem for a 6lbr point of view that the eth0 if is called enxb827eb24c931 instead of eth0 in the Ubuntu Mate OS which I use on my RPI3?

laurentderu, 2016-05-03:
The name of the interface should not matter, as long as it is properly referenced in 6lbr.conf

You can use the routed interface mode to connect your 6LBR to your WiFi (see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration) This configuration does not require a bridge but you need additional configuration to set up the routes

janeksz, 2017-01-11:
should the routed interface mode work on RPi3? I am having the same problem
@flygbengan, do you have it working?

closed 2017-06-15 by laurentderu