cetic/6lbr #165: Is there some good source of step-by-step documentation in setting up 6LBR

rohitsivakumar, 2016-05-03:
I am spent two weeks without having made any progress in trying to figure out in which modes I must configure 6LBR to connect with the TI's sensortags to try out how easy/complex it is to setup contiki. I am confused it should be configured in Router mode or Smart-bridge mode. If in Router mode, I can access the http://[bbbb::100] webpage but if in Smart-bridge, even that is lost. In rourter mode, I am unable to ping the wsn nodes (Ti sensorTags). TI folks ask me to find out from contiki guys while contiki is a protocol and 6 LBR is a vm layer meant to route wsn network to ipv6/ipv4 external internet.

It is all so confusing to set things up. Don't we have a simple guide on how to get going? I have not yet entered the contiki layer and still stuck on getting WSN nodes to ping from a outside win/linux PC (external host)

laurentderu, 2016-05-03:
You can follow the wiki pages, the various operating mode are described as well as the interface configurations. Each mode has its pro and cons and can not be used in place of another, the whole network configuration must match the 6LBR configuration, but this is the case for any network element you use.

TI made a exhaustive step by step tutorial for BeagleBone and CC26xx but it could also be applied for other platforms : http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples

(Conceptually 6LBR could be seen as a VM running on your RPi/BeagleBone, hence the need of a tap interface and a bridge to link to the external world)

alterschwede, 2016-09-05:
HI @rohitsivakumar 

Have you solved your problem?
You probably had the same problem as I had... :-)
You must change all: Prefix, 6LoWPAN contex 0, Manual address from aaaa:: to fd00::
(This is default in Contiki since ... May 2016 ... i think.)
And...
You have to add the appropriate route on the computer from where you interact with the node too!
On a Mac: sudo route add -inet6 -prefixlen 64 fd00:: bbbb::100

Read this post:
https://github.com/cetic/6lbr/issues/150

//Stefan

closed 2017-06-15 by laurentderu