cetic/6lbr #168: service 6lbr don't start!!!

Ulisseruiz, 2016-05-03:
Hi i have a problem with 6lbr.

i have ubuntu last 64bit in pc intel atom.
the hardware as router is smartrf06eb-cc13xxem.
the system contain 150 nodes with cc1310.
But i not are expert of this and contiki.

can i help me!?

mar 3 mag 2016, 21.59.21, CEST : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyUSB0 -r -t eth0 -B 115200 -U /u$
Contiki-develop-20160421-3-g4713ba1 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-05-03 21:59:21.370480: NOTICE: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20160421-3-g4713ba1)
2016-05-03 21:59:21.370666: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-05-03 21:59:21.370812: INFO: NVM: NVM Magic : 2009
2016-05-03 21:59:21.370819: INFO: NVM: NVM Version : 5
2016-05-03 21:59:21.370822: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-05-03 21:59:21.370844: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-05-03 21:59:21.370856: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-05-03 21:59:21.370860: INFO: MAC: Using 'csma' mac driver
2016-05-03 21:59:21.371213: INFO: SLIP: SLIP started on /dev/ttyUSB0
2016-05-03 21:59:21.381727: INFO: SCMD: Started br-cmd process
2016-05-03 21:59:21.381765: INFO: BR-RDC: Reset SLIP Radio
2016-05-03 21:59:21.381781: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:22.381128: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:23.381098: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:24.381129: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:25.381139: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:26.381099: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:27.381123: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:28.381074: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:29.381068: INFO: BR-RDC: Fetching MAC address
2016-05-03 21:59:30.381075: INFO: BR-RDC: Fetching MAC address

this is log by /var/log/6lbr.log

AndrewBrannan, 2016-05-03:
It can't find your slip-radio.  Check 6lbr.conf to make sure that DEV_RADIO is set to the correct interface.

Ulisseruiz, 2016-05-04:
yes is set exactly.

laurentderu, 2016-05-04:
What firmware are you using on the slip-radio ? could you check using a serial terminal that it starts up and wait for commands ?

Ulisseruiz, 2016-05-05:
i don't have a ftdi232 but smartrf06eb with cc1310 on board.
with serial terminal no are text.

Ulisseruiz, 2016-05-05:
someone has the .bin code already compiled to comunicate with serial of board?

Ulisseruiz, 2016-05-06:
RPL-BORDER-ROUTER work good with tool tunslip6

laurentderu, 2016-05-10:
You need a slip-radio firmware on the CC13xx adapted for the CC13xx and also have the art configuration set properly, otherwise the 6lbr can not communicate with the slip-radio firmware.

Ulisseruiz, 2016-05-10:
I have a comunication with 6lbr and slip-radio.
I view  all web page of 6lbr and view node but dont comunicate with this with web server or mqtt

laurentderu, 2016-05-10:
https://github.com/cetic/6lbr/issues/173

closed 2016-05-10 by laurentderu