cetic/6lbr #170: Implement CoAP on cc2538dk and view in 6LBR server page

Praveen-Kumar-K, 2016-05-09:
Hi,
I have configure 6lbr on beagle bone black in Router mode, I am able to view [bbbb::100] web page from my PC. In web page how can i access webserver or coap page placed at sensor tab.
actually i want to connect device through coap, how it could be achieved, please any one help me out regarding this.

Thanks,
Praveen Kumar K

Praveen-Kumar-K, 2016-05-11:
Hi,
I have flashed ER-REST example from Contiki-3.0 to CC2538dk's, One with server and other with client. 6LBR is running in beagle bone black, slip is running on cc2538dk. How could i communicate with both nodes with CoAP. can any one help me in this regard.
Thanks,
Praveen Kumar K

laurentderu, 2016-05-12:
er-rest-example has many hardcoded network parameters and can not be used on a real network. Instead you should use a better demo, like 6lbr-demo.

Praveen-Kumar-K, 2016-05-17:
Thanks  laurentderu,
may i know what are the functionalities performed by 6lbr-demo, and how could i configure for CoAP server for one node and CoAP client for other node, Can any one help me in this regard.
Thanks,
Praveen kumar k

stnitomy, 2017-04-24:
@Praveen-Kumar-K : Hello Praveen, Did you figured out how to configure CoAP server for one node and client for other???
Thanks

closed 2017-06-15 by laurentderu