cetic/6lbr #172: Issue adding route to aaaa: prefix address

sharidas, 2016-05-09:
Hi,

For last few days I am trying to figure out how to get 2 CC2531 devices ping each other. So my scenario goes like this:

1) I have an embedded board sabre sd. A cc2531 device is connected to this board.
2) I have an embedded board Sabre AI Dul. A cc2531 device is connected to this board.

Both the embedded boards are connected to the switch. I have following configurations on each of the baord.

6lbr ( same for both the boards):
`MODE=ROUTER`
`RAW_ETH=0`
`BRIDGE=1`
`DEV_BRIDGE=br0`
`DEV_ETH=eth0`
`DEV_TAP=tap0`
`RAW_ETH_FCS=1`
`DEV_RADIO=/dev/ttyACM0`
`BAUDRATE=115200`
`LOG_LEVEL=3`

There is an radvd.conf ( on one of the boards I run radvd):

`interface br0`
`{`
 `   AdvSendAdvert on;`
`    AdvManagedFlag off;`
`    AdvOtherConfigFlag on;`
`    MaxRtrAdvInterval 10;`
`    prefix 2001:db8:2::/64`
`    {`
`        AdvOnLink on;`
`        AdvAutonomous on;`
`        AdvRouterAddr on;`
`    };`
`};`

sysctl.conf ( for both the boards ):
`net.ipv6.conf.all.accept_ra_rt_info_max_plen=64`
`net.ipv4.ip_forward=1`
`net.ipv6.conf.all.forwarding=1`
`net.ipv6.conf.all.accept_ra=2`

tail -30 /var/log/6lbr.log of Sabre SD board:

`2016-04-09 10:30:26.774039: INFO: BR-RDC: Reset SLIP Radio`
`2016-04-09 10:30:26.774109: INFO: BR-RDC: Fetching MAC address`
`2016-04-09 10:30:26.774274: WARN: PF: No input function set`
`2016-04-09 10:30:26.774460: WARN: PF: No input function set`
`2016-04-09 10:30:26.774642: WARN: PF: No input function set`
`2016-04-09 10:30:26.774711: WARN: PF: No input function set`
`2016-04-09 10:30:26.774810: WARN: PF: No input function set`
`2016-04-09 10:30:26.774925: WARN: PF: No input function set`
`2016-04-09 10:30:26.807653: INFO: SLIP: cc2531: Reboot requested`
`2016-04-09 10:30:26.808638: INFO: SLIP: cc2531: Reverting to h/w MAC but not rebooting`
`2016-04-09 10:30:26.837646: INFO: BR-RDC: Got MAC: 00:12:4b:00:07:68:d3:81`
`2016-04-09 10:30:27.159107: WARN: PF: No input function set`
`2016-04-09 10:30:27.589108: WARN: PF: No input function set`
`2016-04-09 10:30:27.649042: WARN: PF: No input function set`
`2016-04-09 10:30:27.729046: WARN: PF: No input function set`
`2016-04-09 10:30:27.774154: INFO: ETH: Eth MAC address : 02:00:07:68:d3:81`
`2016-04-09 10:30:27.774453: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:768:d381`
`2016-04-09 10:30:27.774551: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:768:d381`
`2016-04-09 10:30:27.774634: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::200`
`2016-04-09 10:30:27.774690: INFO: 6LBR: RA Daemon enabled`
`2016-04-09 10:30:27.774721: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'`
`2016-04-09 10:30:27.783534: INFO: 6LBR: Configured as DODAG Root`
`2016-04-09 10:30:27.783601: INFO: 6LBR: Starting as RPL ROUTER`
`2016-04-09 10:30:27.788068: INFO: 6LBR: CETIC 6LBR Started`
`2016-04-09 10:30:27.788159: INFO: SLIP: cc2531: Setting RF channel 0x19`

tail -30 /var/log/6lbr.log of Sabre AI dual board:

`2016-04-06 9:18:09.9445: INFO: BR-RDC: Reset SLIP Radio`
`2016-04-06 9:18:09.9530: INFO: BR-RDC: Fetching MAC address`
`2016-04-06 9:18:09.9720: WARN: PF: No input function set`
`2016-04-06 9:18:09.9912: WARN: PF: No input function set`
`2016-04-06 9:18:09.9987: WARN: PF: No input function set`
`2016-04-06 9:18:09.10090: WARN: PF: No input function set`
`2016-04-06 9:18:09.10155: WARN: PF: No input function set`
`2016-04-06 9:18:09.10254: WARN: PF: No input function set`
`2016-04-06 9:18:09.10379: WARN: PF: No input function set`
`2016-04-06 9:18:09.10604: WARN: PF: No input function set`
`2016-04-06 9:18:09.10679: WARN: PF: No input function set`
`2016-04-06 9:18:09.42589: INFO: SLIP: cc2531: Reboot requested`
`2016-04-06 9:18:09.43584: INFO: SLIP: cc2531: Reverting to h/w MAC but not rebooting`
`2016-04-06 9:18:09.72654: INFO: BR-RDC: Got MAC: 00:12:4b:00:07:68:d0:da`
`2016-04-06 9:18:09.161921: WARN: PF: No input function set`
`2016-04-06 9:18:09.461916: WARN: PF: No input function set`
`2016-04-06 9:18:09.641918: WARN: PF: No input function set`
`2016-04-06 9:18:10.9141: INFO: ETH: Eth MAC address : 02:00:07:68:d0:da`
`2016-04-06 9:18:10.9728: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:768:d0da`
`2016-04-06 9:18:10.9975: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:768:d0da`
`2016-04-06 9:18:10.10179: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100`
`2016-04-06 9:18:10.10304: INFO: 6LBR: RA Daemon enabled`
`2016-04-06 9:18:10.10377: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'`
`2016-04-06 9:18:10.11255: INFO: 6LBR: Configured as DODAG Root`
`2016-04-06 9:18:10.11381: INFO: 6LBR: Starting as RPL ROUTER`
`2016-04-06 9:18:10.41612: INFO: 6LBR: CETIC 6LBR Started`
`2016-04-06 9:18:10.41739: INFO: SLIP: cc2531: Setting RF channel 0x19`

route -n -6 of sabre sd:

`root@mx6q:~# route -n -6`
`Kernel IPv6 routing table`
`Destination                    Next Hop                   Flag Met Ref Use If`
`::1/128                        ::                         Un   0   1  1581 lo`
`2001:db8:2::/128               ::                         Un   0   1     0 lo`
`2001:db8:2:0:204:9fff:fe02:6b21/128 ::                         Un   0   1    30 lo`
`2001:db8:2::/64                ::                         UAe  256 0     0 br0`
`aaaa::/64                      bbbb::100                  UG   1   0     0 br0`
`bbbb::/128                     ::                         Un   0   1     0 lo`
`bbbb::204:9fff:fe02:6b21/128   ::                         Un   0   1     6 lo`
`bbbb::/64                      ::                         UAe  256 0     1 br0`
`fe80::/128                     ::                         Un   0   1     0 lo`
`fe80::/128                     ::                         Un   0   1     0 lo`
`fe80::204:9fff:fe02:6b21/128   ::                         Un   0   1   108 lo`
`fe80::204:9fff:fe02:6b21/128   ::                         Un   0   1     0 lo`
`fe80::/64                      ::                         U    256 0     0 br0`
`fe80::/64                      ::                         U    256 0     0 eth0`
`ff00::/8                       ::                         U    256 0     0 tap0`
`ff00::/8                       ::                         U    256 7     0 br0`
`ff00::/8                       ::                         U    256 0     0 eth0`
`::/0                           ::                         !n   -1  1   199 lo`
`root@mx6q:~# `

route -n -6 of sabre AI dual:

`root@mx6q:~# route -n -6`
`Kernel IPv6 routing table`
`Destination                    Next Hop                   Flag Met Ref Use If`
`::1/128                        ::                         Un   0   1  2381 lo`
`aaaa::/64                      bbbb::200                  UG   1   0     0 br0`
`bbbb::/128                     ::                         Un   0   1     0 lo`
`bbbb::204:9fff:fe02:7732/128   ::                         Un   0   1     4 lo`
`bbbb::/64                      ::                         UAe  256 0     1 br0`
`fe80::/128                     ::                         Un   0   1     0 lo`
`fe80::/128                     ::                         Un   0   1     0 lo`
`fe80::204:9fff:fe02:7732/128   ::                         Un   0   1     0 lo`
`fe80::204:9fff:fe02:7732/128   ::                         Un   0   1     0 lo`
`fe80::/64                      ::                         U    256 0     0 br0`
`fe80::/64                      ::                         U    256 0     0 eth0`
`ff00::/8                       ::                         U    256 0     0 tap0`
`ff00::/8                       ::                         U    256 4     0 br0`
`ff00::/8                       ::                         U    256 0     0 eth0`
`::/0                           ::                         !n   -1  1     1 lo`
`root@mx6q:~#`

I am able to ping6 aaaa::212:4b00:768:d0da from sabre SD board using interface br0. Similarly aaaa::212:4b00:768:d381 can be ping-ed from sabre ai dual board using br0 board. 

Query: I would like to ping between two cc2531 devices. For this what extra configuration should be required from 6lbr and radvd so that I can have 2 slip radio devices ping each other. Any helping hand here would be helpful for me. In case of more input required for my query kindly let me know so that I can provide the same.

laurentderu, 2016-05-10:
This looks like a similar request to https://github.com/cetic/6lbr/issues/153 This is not possible with 6LBR in RPL Router mode (unless you just need to reach a node and not a while subnet)
You need to have the two 6LBR run in NDP Router mode, or one in NDP Router mode and the second as bridge.

sharidas, 2016-05-18:
@laurentderu Hi thanks for the response. Is it possible to share a sample configuration ( 6lbr.conf ) that needs to be done for NDP-ROUTER mode?

sharidas, 2016-05-19:
@laurentderu Below is the config file that I have for NDP-ROUTER mode:

`#MODE=ROUTER`
`MODE=NDP-ROUTER`
`RAW_ETH=0`
`BRIDGE=1`
`DEV_BRIDGE=br0`
`DEV_ETH=eth0`
`DEV_TAP=tap0`
`RAW_ETH_FCS=1`
`DEV_RADIO=/dev/ttyACM0`
`BAUDRATE=115200`
`LOG_LEVEL=3`

With this setting on both the target machines, I wasn't able to get aaaa:: addresses to ping after removing the ethernet cable. I wonder if my configuration is wrong or so for NDP-ROUTER settings. A helping hand would be appreciated.

laurentderu, 2016-05-24:
THe 6lbr.conf file is ok, now you need to modify the NVM of both NDP Router :

On NDP Router 1 you need to set for example the Ethernet prefix to aaaa::/64 and the WSN prefix to bbbb::/64 and the WSN IP address to bbbb::1
On NDP Router 2 you need to set the Ethernet prefix to cccc::/64 and the WSN prefix to bbbb::/64 and the WSN IP address to bbbb::2. This can be done either using the web interface or the nvm_tool.
Then on NDP Router 1 you need to add a static route cccc::/64 via bbbb::2 and on NDP Router 1, the opposite route : aaaa::/64 via bbbb::1 This must be added in the new nvm.conf file, but the documentation is not yet up to date, I have to complete the page on how to do that.

sharidas, 2016-05-31:
@laurentderu I tried with the radvd.conf for NDP Router 1:
interface br0
{
    AdvSendAdvert on;
    AdvManagedFlag off;
    AdvOtherConfigFlag on;
    MaxRtrAdvInterval 10;
    prefix aaaa::0/64
    {
        AdvOnLink on;
        AdvAutonomous on;
        AdvRouterAddr on;
    };
};

and NDP Router 2 as:

interface br0
{
    AdvSendAdvert on;
    AdvManagedFlag off;
    AdvOtherConfigFlag on;
    MaxRtrAdvInterval 10;
    prefix cccc::0/64
    {
        AdvOnLink on;
        AdvAutonomous on;
        AdvRouterAddr on;
    };
};

With this I wasn't able to add static routes for NDP Router 1: cccc::/64 via bbbb::2 and on NDP Router 2: aaaa::/64 via bbbb::1. It was failing to add the static route. One reason was that for NDP Router 1 the br0 interface didn't had cccc:: prefix IP address and NDP Router 2 didn't had the aaaa:: prefix IP address. Let me know if I need to do anything on the radvd configuration. More over on both the routers, "Eth default router" section is empty. Do I need to update that section? Sorry to ask these questions.

laurentderu, 2016-06-01:
Sorry, I should have been more complete in my explanation, you don't need a RADVD, instead the configuration must be done in each NDP Router, either via the web server or the nvm_tool.
I don't think you need to set the "Eth default router" on the other hand.
I will add a complete example with the configuration file this week-end, it will be much easier to start from that.

phsontung, 2016-09-30:
@laurentderu, Thanks for your information. 
Currently, do we have any update for this sample configuration?

closed 2017-12-14 by laurentderu

Yoshinee, 2018-02-12:
Can we modify the prefix of the WSN and ethernet network