cetic/6lbr #173: 6lbr service & routing

Ulisseruiz, 2016-05-09:
HI i have a system.
view a 6lbr webpage to setting but nodes is unavailable with mqtt or coap or web server.
i need help to setting board server with ubuntu..
i not have a setting of interface eth0 or other setting..
Help me please.
[image]

laurentderu, 2016-05-10:
If you can reach 6LBR but not the node it can either be a radio configuration issue or a network configuration issue.
If you don't see your node in the sensors or the network status page of 6LBr, it's a radio issue (either radio channel or wrong RDC)
If you see your nodes but can not reach it, you need to configure routes on your hosts to reach them, see https://github.com/cetic/6lbr/wiki/6LBR-Connection
If you need IPv4, you need to configure NAT64

Ulisseruiz, 2016-05-10:
i view node in 6lbr page but don't have a correct setting for host.....
can help me!
[image]
[image]

closed 2016-05-12 by Ulisseruiz