cetic/6lbr #176: unable to view nodes in server tab in 6LBR page

Praveen-Kumar-K, 2016-05-10:
Hi,
I am implementing 6LBR on Beagle bone black. I have clone source from github and then build with configuration 
--------------------
MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3
-------------------------------
by starting service 6LBR iam able to view bbbb::100 page.
then i have compiled helloworld code from Contiki-3.0 to cc2538dk and i am able see it from serial window, but i am unable to view from bbbb::100 page servers tab.
my slip radio is cc2531 USB dongle, which is connected to Beagle bone black.

logs of 6LBR is as shown 
--------------------------------------------------------------------------------------
********************************************************************************
Tue May 10 12:12:41 UTC 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins   
Contiki-develop-20160421-3-g4713ba1 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-05-10 12:12:42.035873: NOTICE: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20160421-3-g4713ba1)
2016-05-10 12:12:42.036662: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-05-10 12:12:42.045215: INFO: NVM: NVM Magic : 2009
2016-05-10 12:12:42.045254: INFO: NVM: NVM Version : 5
2016-05-10 12:12:42.045265: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-05-10 12:12:42.045322: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-05-10 12:12:42.045406: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-05-10 12:12:42.045421: INFO: MAC: Using 'nullmac' mac driver
2016-05-10 12:12:42.046325: INFO: SLIP: SLIP started on /dev/ttyACM0
2016-05-10 12:12:42.056870: INFO: SCMD: Started br-cmd process
2016-05-10 12:12:42.057005: INFO: BR-RDC: Reset SLIP Radio
2016-05-10 12:12:42.057043: INFO: BR-RDC: Fetching MAC address
2016-05-10 12:12:42.059755: INFO: SLIP: cc2531: Reboot requested
2016-05-10 12:12:42.061090: INFO: SLIP: cc2531: Reverting to h/w MAC but not rebooting
2016-05-10 12:12:42.062676: INFO: BR-RDC: Got MAC: 00:12:4b:00:07:68:d2:69
2016-05-10 12:12:43.057303: INFO: ETH: RAW Ethernet interface init
2016-05-10 12:12:43.067882: INFO: TAP: opened device /dev/eth0
2016-05-10 12:12:43.068638: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr 90:59:af:4f:7f:94  
          inet addr:192.168.1.190  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fd35:7e3d:9559:0:9259:afff:fe4f:7f94/64 Scope:Global
          inet6 addr: fe80::9259:afff:fe4f:7f94/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:172315 errors:0 dropped:0 overruns:0 frame:0
          TX packets:103753 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:168566909 (160.7 MiB)  TX bytes:31851232 (30.3 MiB)
          Interrupt:40 

2016-05-10 12:12:43.227613: INFO: TAP: Eth MAC address : 90:59:af:4f:7f:94
2016-05-10 12:12:43.228433: INFO: ETH: Eth MAC address : 90:59:af:4f:7f:94
2016-05-10 12:12:43.229303: INFO: LLSEC: Using 'nullsec' llsec driver
2016-05-10 12:12:43.229760: INFO: 6LBR: Security layer initialized
2016-05-10 12:12:43.230752: INFO: SLIP: cc2531: Setting RF channel 0x19
2016-05-10 12:12:43.231511: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:768:d269
2016-05-10 12:12:43.232237: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:768:d269
2016-05-10 12:12:43.233096: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-05-10 12:12:43.233686: INFO: 6LBR: RA Daemon enabled
2016-05-10 12:12:43.234135: INFO: 6LBR: Checking addresses duplication
2016-05-10 12:12:54.235932: INFO: 6LBR: Configured as DODAG Root ::
2016-05-10 12:12:54.237718: INFO: 6LBR: Starting IP64
Starting DHCPv4
2016-05-10 12:12:54.239091: INFO: 6LBR: Starting MDNS
2016-05-10 12:12:54.240546: INFO: 6LBR: Starting as RPL ROUTER
2016-05-10 12:12:54.244807: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-05-10 12:12:54.246476: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-05-10 12:12:54.247690: INFO: NODECFG: Node Config init
2016-05-10 12:12:54.248862: INFO: NODECFG: No node_config.conf file specified
2016-05-10 12:12:54.249819: INFO: HTTP: Starting webserver on port 80
2016-05-10 12:12:54.252052: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2016-05-10 12:12:54.253410: INFO: DNS: DNS proxy started
2016-05-10 12:12:54.254420: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-05-10 12:12:54.255536: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-05-10 12:12:54.256891: INFO: 6LBR: CETIC 6LBR Started
RD client started
2016-05-10 12:12:55.277786: INFO: 6LBR: Set IPv4 address : 192.168.1.102
2016-05-10 12:12:56.506226: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
2016-05-10 12:12:56.757184: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (6)
2016-05-10 12:12:57.256585: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (8)
2016-05-10 12:12:57.759670: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (14)
2016-05-10 12:13:02.324534: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:13:07.035407: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:13:19.324720: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:13:57.841367: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:14:42.076910: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:16:20.532521: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:20:27.523106: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
2016-05-10 12:26:08.461525: INFO: NODECFG: Node has been rejected : aaaa::212:4b00:413:3b01
***************************************************************
and my config file is 
------------------------------------
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : aaaa::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : True
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 2
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

MAC layer : 0
Security layer : 0
Security level : 3
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay disabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : True

NAT 64 Enabled : True
NAT 64 DHCP Enabled : True
NAT 64 static port mapping Enabled : True
NAT 64 RFC-6052 prefix : True
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80
--------------------------------------------------

can any one help me in this regard

Thanks, 
Pravee Kumar k

laurentderu, 2016-05-10:
You have activated the filtering of the node and so nodes not registered in Node Config are rejected by 6LBR. You can disable this in the Configuration page.

Praveen-Kumar-K, 2016-05-11:
Hi,
Thank you  Laurent Deru,
by disabling node filter i am able to see the nodes in web page, but slip acknowledgement timed out Error is not resolved, it  was observed from log file as shown
-------------------------------------
br-rdc: send failed, slip ack timeout (4)
ERROR: BR-RDC: br-rdc: send failed, slip ack timeout
(4)
2015-03-02 0:04:10.634881: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout
(6)
2015-03-02 0:04:11.134765: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout
(8)
2015-03-02 0:04:11.636750: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout 
----------------------------------

laurentderu, 2016-05-12:
These errors may happen at startup when the slip-radio is not correctly synced, also slip-radio built outside of 6lbr tree may not contain yet all the fixes for the slip connection

Praveen-Kumar-K, 2016-05-13:
Hi laurentderu,
Thanks for comment on this. is there any solution to eliminate these errors, How could we fix these issues. please let me know, on this regard.

Thanks,
Praveen Kumar Kamma.

closed 2017-06-15 by laurentderu