cetic/6lbr #177: Using cc2520 with SPI instead of slip-radio?

liulibing, 2016-05-12:
Hi,

I have used Raspberry Pi 2(RPi2) to support 802.15.4 device. The detailed information is as follows:

Hardware: 
        1. Raspberry Pi 2
        2. TI cc2520 (Transceiver only, without MCU 8051)

Steps:
1.connect cc2520 to RaspberryPi with SPI.

> RPi2(PIN)  cc2520
17	VCC
20	GND
11	RST
19	SI
21	SO
23	SCK
24	CSN
22	IO1
18	IO2
15	IO3
16	IO4
13	VEN

2.git clone RPi linux kernel and modify kernel config file and DTS file for TI cc2520
3.build RPi linux kernel and update image files (DTB, kernel, and linux modules) to SDCard
4.run test (discovery nodes, capture packets and sniffer) according to http://wpan.cakelab.org/. 

It means that 6lowpan is supported well in Linux 4.x, but the RPL implemented in Linux is not a good status.

The current solution of border routers almost are using 6lbr, which is based on slip-radio device when we want to develop an advanced border router, such as, using Raspberry Pi with Linux OS. Slip-radio is slip over 802.15.4 radio, but why not to use cc2520 direct (I think it will have better performance and good application)? Do you think it is possible or 6lbr will support this feature?

laurentderu, 2016-05-12:
SPI radio support is not planned in short term for 6LBR, but it should be feasible. Another possibility would be to add support for 802.15.4 linux driver. There is a driver now in Contiki so it could be implemented once release 1.4.0 is out.

liulibing, 2016-05-13:
It's a good news, thanks.

closed 2016-05-13 by liulibing