cetic/6lbr #181: SRF06-CC26xx+ENC28J60 does not work

bbin08, 2016-05-15:
Hi 

Does anybody test the /6lbr/example/6lbr with srf06-cc26xx? It works correctly with cc2538dk, but doesn't work with srf06-cc26xx. 

I am running the command as: make TARGET=srf06-cc26xx BOARD=srf06 all CPU_FAMILY=cc26xx. 

Anyone can help on it? Thanks!

laurentderu, 2016-05-17:
The build command is ok, could you describe with more details what the issue is ? The support for CC26xx is still experimental so there are still bugs lingering.

bbin08, 2016-05-18:
Hi Laurentderu

It doesn't work and no output from the uart and no led flash even. It seems that the contiki main.c doesn't startup.

chenek, 2016-07-11:
@laurentderu I see the same problem. I use the latest 6lbr and do "make TARGET=srf06-cc26xx all". I don't see any UART output on my CC2650DK. I test CC2538DK with the same source code and it works fine. Do you have any suggest to fix the issue?

chenek, 2016-11-24:
@laurentderu any news on this bug. I test this today but it still doesn't work.

laurentderu, 2016-12-16:
I finally found (and solved) the bug, it was a wrong path in the linker scripts that caused some part of the program to be misplaced in memory. But it occurred only when compiling inside 6LBR tree, when doing out of tree build, everything was fine.
The fix is pushed on the dev-1.4.x and develop branches

chenek, 2016-12-16:
Cool! I would test it when I can access to my EVB.

chenek, 2016-12-19:
@laurentderu I can build and run 6lbr () on my CC1310DK with ENC28J60 now but it shows the following page when I access to [bbbb::100]. It seems something wrong because I don't see transitional 6lbr web page. Do you know what's wrong with my build?

[image]

chenek, 2016-12-19:
@laurentderu I also find that my CC1310 sensor node running cc26xx-web-demo cannot join CC1310DK 6lbr. Do you see the same issue? The followings are my UART logs from CC1310 6lbr and cc26xx-web-demo running on CC1310 LaunchPad

**UART logs from CC1310 6lbr:**
Starting Contiki-contiki-merge-20160622-92-gec57ea0
With DriverLib v0.44336
TI SmartRF06EB + CC13xx EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 RF: Channel 25
NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-merge-20160622-92-gec57ea0)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : ffff
INFO: NVM: NVM Version : ffff
ERROR: NVM: Invalid NVM magic number or unsupported NVM version, reseting it...
WARN: NVM: Migrate NVM version 0 towards 1
WARN: NVM: Migrate NVM version 1 towards 5
INFO: NVM: Flashing 6LBR NVM
NOTICE: 6LBR: Log level: 30 (services: ffffffff)
INFO: ETH: ENC28J60 init
INFO: ETH: Eth MAC address : 06:00:08:68:0f:14
INFO: ENC: resetting chip
INFO: ETH: ENC-28J60 Process started
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:868:f14
INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:868:f14
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
INFO: 6LBR: RA Daemon enabled
INFO: 6LBR: Checking addresses duplication
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root fd00::212:4b00:868:f14
INFO: 6LBR: Starting as RPL ROUTER
INFO: 6LBR: CETIC 6LBR Started
INFO: HTTP: Script : /

UART logs from CC1310 LaunchPad running cc26xx-web-demo:
Starting Contiki-3.x-3057-g8fca9ce
With DriverLib v0.46593
TI CC1310 LaunchPad
IEEE 802.15.4: No, Sub-GHz: Yes, BLE: No, Prop: No
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 RF: Channel 25
 Node ID: 8978
CC26XX Web Demo Process
CC26XX Web Server
CC26XX CoAP Server
6LBR Client Process
CC26XX MQTT Client Process

laurentderu, 2016-12-22:
On CC26xx and CC13xx, the flash and ram available is really limited, so by default the full web server is not enabled and instead a light version is started, which only display the status.

Now a node should be able to join the network created by the CC1310 BR, but without more log it's hard to tell what going wrong here...

chenek, 2016-12-22:
Understand that this web thing is due to lack of resources. When you test CC1310 BR, do you let a CC1310 sensor node join successfully? By the way, how can I provide more log for you?

chenek, 2017-01-11:
Issue solved. If anyone have similar question, refer to http://sunmaysky.blogspot.tw/2017/01/build-6lbr-for-cc2650dk-or-cc1310dk.html

whu-inf, 2017-04-11:
@chenek  
Hi
I'm trying to run cetic_6lbr_router.bin on my CC1310DK---ENC28J60---openwrt router kit. I followed each and every instruction in [http://sunmaysky.blogspot.tw/2017/01/build-6lbr-for-cc2650dk-or-cc1310dk.html](url). It failed, and seems that it got stuck at the 'ENC:resetting chip' process. I've already double-checked the  I/O mapping MACROs in head files but NOT able to see any wrong hardware connection.
I got really confused, any idea of this?

hardware connection:
IOID_10------ENC28J60_SCK
IOID_9-------ENC28J60_SI
IOID_8-------ENC28J60_SO
IOID_14------ENC28J60_CS
 
board.h file in 6lbr/platform/srf06-cc26xx/srf06/cc13xx:
#define BOARD_IOID_SPI_SCK        IOID_10
#define BOARD_IOID_SPI_MOSI       IOID_9
#define BOARD_IOID_SPI_MISO       IOID_8

enc28j60-arch-gpio.c in 6lbr/examples/6lbr/platform/srf06-cc26xx:
#undef SPI_CS_PORT
#ifndef CC26XX_ENC28J60_CONF_CS_PORT
#define SPI_CS_PORT   IOID_14
#else
#define SPI_CS_PORT   CC26XX_ENC28J60_CONF_CS_PORT
#endif

UART log from cc1310dk:
[13:50:05.450] Starting Contiki-contiki-base-develop-20170120-2305-g98a3eac
[13:50:05.450] Wit
[13:50:05.459] h DriverLib v0.47020
[13:50:05.459] TI SmartRF06EB + CC13xx EM
[13:50:05.459] IEEE 802.15.4: No, Sub-GHz: Yes, BLE: No, Prop: No
[13:50:05.459]  Net: sicslowpan
[13:50:05.556]  MAC: CSMA
[13:50:05.556]  RDC: nullrdc
[13:50:05.556]  RF: Channel 25
[13:50:05.556]  Node ID: 39795
[13:50:05.556] NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-base-develop-20170120-2305-g98a3eac)
[13:50:05.659] INFO: NVM: Reading 6LBR NVM
[13:50:05.659] INFO: NVM: NVM Magic : 2009
[13:50:05.669] INFO: NVM: NVM Version : 5
[13:50:05.669] NOTICE: 6LBR: Log level: 30 (services: ffffffff)
[13:50:05.680] INFO: ETH: ENC28J60 init
[13:50:05.680] INFO: ETH: Eth MAC address : 06:00:xx:xx:xx:73
[13:50:05.680] INFO: ENC: resetting chip
[13:50:05.680] 
-------------------------------------------------------------------------------------------
then it freezes. I tried resetting the cc1310dk but nothing changed

chenek, 2017-04-11:
This is usually caused by bad SPI connection between LaunchPad and ENC28J60.

laurentderu, 2017-04-11:
As @chenek said, freezing at "ENC: resetting chip" usually means your SPI wiring is wrong. The reset sequence waits until a register in the ENC changes, so it your wiring is wrong you will always read all zero (or one).

closed 2017-12-14 by laurentderu