cetic/6lbr #184: Reading slip-radio packets (like port mirroring)

lucasscola, 2016-05-23:
Hi, I don't know if this is the place to ask/suggest this, I hope it is. I am looking for a way to analyze the 6lowpan traffic on the 6lbr slip radio and I can't find a way to do it. Does someone know if it's possible (without the need of an external sniffer)? I am currently running 6lbr on BBB + cc2531 USB Dongle.
I think that one good way to make it would be if 6lbr has an option to implement something like port mirroring, copying the packet to another interface (say, tap1) and then we can use Wireshark to analyze the traffic.  

Thanks!

laurentderu, 2016-05-24:
Actually you can ask log_6lbr to output the packet in a format parable by Wireshark, sadly you have to recompile 6blr to enable it, I guess we need to add a runtime option to enable it.

In your project-conf.h file you need to add :

    #define WIRESHARK_IMPORT_FORMAT 1

If it works correctly, you should have your packet dumped as something like 

0000 06 AB 43 FC ....

Which can then be imported by wireshark as text input.

You can use the log 6lbr services mask to disable logs from unwanted interfaces.

lucasscola, 2016-05-24:
That's great thank you! That is what I'm looking for

closed 2016-05-27 by lucasscola

aniketdali, 2017-10-04:
where will i find the dumped packets?