cetic/6lbr #185: Lists all the nodes known by the Border Router

BaN-Co, 2016-05-29:
Hello,

Is there a way for 6lbr to lists all the nodes known by the Border Router from console the same way as in the Sensors table of the 6lbr embedded webserver without parsing the 6lbr.log?

BaN-Co, 2016-05-29:
I checked the code and there seems to be that the webserver is the only way to access the node-info module.
I need to get this information about the nodes inside C code app, that will communicate with the available nodes. Is it a good practice to parse the web page to get this data?

laurentderu, 2016-06-01:
My original reply got lost...

You can parse the sensors table of the 6lbr web server, it is proper XHTML. But I agree it is not the best practice as the format of the table could change in the future. I'm a bit reluctant to add a json or xml or... interface as there is not yet a standardized format.

Instead you could add a plugin in 6lbr that will either scan the node info table or register to the route notification callback.
The route notification callback could be used as follow :

<pre>
static struct uip_ds6_notification route_notification;

void
route_notification_cb(int event,
                                uip_ipaddr_t * route,
                                uip_ipaddr_t * nexthop, int num_routes)
{
  if(event == UIP_DS6_NOTIFICATION_ROUTE_ADD) {
    ...
 } else if(event == UIP_DS6_NOTIFICATION_ROUTE_RM) {
  ....
  }
}
...
  uip_ds6_notification_add(&route_notification,  route_notification_cb);

</pre>

BaN-Co, 2016-06-01:
Thank you for your response!
I hadn't thought about the possibility to use plugins, it fully meets my needs.
But this solution doesn't remove the potential need to get the BR node info from the other network nodes. So I think we should leave this issue open until there will be the free time to add proper interface.

laurentderu, 2016-06-02:
Actually the list of nodes does not require explicit info from the node, the list is created using information gathered by RPL. The additional UDP client in the nodes is there only to retrieve more information from the node (the current parent, the prr, ...) If you deactivate it you will still have the list of nodes available.
Also, I'm planning in the future to retrieve some data directly from coap or tcp communication

steven3w, 2016-10-12:
Hi, Laurentderu, 

what time will the coap or tcp solution be done? I also have this requirement, I am doing a gateway  and a app web server beside the 6lbr in the BBB, it needs the node info. I could use udp like the cetic client, but the RPL should have the node info already there, why waste the through put bandwidth?