cetic/6lbr #186: implement nat64 and ping IPv4 mapped IPv6 address from node

Praveen-Kumar-K, 2016-05-30:
Hi,
As from the new version of 6lbr, NAT64 is included in the 6lbr itself. I have configured Beagle bone as 6LBR with router mode. i am able to see the 6lbr web page from my PC ( i.e bbbb::100 ), now i want to ping IPv4 mapped IPv6 address of google from my node ( i am using cc2538dk as node with ping6 application running in it), for that i need to use NAT64, where i need to use prefix which has to propagate in IPv4 mapped IPv6 address. can any one help me in this regard, the way to get ping IPv6 address of google from node.

Thanks,
Praveen Kumar K

laurentderu, 2016-06-01:
I guess you mean ping IPv4 address of google ? To do that, you have to use ip64_addr_4to6() on your node to convert the IPv4 address into a NAT'ed IPv6 address that you can use in your ping6 application

Praveen-Kumar-K, 2016-06-06:
Thanks laurentderu,
I have done as your comments, i know that there is prefix in code "::ffff", it is enabled when nat64 is enabled in config page. I am able to get mac learn aaaa::<mac of node> in ping6 application, but i am unable to ping 0::ffff:808:808 from Beagle bone black, may i know the reason, and how could i implement that, can any one help in this regard.

Thanks,
Praveen Kumar K

closed 2017-06-15 by laurentderu