cetic/6lbr #193: Issue with automatic connection

Mecabot, 2016-06-16:
Hello everyone,
I'm following this tutorial:
[https://github.com/cetic/6lbr/wiki/6LBR-Connection#automatic-connection]
when i use this command in Raspberry Pi :

`sudo sysctl -w net.ipv6.conf.eth0.accept_ra_rt_info_max_plen=64`

I face the following error:

`sysctl: cannot stat /proc/sys/net/ipv6/conf/eth0/accept_ra_rt_info_max_plen: No such file or directory`

Please help me on this problem.

My settings are ase follows:

1) interfaces:

source-directory /etc/network/interfaces.d
auto lo
iface lo inet loopback
auto br0
iface br0 inet dhcp
       bridge_ports eth0
       bridge_stp off
       up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
       post-up ip link set br0 address `ip link show eth0 | grep ether | awk '{print $2}'`

iface eth0 inet6 static
address bbbb::100
netmask 64
up ip -6 route add via default via bbbb::101/64 dev eth0
up sysctl -p 

2) 6lbr.conf:

MODE=ROUTER
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0 
CREATE_BRIDGE=0 
DEV_RADIO=/dev/ttyACM0 
BAUDRATE=115200
LOG_LEVEL=3

laurentderu, 2016-06-21:
As your eth0 is part of the bridge br0, you can not configure it, you must instead configure the br0 interface. Also, as you are using static address on br0 you don't need the accept_ra flag on br0 either.

Mecabot, 2016-06-28:
@laurentderu thanks for the help, your response was very helpful.

closed 2016-06-28 by Mecabot