cetic/6lbr #194: How to build and run 6lbr on CC2650DK+ENC28J60?

chenek, 2016-06-16:
I am able to build and run 6lbr on CC2538DK+ENC28J60 and I would like to know if it's possible to run 6lbr on CC2650DK+ENC28J60.

laurentderu, 2016-06-24:
It's possible to compile 6LBR for CC2650 since release V1.4.0, but this is still experimental. You also have to set up the ENC28J60 configuration

chenek, 2016-06-24:
I had set up the ENC28J60 with CC2538 and it worked. Is it the same for CC2650? By the way, what is the make command to build binary for CC2650DK+ENC28J60?

chenek, 2016-07-03:
@laurentderu Do you know how to build binary for CC2650DK-ENC28J60?

chenek, 2016-07-08:
I solve the problem. Since I use Cygwin, I have to install srec_cat into Cygwin/bin to make it build successfully.

closed 2016-07-08 by chenek