cetic/6lbr #195: connecting contiki websense to 6lbr

marcomilazzo, 2016-06-17:
Dear Sir
i'm using 6lbr and sky and it works fine
i would like to use the contiki  websense in the ipv6 example dir
6lbr does see the node but i doesen't connect to the route
i did add  
#define NETSTACK_CONF_MAC   csma_driver
#define NETSTACK_CONF_RDC   nullrdc_driver
#define RF_CHANNEL          26

but still nothing
[image]

What should i do ?
thank's
marco

laurentderu, 2016-06-21:
The 'Last seen' value is not troublesome if there is no traffic from/to your node. Can you ping the node from your PC ? if not, it means your route towards the WSN are not properly configured, see https://github.com/cetic/6lbr/wiki/6LBR-Connection

marcomilazzo, 2016-06-22:
Hi Laurent 
thank's a lot for the response
So there is nothing missing in the normal web-sense of contiky?
But the point is that it doesn't form the node tree ?
This is way i think there is something missing
I also tried the contiki udp client server and it workk
6lbr does see the node but no tree !
what am i missing?
thank's again
marco

laurentderu, 2016-06-24:
If you mean the tree in the web server page, it's normal. You need a small application on your node to periodically send to 6lbr the information about the node's current parent. 6LBR by itself can not infer this information.

marcomilazzo, 2016-06-26:
hi
can you tell me what application?
do you have an example?
thank's marco

marcomilazzo, 2016-06-27:
Also
I'm using the cc2530 . is there a way to compile 6lbr-demo for this chip?
Marco

laurentderu, 2016-06-29:
The CC2530 is quite a limited platform and we don't plan to support it

fzy-Line, 2017-03-14:
I am using CC2538dk, and I also meet the same question. Have you solved it yet? Thanks!

closed 2017-06-15 by laurentderu