cetic/6lbr #196: Issues with CC2650STK and local MQTT Broker.

Mecabot, 2016-06-28:
Hi guys:

I have tried to connect CC2650STK with local Mosquitto but i have not been succesful.

In the CC2650STK´s debug i received the following:

_Starting Contiki-3.x-2370-ge3d5b74
With DriverLib v0.44336
TI CC2650 SensorTag
 Net: sicslowpan
 MAC: CSMA
 RDC: ContikiMAC, Channel Check Interval: 16 ticks
 RF: Channel 25
CC26XX Web Demo Process
CC26XX Web Server
CC26XX CoAP Server
6LBR Client Process
CC26XX MQTT Client Process
CC26XX Net UART Process
MQTT - Registered successfully
Init
Registered. Connect attempt 1
MQTT - Got mqtt_do_connect_tcp_event!
Connecting (1)
Connecting (1)
Connecting (1)
............
MQTT - Disconnected by tcp event 2
APP - MQTT Disconnect. Reason 2
Disconnected
Disconnected. Attempt 2 in 1024 ticks
MQTT - Abort
Registered. Connect attempt 2
MQTT - Got mqtt_do_connect_tcp_event!
Connecting (2)
Connecting (2)
Connecting (2)
......._

My Mosquitto´s configuration is:

**bind_address 127.0.0.2
port 1885**

I have monitored the port with the command **sudo tcpdump -i eth0 -v -X port 1885** and I received the following:

_192.168.0.110.10262 > 127.0.0.2.1885: Flags [S], cksum 0x2bbf (correct), seq 400, win 128, options [mss 128], length 0_
	
	0x0000:  4500 002c 011a 0000 3f06 3a9a c0a8 006e  E..,....?.:....n	
	0x0010:  7f00 0002 2816 075d 0000 0190 0000 0000  ....(..]........	
	0x0020:  6002 0080 2bbf 0000 0204 0080            `...+.......

_00:06:31.061237 IP (tos 0x0, ttl 63, id 283, offset 0, flags [none], proto TCP (6), length 44)   
192.168.0.110.10262 > 127.0.0.2.1885: Flags [S], cksum 0x2bbf (correct), seq 400, win 128, options [mss 128], length 0_
	
	0x0000:  4500 002c 011b 0000 3f06 3a99 c0a8 006e  E..,....?.:....n	
	0x0010:  7f00 0002 2816 075d 0000 0190 0000 0000  ....(..]........	
	0x0020:  6002 0080 2bbf 0000 0204 0080            `...+.......

_00:06:55.586264 IP (tos 0x0, ttl 63, id 284, offset 0, flags [none], proto TCP (6), length 40)   
192.168.0.110.12001 > 127.0.0.2.1885: Flags [R.], cksum 0x376a (correct), seq 400, ack 0, win 128, length 0_
	
	0x0000:  4500 0028 011c 0000 3f06 3a9c c0a8 006e  E..(....?.:....n	
	0x0010:  7f00 0002 2ee1 075d 0000 0190 0000 0000  .......]........	
	0x0020:  5014 0080 376a 0000                      P...7j..

Should I add some additional settings to mosquitto´s configuration?

Best regards,
Mecabot.

laurentderu, 2016-06-29:
If you are using 6LBR in Raw Ethernet mode you can not access servers on the same host, if your host is on another machine on the lan, you can not use 127.0.0.2 as it is a host only address

chenek, 2016-07-03:
@Mecabot What HW do you run 6lbr? If it's BBB, Do you run wrapsix or6lbr NAT64 on BBB. You have to enable NAT64 to access MQTT server.

Mecabot, 2016-07-13:
Hi, sorry for the late.

@laurentderu  , i use other Raspberry Pi like a Mosquitto Broker, it is working but i don´t know why. My 6lbr.conf is the following:

_MODE=ROUTER
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0 
CREATE_BRIDGE=0 
DEV_RADIO=/dev/ttyACM0 
BAUDRATE=115200
LOG_LEVEL=3_ 

What should be my 6lbr.conf to have a Mosquittto Broker in the same Raspberry Pi? I only want to use a Raspberry pi.

@chenek my HW configuration is the following:

CC2650STK(CC26xx-web-demo) <-> CC2650STK(Slip-radio) <->  Raspberry Pi2 (Router with NAT64) <-> Raspberry Pi 2 (Mosquitto Broker) .

chenek, 2016-07-13:
Try the following values to replace those in your 6lbr.conf

MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

LOG_LEVEL=3

Mecabot, 2016-08-18:
Hi @chenek , i´ve tried your suggestion, but i´ve not been successful.

I got the following:

********************************************************************************
Thu 18 Aug 18:32:40 UTC 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins   
Contiki-develop-20160421-10-ge3648dc started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-08-18 18:32:40.396599: NOTICE: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20160421-10-ge3648dc)
2016-08-18 18:32:40.397328: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-08-18 18:32:40.401271: INFO: NVM: NVM Magic : 2009
2016-08-18 18:32:40.401305: INFO: NVM: NVM Version : 5
2016-08-18 18:32:40.401362: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-08-18 18:32:40.401489: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-08-18 18:32:40.401536: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-08-18 18:32:40.401555: INFO: MAC: Using 'csma' mac driver
2016-08-18 18:32:40.402588: INFO: SLIP: SLIP started on /dev/ttyACM0
2016-08-18 18:32:40.431383: INFO: SCMD: Started br-cmd process
2016-08-18 18:32:40.431532: INFO: BR-RDC: Reset SLIP Radio
2016-08-18 18:32:40.431618: INFO: BR-RDC: Fetching MAC address
2016-08-18 18:32:40.436267: ERROR: SLIP: Unknown command
2016-08-18 18:32:40.437781: ERROR: SLIP: Unknown command
2016-08-18 18:32:41.431210: INFO: BR-RDC: Fetching MAC address
2016-08-18 18:32:41.434389: INFO: BR-RDC: Got MAC: 00:12:4b:00:07:b5:3c:03
2016-08-18 18:32:42.431235: INFO: ETH: RAW Ethernet interface init
2016-08-18 18:32:42.482376: INFO: TAP: opened device /dev/eth0
2016-08-18 18:32:42.483170: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b8:27:eb:1c:37:64  
          inet addr:169.254.163.175  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:612 errors:0 dropped:0 overruns:0 frame:0
          TX packets:276 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:102548 (100.1 KiB)  TX bytes:39028 (38.1 KiB)

2016-08-18 18:32:42.611603: INFO: TAP: Eth MAC address : b8:27:eb:1c:37:64
2016-08-18 18:32:42.612450: INFO: ETH: Eth MAC address : b8:27:eb:1c:37:64
2016-08-18 18:32:42.613082: INFO: LLSEC: Using 'nullsec' llsec driver
2016-08-18 18:32:42.613484: INFO: 6LBR: Security layer initialized
2016-08-18 18:32:42.614172: ERROR: SLIP: Unknown command
2016-08-18 18:32:42.614720: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:7b5:3c03
2016-08-18 18:32:42.615362: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:7b5:3c03
2016-08-18 18:32:42.616171: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-08-18 18:32:42.616735: INFO: 6LBR: RA Daemon enabled
2016-08-18 18:32:42.617135: INFO: 6LBR: Checking addresses duplication
2016-08-18 18:32:53.619880: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:7b5:3c03
2016-08-18 18:32:53.620563: INFO: 6LBR: Starting IP64
2016-08-18 18:32:53.620961: INFO: 6LBR: IPv4 address : 192.168.0.115
2016-08-18 18:32:53.621483: INFO: 6LBR: Starting as RPL ROUTER
2016-08-18 18:32:53.623573: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-08-18 18:32:53.624158: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-08-18 18:32:53.624711: INFO: NODECFG: Node Config init
2016-08-18 18:32:53.625128: INFO: NODECFG: No node_config.conf file specified
2016-08-18 18:32:53.625553: INFO: HTTP: Starting webserver on port 80
2016-08-18 18:32:53.626472: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2016-08-18 18:32:53.627069: INFO: DNS: DNS proxy started
2016-08-18 18:32:53.627785: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-08-18 18:32:53.628302: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-08-18 18:32:53.628846: INFO: 6LBR: CETIC 6LBR Started
RD client started

Do you have any other suggestion?

Kind regards.

Mecabot, 2016-12-29:
Currently i use an external Broker MQTT because my Raspberry Pi(configured like Mosquitto Broker and Router with NAT64 at the same time) cannot get the MQTT Messagges with the following configuration:

MODE=ROUTER
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
CREATE_BRIDGE=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3

And, when i use the next configuration i lose BR0 and TAP0:

MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3

Any have a suggest for me?

Kind regards.

laurentderu, 2017-01-06:
The second configuration configures 6LBR in Raw Ethernet mode, in this mode the BR access directly the Ethernet device and therefore does not create a tap interface and a bridge to connect it the the Ethernet interface. That explain why the host can not communicate with the BR.

Normally the first configuration should work, unless the bridge interface is not properly configured on the host. In that case the external hosts can access 6LBR through the bridge, but the BR host can not use the br0 interface.

closed 2017-06-15 by laurentderu

rhorning, 2017-06-20:
Hey!
I believe I'm facing a similar issue.
I'm using CC2650 connected to a Linux Mint machine where 6lbr and mosquitto.rsmb are running.
My 6lbr setup is the following:

_MODE=ROUTER
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=enp7s0
ETH_JOIN_BRIDGE=1
CREATE_BRIDGE=1
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=4_

I'm using as node the CC2530. I'm able to send udp packages to my Linux machine(router), but it's not being possible to connect to the MQTT broker. Checking the network traces and even the console while MQTT is running, it's possible to see MQTT receiving a connection request from my node IP and sending the CONNACK, but my node doesn't catch it, it keeps trying to connect to the broker. **In other words, the broker sees the connection as established but my node doesn't because the CONNACK did not arrive at it.**

Analyzing a lit bit further, it seems that my node is not receiving packages at all, it can only send them. I did a simple python script (running on the router) that sends a udp package to the node. On my node firmware, I print on the terminal every package that I receive, but it's not printing anything when I run the python script (several times).. 

@laurentderu, do you have any clues? It seems to me that the issue is before the MQTT connection, because it is receiving any packages..

Thanks in advance!

rhorning, 2017-06-24:
I found out the problem. The IP that I was binding was not the same IP that my Linux was using to respond. I changed to the one that it was using and it worked.
I know that it's not related to this issue anymore, but how can I communicate always with the same IP on the Linux? I need to connect to the broker every time my nodes start. For that, I would need a static IP that reaches this sensor network. How can I do that? I've tried creating a static IP with the prefix bbbb and even binding with the border IP (bbbb::100), but it didn't work..

laurentderu, 2017-06-28:
It's a known problem on IPv6 network when privacy feature is enabled, the IP address of the host is periodically changed to avoid tracking, but if this host hosts servers, the client must be aware of the IP change.

The easiest workaround is to disable the privacy feature as it's not intended for server, you can disable it using the following sysctl parameter : net.ipv6.conf.eth0.use_tempaddr=0 (replace eth0 with your actual interface identifier).

rahultt007, 2017-07-07:
Hi,
 We are also facing the same issue like not able to get connect to mosquitto broker running on the same BBB where 6LBR is also running. 
Actually we are having another IOT SDK running on same BBB to connect it to one of our IOT HUB. So please let me know which interfaces i need to configure the configuration for 6LBR etc.

I tried with the following but not able to get the data in BBB MQTT Broker

6LBR Configuration is as follows,
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200
LOG_LEVEL=3

also tried
RAW_ETH=0
BRIDGE=1

In the interfaces we tried both,
auto br0
iface br0 inet dhcp
#auto eth0
#iface eth0 inet dhcp
and,
#auto br0
#iface br0 inet dhcp
auto eth0
iface eth0 inet dhcp

@laurentderu Please let me know which configuration is correct and let me know your thoughts.

Thanks