cetic/6lbr #197: 6lbr-unable to access the 'web' ibm cloude page- (sensortag2+bbb-edge router)

shosant, 2016-06-29:
Hi, I am using sensortag with debug devpack connected to the bbb to for the edge router.
All the configuration is done and I am able to access the 6lbr webpage. But when click the web option, It takes to sensor node address page, loads only the menu and loading fails./The connection has timed out/ error.
sometimes even the menu does'nt load and i get page expired or The connection has timed out error.
If the menu loads, and I quickly click the ibm quickstart link, i see my device id. If I click go, nothing happens.
It is so erratic that sometimes, form the menu (when loaded)if I click any of the config page, it loads and then in few seconds pages expires.
**Only this top menu loads, randomly and not always:**
[ Index ] | [ Device Config ] | [ Net-UART Config ] | [ MQTT/IBM Cloud Config ] | [ IBM Quickstart ]
**I am unable to ping6 the sensor node:**
debian@beaglebone:~$ ping6 fd00::212:4b00:7a8:d606
PING fd00::212:4b00:7a8:d606(fd00::212:4b00:7a8:d606) 56 data bytes
From bbbb::101 icmp_seq=1 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=2 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=3 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=4 Destination unreachable: Address unreachable
Sometimes i get the error, network not found if I ping6.
**6lbr.log:**
2016-06-29 1:47:16.339392: INFO: BR-RDC: Reset SLIP Radio
2016-06-29 1:47:16.339432: INFO: BR-RDC: Fetching MAC address
2016-06-29 1:47:16.342846: ERROR: SLIP: Unknown command
2016-06-29 1:47:16.344126: ERROR: SLIP: Unknown command
2016-06-29 1:47:17.339841: INFO: BR-RDC: Fetching MAC address
2016-06-29 1:47:17.342208: INFO: BR-RDC: Got MAC: 00:12:4b:00:07:bb:81:03
2016-06-29 1:47:18.339355: INFO: ETH: RAW Ethernet interface init
2016-06-29 1:47:18.352689: INFO: TAP: opened device /dev/eth0
2016-06-29 1:47:18.353390: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr 68:9e:19:99:8e:bc
          inet addr:192.168.1.7  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::6a9e:19ff:fe99:8ebc/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:245 errors:0 dropped:0 overruns:0 frame:0
          TX packets:186 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:25176 (24.5 KiB)  TX bytes:26779 (26.1 KiB)
          Interrupt:40
2016-06-29 1:47:18.539294: INFO: TAP: Eth MAC address : 68:9e:19:99:8e:bc
2016-06-29 1:47:18.540268: INFO: ETH: Eth MAC address : 68:9e:19:99:8e:bc
2016-06-29 1:47:18.540970: INFO: LLSEC: Using 'nullsec' llsec driver
2016-06-29 1:47:18.541406: INFO: 6LBR: Security layer initialized
2016-06-29 1:47:18.542884: ERROR: SLIP: Unknown command
2016-06-29 1:47:18.543613: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:7bb:8103
2016-06-29 1:47:18.544481: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:7bb:8103
2016-06-29 1:47:18.545159: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-06-29 1:47:18.545716: INFO: 6LBR: RA Daemon enabled
2016-06-29 1:47:18.546249: INFO: 6LBR: Checking addresses duplication
2016-06-29 1:47:29.549237: INFO: 6LBR: Configured as DODAG Root fd00::212:4b00:7bb:8103
2016-06-29 1:47:29.551293: INFO: 6LBR: Starting IP64
2016-06-29 1:47:29.552355: INFO: 6LBR: IPv4 address : 10.15.1.66
2016-06-29 1:47:29.553505: INFO: 6LBR: Starting as RPL ROUTER
2016-06-29 1:47:29.558011: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-06-29 1:47:29.559397: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-06-29 1:47:29.560766: INFO: NODECFG: Node Config init
2016-06-29 1:47:29.561882: INFO: NODECFG: No node_config.conf file specified
2016-06-29 1:47:29.562803: INFO: HTTP: Starting webserver on port 80
2016-06-29 1:47:29.564707: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2016-06-29 1:47:29.566122: INFO: DNS: DNS proxy started
2016-06-29 1:47:29.566985: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-06-29 1:47:29.568066: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-06-29 1:47:29.569255: INFO: 6LBR: CETIC 6LBR Started
RD client started
2016-06-29 1:50:36.888482: WARN: HTTP: File '/favicon.ico' not found, from bbbb::4da6:9c42:5d5e:610d
2016-06-29 1:50:36.900480: WARN: WEB: Buffer overflow
2016-06-29 1:50:36.933264: WARN: HTTP: File '/favicon.ico' not found, from bbbb::4da6:9c42:5d5e:610d
2016-06-29 1:50:36.944524: WARN: WEB: Buffer overflow
2016-06-29 1:52:44.868575: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
2016-06-29 1:52:45.202513: ERROR: BR-RDC: br-rdc: ack received for unknown packet (4)
2016-06-29 1:52:47.060360: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (5)
**Slipradio time out when accessing the cloud page**
[image]
Could somebody please help in troubleshooting the problem.

laurentderu, 2016-06-29:
What is the slip-radio you are using and what is the RDC used, ContikiMAC, NullRDC, ...? Do you always see slip errors when you try to access the node ? In the statistics page of 6LBR do you see lots of errors in the CSMA or RDC or SLIP layer ?

shosant, 2016-06-29:
I am using the sensortag2(cc2650) + debug dev pack connected to the BBB.
RDC used is NULLRDC.
and yes always it timesout when I try to open the web page to access the clould.
Rarely it loads only the menu, but not the contents of the index.html of the sensor node.
**Statistics:**
**IP**
Received packets : 502
Sent packets : 338
forwarded packets : 90
Dropped packets : 67
Wrong IP version or header length : 0
Dropped IP fragments : 0
Checksum errors : 0
Unsupported protocol : 0
**ICMP**
Received packets : 79
Sent packets : 19
Dropped packets : 43
Unsupported type : 43
Checksum errors : 0
**TCP**
Received packets : 279
Sent packets : 229
Dropped packets : 0
Checksum errors : 0
Ack errors : 0
Received RST : 0
retransmitted segments : 0
Dropped SYNs : 0
SYNs for closed ports : 0
**UDP**
Received packets : 0
Sent packets : 0
Dropped packets : 0
Checksum errors : 0
**NDP**
Received packets : 14
Sent packets : 28
Dropped packets : 0
**RPL**
Memory overflow : 0
Local repairs : 0
Global repairs : 0
Invalid packets : 0
DIO timer resets : 5
Parent switch : 0
Forward errors : 0
Loop errors : 0
Loop warnings : 0
Root repairs : 0
**CSMA**
Allocated packets : 11
Allocated neighbors : 1
Packet overflow : 0
Neighbor overflow : 0

Send packets : 121
Received packets : 45
Not acked packets : 93
Collisions : 1
Retransmissions : 76
Dropped packets : 18
Deferred packets : 0
**RDC**
Callback count : 1
Ack timeout : 0
Parse error : 0
**SLIP**
Messages sent : 126
Messages received : 170
Bytes sent : 11872
Bytes received : 4236

One more query:
Why is my parent node empty and the node tree is not shown?

shosant, 2016-06-29:
my ifconfig output:
debian@beaglebone:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr 68:9e:19:99:8e:bc
          inet addr:192.168.1.7  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::6a9e:19ff:fe99:8ebc/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2788 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1037 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:608317 (594.0 KiB)  TX bytes:215992 (210.9 KiB)
          Interrupt:40

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

usb0      Link encap:Ethernet  HWaddr b6:c2:72:73:f6:7a
          inet addr:192.168.7.2  Bcast:192.168.7.3  Mask:255.255.255.252
          inet6 addr: fe80::b4c2:72ff:fe73:f67a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:389 errors:0 dropped:0 overruns:0 frame:0
          TX packets:45 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:33636 (32.8 KiB)  TX bytes:11109 (10.8 KiB)
------------------------------------------------------------------------------------------------
6lbr.conf:
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200

LOG_LEVEL=3

shosant, 2016-06-29:
netstat:

debian@beaglebone:~$ netstat -rn6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
fe80::/64                      ::                         U    256 0     0 usb0
fe80::/64                      ::                         U    256 0     0 eth0
::/0                           ::                         !n   -1  1  3729 lo
::1/128                        ::                         Un   0   1     4 lo
fe80::6a9e:19ff:fe99:8ebc/128  ::                         Un   0   1     0 lo
fe80::b4c2:72ff:fe73:f67a/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 usb0
ff00::/8                       ::                         U    256 0     0 eth0
::/0                           ::                         !n   -1  1  3729 lo

laurentderu, 2016-06-30:
The number of retransmission and not acked packets in the CSMA layer is quite important, maybe your network is conflicting with a WiFi network, or the firmware in the CC2650 node is a bit old and still has the infinite retransmission bug which was solved recently in Contiki.
If you use another channel, does the number of lost packets diminish ?

birdmanhoo, 2016-07-01:
I've been troubleshooting a similar issue [here](https://github.com/contiki-os/contiki/issues/1594) and think it could be interference from an existing wireless network. Does anyone have a suggestion for documentation on what channels can be used and what slice of frequency spectrum they correspond to? Thx!

shosant, 2016-07-02:
Now I am able to see the sensor data in the web page. But , I am unable to get it in the IBM quick start page.
_**The device ID you have entered is valid, but we are waiting for your device to send us a message.

Get your device to send us some valid data - we'll visualize it for you!**_
It keeps waiting but no data is received.

I am using windows laptop,bbb running the 6lbr connected to it with sensortag as the mesh device. 
tcpdump, it says :
0 packets captured
0 packets received by filter
0 packets dropped by kernel

Can I run wireshark in windows , to check if mqtt data is sent? Which filter option should I use as I have not used wireshark before.
Is there any other way to connect to the cloud and to troubleshoot why I am unable to see the sesnordata in quickstart page?

chenek, 2016-07-14:
@shosant you can refer to http://sunmaysky.blogspot.tw/2015/08/how-to-setup-contiki-sniffer-for.html which show you how to setup 6LowPAN sniffer with CC2531 or CC2538.

balarsbp, 2016-10-06:
Hai everyone,
I followed the instruction in this link http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
I received the sensor data in IBM cloud, but the data is not continuously storing in cloud.I received the data continuously for 24 hours only then sensor freeze.
I flashed the cc26xx-web-demo.bin file in cc2650.

closed 2017-06-15 by laurentderu