cetic/6lbr #198: unable to see the sensortag data in IBM quickstart page

shosant, 2016-07-05:
Hi I am unable to get the cc2650 sensortag data in the ibm quickstart page, from 6lbr.
I am able to get the sensor data in the 'web' page, but not in ibm quickstart.
I am running NAT64- wrapsix. The MQTT settings are default, I have changed anything.
Could someone please, help me in troubleshooting the issue, what settings to look for .
I am struck at this point. Rest all is working fine.

chenek, 2016-07-14:
Try to disable wrapsix and use NAT64 on 6lbr which you can refer to https://github.com/cetic/6lbr/wiki/NAT64

shosant, 2016-07-14:
[image]
[image]
[image]

shosant, 2016-07-14:
Hi I have done all the settings. but it doesnt get connected the quick start
[image]

shosant, 2016-07-14:
I am using windows PC, and my netstat output on windows it 

IPv6 Route Table
===========================================================================
Active Routes:
 If Metric Network Destination      Gateway
  1    306 ::1/128                  On-link
  8    281 bbbb::/64                On-link
  2    266 bbbb::/64                On-link
  8    281 bbbb::101/128            On-link
  2    266 bbbb::3dbc:5bcf:da1e:6c20/128
                                    On-link
  8    281 bbbb::9911:1311:7b21:22cb/128
                                    On-link
  2    266 bbbb::b8a7:371e:2756:12ea/128
                                    On-link
  8    281 bbbb::d06a:f061:5e21:fcd1/128
                                    On-link
  8    281 fd00::/64                bbbb::100
  8    281 fd00::/64                fe80::6a9e:19ff:ff99:8ebc
  2    266 fd00::/64                fe80::6a9e:19ff:ff99:8ebc
  2    266 fe80::/64                On-link
  8    281 fe80::/64                On-link
 21    276 fe80::/64                On-link
 14    276 fe80::/64                On-link
 22    266 fe80::/64                On-link
 14    276 fe80::302b:aee0:131d:8b31/128
                                    On-link
  2    266 fe80::3dbc:5bcf:da1e:6c20/128
                                    On-link
 22    266 fe80::45ab:c1fe:2218:7446/128
                                    On-link
  8    281 fe80::d06a:f061:5e21:fcd1/128
                                    On-link
 21    276 fe80::ed60:4abd:882:f221/128
                                    On-link
  1    306 ff00::/8                 On-link
  2    266 ff00::/8                 On-link
  8    281 ff00::/8                 On-link
 21    276 ff00::/8                 On-link
 14    276 ff00::/8                 On-link
 22    266 ff00::/8                 On-link
===========================================================================
Persistent Routes:
 If Metric Network Destination      Gateway
  0 4294967295 fd00::/64                bbbb::100
===========================================================================

Is there any other network setting that I have missed doing?
I am unable ping the node from my bbb.
debian@beaglebone:~$ ping6 fd00::212:4b00:7a8:d606
connect: Network is unreachable

chenek, 2016-07-14:
I suggest you to setup sniffer to make sure your SensorTag sending MQTT messages to your 6lbr.

closed 2017-06-15 by laurentderu