cetic/6lbr #200: Low Down PRR value when I test 6lbr with CC1310

chenek, 2016-07-13:
I always get Low Down PRR value when I test 6lbr with CC1310. For Up PRR, I can get 100% but it's 4.3% on Down PRR. Is this normal? Doesn't higher value in Down PRR means better delivery rate?

pvergers, 2016-07-13:
I am very interested in this as well - I have seen and am concerned by the same thing (observed with CC1310 as well as CC2650).
I notice the value seems to drop once every 60 seconds, starting at 100%  -> 50% (1/2) -> 33% (1/3) -> 1/4 -> 1/5 ......

birdmanhoo, 2016-07-13:
I have this issue when using a CC2650 as well. I have some other symptoms mentioned [here](https://github.com/contiki-os/contiki/issues/1594) if it helps.

morphal, 2016-10-18:
The same problem with 6lbr + cc2650 sensortags.
And even further: after a while 6lbr moves node from OK status to NR (no route)

laurentderu, 2016-10-18:
The Down PRR of CC26xx and CC13xx displayed is wrong. Actually, the 6lbr-client provided in cc26xx-web-demo does not transmit the server sequence number and so 6lbr assumes that all the downstream packets are lost. The wrong value should not be displayed and the cc26xx-web-demo updated.

On the other hand, if your node status switches from OK to NR, there is an issue there to be investigated !

morphal, 2016-10-18:
@laurentderu , sorry, i didn't catch what you mean.
could you be a bit more specific on what is wrong and what should be changed in cc26xx-web-demo to make 6lbr PRR calculations correct?

Maybe you can also give me a suggestion how can I investiagate NR problem? I have cc2650 sensortag modules, they do not support step-by-step debugging, only 'printfs'. So, any suggestion where to start would be extremely helpful.

thanks

mdmobashir, 2016-12-05:
I observed the same. The Down PRR starts with 100, but gradually drops until close to 0. Any update/fix/explanation for this?

Thanks

laurentderu, 2016-12-05:
@mdmobashir see https://github.com/cetic/6lbr/issues/200#issuecomment-254498984 for the explanation (In short the downstream sequence index is not managed by CC26xx-web-demo and so the PRR is wrong).

The fix for 6LBR is available on the branch https://github.com/cetic/6lbr/tree/dev-1.4.x but I forgot to mention it here.

mdmobashir, 2016-12-05:
Thanks @laurentderu 
Eager to try out the new branch. 
Also, have you faced a scenario where the nodes are OK (last seen getting updated), but doesn't connect to the parent (and thus the metrics are all empty). I'm testing a very simple single hop topology with 3 nodes connecting to the sink.

Thanks again :)

laurentderu, 2016-12-05:
It could be many things, you may have a discrepency between the 6lowpan context set by 6lbr and the one hardcoded in your nodes, or DAO-Ack activated on the nodes but not on the BR, ...

mdmobashir, 2016-12-05:
Sure, thanks. I'd try to look more into the logs to figure out. One last question regarding the application that you suggest I should use with 6lbr. I'm just trying to collect data through the border router and using cc2650-web-demo currently.

PasqualeSalierno, 2017-01-13:
Hello! I’m a young hardware tester and now i am working on X.ip5 SmartMachine® a new board based on a powerful ARM® Cortex®-M3, very low active RF, MCU current and lowpower mode current consumption provide excellent battery lifetime. The X.ip5 is a 6LoWPAN device: low-power, IP-driven nodes and large mesh network support make this X.ip5 SmartMachine® the ideal solution for Internet of Things applications, thanks to its small size, the compact integrated antenna on board and battery supply.
More information on

http://telecontrolli.com/resource/downloads/datasheet/iot-system/x-ip-transceiver-family/82-x-ip5-datasheet/file.html

closed 2017-12-14 by laurentderu