cetic/6lbr #201: Unable to start 6LBR on Contiki

supriya20, 2016-07-16:
Hi,

I am trying to build an Edge Router with a SLIP radio using TI Sensortag CC2650 and 6LBR installed on my contiki running machine. 

I have flashed the sensortag as SLIP Radio and installed 6LBR on to my contiki machine following the steps mentioned in the link [https://github.com/cetic/6lbr/wiki/Other-Linux-Software-Configuration](url). When I start 6LBR I get a message as below on my terminal. As mentioned in the link this is supposed to create a file "6lbr.ip" in which the IP of the webserver will be stored and connectivity can be checked by pinging to that IP. But this file is not created for me and hence not able to check the connectivity and ensure the working of the edge router.

[image]

I didn't get any  error while installing 6lbr onto contiki nor when flashing sensortag as SLIP radio.
Don't know where I am going wrong and no clue to debug the issue.

It would be of great help if I can get guidance on this.

Thanks in Advance...!!!

rishi-electronut, 2017-01-12:
I am facing similar issues. However, I have setup 6RBL on my Raspberry Pi.
6lbr.log file is present in /var/log/ directory however, 6lbr.ip file is missing.
Guidance would be much appreciated.

laurentderu, 2017-01-12:
See #238

closed 2017-01-12 by laurentderu