cetic/6lbr #213: Communication issues with CC1310 nodes

rdeterre, 2016-08-18:
Hi,

I have some communication issues with 6lbr and CC1310 launchpads. A Raspberry Pi 3 runs 6lbr with a CC1310 launchpad as the slip radio and CC1310 launchpads as well for the node. The node runs the web demo code sample from contiki.

I can see the node in 6lbr's web interface, but I cannot open the node's web page. In addition, I can ping the node, but one every two pings fails with a 'Destination unreachable' error.

```
...
64 bytes from fd00::212:4b00:a27:ddf3: icmp_seq=545 ttl=63 time=106 ms
From bbbb::ede4:43fb:f5f9:f78d icmp_seq=546 Destination unreachable: No route
64 bytes from fd00::212:4b00:a27:ddf3: icmp_seq=546 ttl=63 time=106 ms
From bbbb::ede4:43fb:f5f9:f78d icmp_seq=547 Destination unreachable: No route
64 bytes from fd00::212:4b00:a27:ddf3: icmp_seq=547 ttl=63 time=106 ms
From bbbb::ede4:43fb:f5f9:f78d icmp_seq=548 Destination unreachable: No route
...
```

6lbr is compiled from the current master branch (commit 58c673660). I had tried the `dev-1.4.x` branch before without success. The `6lbr.conf` file is copied below. The RF channel is set to 13 in 6lbr's configuration, the slip radio's projet-conf.h file as well as the web demo project-conf.h. The prefix is set to `fd00`. Other than that, all configuration elements in 6lbr are left to their default values.

In the web demo firmware, I tried both the default driver and `nullrdc_driver` without any effect except a change in the ping duration.

```
MODE=ROUTER

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only
```

The Contiki version I used to compile the CC1310's firmware is that from `g-oikonomou:bugfix/cc13xx/contikimac-timings`, as this branch solves a communication problem with the CC1310 (https://github.com/contiki-os/contiki/issues/1747), and the web demo code sample does appear to respond to ping correctly when connected via a `rpl-border-router`.

Is there something I can try to get these nodes to work with 6lbr?

Thanks

ustbzm, 2016-10-12:
Hi, I have a custom board runs Debian OS. I install the 6lbr on it. 
I use this command to start the 6lbr: service 6lbr start.
Connect the CC1310 Launchpad to Debian via a USB-UART board. But the 6lbr can not run successfully.
The /var/log/6lbr.log:
********************************************************************************
Wed Oct 12 16:27:16 CST 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyUSB0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins   
Contiki-contiki-merge-20160622-29-g05adca5 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-10-12 16:27:16.345538: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-merge-20160622-29-g05adca5)
2016-10-12 16:27:16.346000: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-10-12 16:27:16.346221: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
2016-10-12 16:27:16.346454: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/lwm2m.so
2016-10-12 16:27:16.346640: INFO: PLUGIN: Initialising lwm2m-client
2016-10-12 16:27:16.346723: INFO: NVM: NVM Magic : 2009
2016-10-12 16:27:16.346738: INFO: NVM: NVM Version : 5
2016-10-12 16:27:16.346753: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-10-12 16:27:16.346777: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-10-12 16:27:16.346821: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-10-12 16:27:16.346834: INFO: MAC: Using 'csma' mac driver
2016-10-12 16:27:16.348937: INFO: SLIP: SLIP started on /dev/ttyUSB0
2016-10-12 16:27:16.359408: INFO: SCMD: Started br-cmd process
2016-10-12 16:27:16.359463: INFO: BR-RDC: Reset SLIP Radio
2016-10-12 16:27:16.359489: INFO: BR-RDC: Fetching MAC address
2016-10-12 16:27:16.362298: ERROR: SLIP: Unknown command
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:17.359467: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:18.359450: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:19.359432: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:20.359454: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:21.359459: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:22.359383: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:23.359423: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:24.359437: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:25.359422: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:26.359435: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:27.359432: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:28.359440: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:29.359432: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:30.359430: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:31.359428: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:32.359370: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:33.359419: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:34.359408: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:35.359437: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 214d (11)
2016-10-12 16:27:36.359407: INFO: BR-RDC: Fetching MAC address

ustbzm, 2016-10-12:
It seems that, the 6lbr can not read the MAC address from the CC1310 launchpad, and the launchpad is always output: EUnknown command 214d (11).

ustbzm, 2016-10-17:
I have solved the problem.
I convert the '\n' to '\r\n' in the "cc26xx_uart_write_byte" interface. So the 6LBR can not communicate with the slip-radio.

PasqualeSalierno, 2017-01-13:
Hello! I’m a young hardware tester and now i am working on X.ip5 SmartMachine® a new board based on a powerful ARM® Cortex®-M3, very low active RF, MCU current and lowpower mode current consumption provide excellent battery lifetime. The X.ip5 is a 6LoWPAN device: low-power, IP-driven nodes and large mesh network support make this X.ip5 SmartMachine® the ideal solution for Internet of Things applications, thanks to its small size, the compact integrated antenna on board and battery supply.
More information on

http://telecontrolli.com/resource/downloads/datasheet/iot-system/x-ip-transceiver-family/82-x-ip5-datasheet/file.html

closed 2017-06-15 by laurentderu