cetic/6lbr #215: How to disable leds in CC2650STK?

Mecabot, 2016-09-13:
Hi guys:

I want to disable the leds in CC2650STK specially the sequence _**red-green-red**_ that starts each time when i pressed the Power Button .

[image]

Best regards,
Mecabot.

msolters, 2016-09-13:
Hey @Mecabot this isn't really a 6LBR question, but a Contiki one.  But the place you're going to want to look at is basically the Contiki `main()` sequence as defined for the srf06-cc26xx platform:  https://github.com/cetic/6lbr/blob/develop/platform/srf06-cc26xx/contiki-main.c#L160

Notice there are a few more instances of the `fade` function littered through `main()`.  If you eliminate those and rebuild, you won't get the flashing.

Mecabot, 2016-09-20:
Hi @msolters, i tried deleting the 4 instances of the fade function that were inside the **main()** , but i got desestabilization in the system. 

Then i´ve reduced the time of leds_on inside of the **fade** function .

[image]

Thanks for your help.

Kind regards.

closed 2016-09-20 by Mecabot