cetic/6lbr #216: IPv6 address not got by beagle bone black

balarsbp, 2016-09-15:
I followed the instruction in this link http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
debian@beaglebone:~$ ifconfig 
eth0      Link encap:Ethernet  HWaddr b0:d5:cc:cd:8a:c6  
          inet addr:20.0.1.6  Bcast:20.0.1.255  Mask:255.255.255.0
          inet6 addr: fe80::b2d5:ccff:fecd:8ac6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST DYNAMIC  MTU:1500  Metric:1
          RX packets:12071 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1761 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1105607 (1.0 MiB)  TX bytes:332017 (324.2 KiB)
          Interrupt:175 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:248 errors:0 dropped:0 overruns:0 frame:0
          TX packets:248 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:20300 (19.8 KiB)  TX bytes:20300 (19.8 KiB)

usb0      Link encap:Ethernet  HWaddr b0:d5:cc:cd:8a:c8  
          inet addr:192.168.7.2  Bcast:192.168.7.3  Mask:255.255.255.252
          inet6 addr: fe80::b2d5:ccff:fecd:8ac8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:303 errors:0 dropped:0 overruns:0 frame:0
          TX packets:177 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:61047 (59.6 KiB)  TX bytes:50750 (49.5 KiB)

debian@beaglebone:~$ 
**all functions working properly.But web server page not display the  sensor data.**

chenek, 2016-09-21:
What sensor do you add to your 6lbr network?

balarsbp, 2016-09-22:
I am using cc2650 sensor tag.
my coap page like this
[image]

chenek, 2016-09-22:
In mid-March 2016, Contiki has changed its default context from aaaa::/64 to fd00::/64, it is mandatory to reflect this change in 6LBR or no proper communication with the node can happen. So, you have to set "Prefix" and "6LoPWAN context 0" both to "fd00::" on the 6lbr side.

closed 2016-11-07 by balarsbp