cetic/6lbr #222: 6lbr smart bridge mode not working on beagle bone black

balarsbp, 2016-10-26:
I followed the instruction in this link http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
I am using
Gateway=beagle bone black
Bluetooth dongle=cc2531
sensor tag=cc2650
I configured in 6lbr(Router mode) beagle bone black.It's working fine.
Issue:
How to configure 6lbr in smart bridge mode.
I followed bridge mode configuration in this link http://processors.wiki.ti.com/index.php/Contiki-6LOWPAN-BBB

My 6lbr.conf files as shown in below
MODE=SMART-BRIDGE
RAW_ETH=0
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3
Tap0 is creating but not providing the data to cloud.
Is there any option for 6lbr project working without pc?.

laurentderu, 2016-11-03:
Are you trying to use 6LBR with Jool or do you plan to bridge 6LBR with your Eth interface ?

balarsbp, 2016-11-04:
Thans for your reply.I am using Wrap six for converting IPV6 to IIPV4 and then already tried with jool but 6lbr setup not working.I followed in this link.http://processors.wiki.ti.com/index.php/Contiki-6LOWPAN-BBB

closed 2016-11-07 by balarsbp