cetic/6lbr #225: Contiki with CC1310

mario9876, 2016-11-24:
Hi I'm new on this job. I started working with Contiki, the CC1310 with SmartFR06EB and Raspberry. I'm trying to load the border router, but as soon as I type "sudo service 6lbr start" does not open my browser with the address bbbb :: 100. Can someone help me?

ronnie1233, 2016-11-29:
Open your log files i can identify where you are going wrong . Do you write nvm.dat in the channel 25 .

mario9876, 2016-11-29:
How?

mario9876, 2016-11-29:
Ok, you have some guide that can be followed for the installation of contiki?

ronnie1233, 2016-11-29:
The installation are avaliable on Github . 

- use cd /var/log 
after that use sudo nano 6lbr.log  . You will be able to generate log files

ronnie1233, 2016-11-29:
http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples

This is the best installation guide .

closed 2017-01-12 by laurentderu