cetic/6lbr #226: How to add two slip-radio on  6lbr

zx2194023, 2016-11-28:
I want to add two slip-radio on  6lbr , but i don't know how to do it .
I just know maybe go to rewrite "slip-dev.c"  

Can someone give me some idea? 
Thanks

laurentderu, 2016-11-29:
The work on multi-radio support will start soon :) It requires per radio configuration (as opposed to global configuration right now), an improved packet forwarding engine and a mac address learner.
I don't have yet a target date for the availability, but I hope before end of the year :)

zx2194023, 2016-11-30:
Thanks for the answer. I have some ideas. I want to  run simultaneously two 6lbr program  on the same machine. The first 6lbr connected to the first slip-radio,and then the second  6lbr  connected to the second slip-radio.
but now i just can run one 6lbr on machine ,how can i implement my idea?

laurentderu, 2016-12-07:
Actually you can run as many 6LBR as you want on a host. The only constraint is that each of them must have its own 6lbr.conf and nvm.dat files. Also, they have to be launched manually as the init.d script is launching only one instance.

So you need to at least change the following parameters in your 6lbr-xxx.conf files :
DEV_RADIO=...
DEV_TAP=...
NVM=...
LOG_6LBR=...

PasqualeSalierno, 2017-01-13:
Hello! I’m a young hardware tester and now i am working on X.ip5 SmartMachine® a new board based on a powerful ARM® Cortex®-M3, very low active RF, MCU current and lowpower mode current consumption provide excellent battery lifetime. The X.ip5 is a 6LoWPAN device: low-power, IP-driven nodes and large mesh network support make this X.ip5 SmartMachine® the ideal solution for Internet of Things applications, thanks to its small size, the compact integrated antenna on board and battery supply.
More information on

http://telecontrolli.com/resource/downloads/datasheet/iot-system/x-ip-transceiver-family/82-x-ip5-datasheet/file.html

closed 2017-11-09 by laurentderu