cetic/6lbr #228: br0 tap0 not getting created on raspberry pi

mdmobashir, 2016-12-09:
Hi ,
Tried to follow the tutorial on "http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples" to create a 6lbr on a raspberry pi with cc1350 Launchpad configured as a slip radio. However, br0 and tap0 are not getting created. The log mentions "ERROR: SLIP: Unknown command" just before the creation of tap interface.

Is there something that I missed which is causing this error or any way to fix it?

Best regards,
MnM

Following is the detailed 6lbr.log

```
Fri 9 Dec 07:16:39 UTC 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
Contiki-contiki-merge-20160622-77-gdc08685 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2016-12-09 7:16:39.108231: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-merge-20160622-77-gdc08685)
2016-12-09 7:16:39.109445: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-12-09 7:16:39.112929: INFO: NVM: NVM Magic : 2009
2016-12-09 7:16:39.112994: INFO: NVM: NVM Version : 5
2016-12-09 7:16:39.113021: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-12-09 7:16:39.113156: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-12-09 7:16:39.113240: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2016-12-09 7:16:39.113274: INFO: MAC: Using 'csma' mac driver
2016-12-09 7:16:39.114216: INFO: SLIP: SLIP started on /dev/ttyACM0
2016-12-09 7:16:39.125621: INFO: SCMD: Started br-cmd process
2016-12-09 7:16:39.125799: INFO: BR-RDC: Reset SLIP Radio
2016-12-09 7:16:39.125928: INFO: BR-RDC: Fetching MAC address
2016-12-09 7:16:39.130396: ERROR: SLIP: Unknown command
2016-12-09 7:16:39.131770: INFO: BR-RDC: Got MAC: 00:12:4b:00:0b:ca:28:83
2016-12-09 7:16:40.125730: INFO: ETH: RAW Ethernet interface init
2016-12-09 7:16:40.134213: INFO: TAP: opened device /dev/eth0
2016-12-09 7:16:40.135555: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b8:27:eb:1f:71:b2
          inet addr:10.2.2.3  Bcast:10.2.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:189 errors:0 dropped:0 overruns:0 frame:0
          TX packets:201 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:23670 (23.1 KiB)  TX bytes:29492 (28.8 KiB)

2016-12-09 7:16:40.386919: INFO: TAP: Eth MAC address : b8:27:eb:1f:71:b2
2016-12-09 7:16:40.388445: INFO: ETH: Eth MAC address : b8:27:eb:1f:71:b2
2016-12-09 7:16:40.389778: INFO: LLSEC: Using 802.15.4 framer
2016-12-09 7:16:40.390600: INFO: LLSEC: Using 'nullsec' llsec driver
2016-12-09 7:16:40.391920: ERROR: SLIP: Unknown command
2016-12-09 7:16:40.393116: INFO: HTTP: Starting webserver on port 80
2016-12-09 7:16:40.394828: INFO: NODECFG: Node Config init
2016-12-09 7:16:40.395711: INFO: NODECFG: No node_config.conf file specified
2016-12-09 7:16:40.396773: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:bca:2883
2016-12-09 7:16:40.398116: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:bca:2883
2016-12-09 7:16:40.399413: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-12-09 7:16:40.400608: INFO: 6LBR: RA Daemon enabled
2016-12-09 7:16:40.401538: INFO: 6LBR: Checking addresses duplication
2016-12-09 7:16:51.404351: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:bca:2883
2016-12-09 7:16:51.405980: INFO: 6LBR: Starting as RPL ROUTER
2016-12-09 7:16:51.410176: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-12-09 7:16:51.411418: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-12-09 7:16:51.412551: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2016-12-09 7:16:51.413709: INFO: DNS: DNS proxy started
2016-12-09 7:16:51.414962: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2016-12-09 7:16:51.415971: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2016-12-09 7:16:51.417228: INFO: 6LBR: CETIC 6LBR Started
RD client started
2016-12-09 7:22:33.068891: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
2016-12-09 7:22:33.136712: ERROR: BR-RDC: br-rdc: ack received for unknown packet (2)
```

ronnie1233, 2016-12-15:
Change it in your 6lbr.conf  file . Also create tap0  in the 6lbr.conf file .

ronnie1233, 2016-12-15:
if if you have already created tap0 in conf file  change your network configuration i had the same issue when keeping eth0 as static  due to firewall issue .

laurentderu, 2016-12-15:
The setup described in "http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples" set 6LBR in Raw Ethernet mode, not Bridging mode, hence the tap0 interface is not created, see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

closed 2017-01-12 by laurentderu