cetic/6lbr #229: Unable to access the Internet from the WSN node through IPv4 internet connection.

vad32m, 2016-12-09:
Hi, everyone. Due to lack of experience with computer networking I ask for your help. For past few days I have been trying to setup 6lowpan network and connect it to the Internet, for my wireless nodes to be able to act as HTTP clients. Desired topology is pretty simple:
[image]
I downloaded InstantContiki VM and installed 6lbr to this VM. As my setup is experimental, I used COOJA to simulate WSN:
[image]
After 6lbr service is launched my ifconfig output looks like this:
[image]
One of my nodes acts like slip-radio while others are flashed with udp-example firmware and modified HTTP example firmware. In my understanding, after 6lbr is properly configured and launched, my HTTP-example node should be able  to resolve hostname "www.contiki-os.org", but even after I launch 6lbr daemon, it cant resolve that hostname. 
[image]
As you may see from the picture, udp client examples work OK, but HTTP-example node fails with resolving hostname. All nodes are visible in 6lbr web interface.
[image]
Another task I can't handle is configuring IP64, which is essential for internet connectivity. Can anyone give me advice on how to do that. My IP64 config looks like this:
[image]
But from web interface I see, that there are some problems with IP64 connectivity:
[image]
[image]
My log and config files:
[6lbrFiles.zip](https://github.com/cetic/6lbr/files/642873/6lbrFiles.zip)
Will appreciate any help or advice.

ronnie1233, 2016-12-15:
Install Wrapsix to see your devices in IPV4 network . Recommend to see both Wrapsix since NAT64  service is nedded for this .

laurentderu, 2016-12-15:
In simulation, the tap0 interface is not connected to any bridge at all, so you won't have internet connectivity by default. You have to see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration to enable the bridging mode and connect tap0 to eth0.

Another way is to use wrapsix as suggested by @ronnie1233 but I'm not familiar with it.

vad32m, 2016-12-15:
@laurentderu thx for advice. I was trying to configure ebtables and iptables, but havent succeed with that. After I modified config file for using bridge between tap0 and eth0 interfaces, 6lbr succesfully retreived ipv4 adress over DHCP. 
RAW_ETH=0
BRIDGE=1

closed 2016-12-15 by vad32m