cetic/6lbr #231: TI CC2650 SensorTag Timeout

larskholte, 2016-12-17:
I am running into what appears to be some kind of timeout issue connecting TI CC2650 SensorTags to 6lbr in router mode on a BeagleBone Black. I compiled 6lbr from source on the BBB as directed here: https://github.com/cetic/6lbr/wiki/Other-Linux-Software-Configuration . The SensorTags are running the cc26xx-web-demo application (details below). I am using a TI CC2531EMK flashed with cc2531-slip-radio_contikimac from this page: http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples .

What Works
---
I'm able to connect to [bbbb::100] and view the webpage from a Windows laptop connected by Ethernet to the BBB. The SensorTags will show up under the "Sensors" tab in the web interface. I can ping their IPv6 addresses from CMD and retrieve a webpage from each sensor individually. That's all working great.

What Doesn't
---
However, 10 minutes after powerup (I've timed it, it seems to be predictably between 600s and 660s) my pings will start to timeout, and I won't be able to get a webpage from the sensors. Oddly, the "Last seen" metric in the web interface will continue to reset, and will never get above 60s. This indicates that the SensorTags are still communicating with the SLIP radio, and checking in every minute, but I'm not able to get through 6lbr to ping them.

I also connected the BBB via Ethernet to a Linux host and verified that I am able to ping the sensors for about 10 minutes. Then things stop working. So it isn't a Windows-host-specific problem.

Once the nodes are unresponsive, I can do two things to fix the problem:
1.	Pressing the power button on the SensorTag twice will cause it to reset. Once it resets, the problem is gone for another 10 minutes or so.
2. `service 6lbr restart` on the BBB fixes the problem.

BBB Setup
---

```
$ uname -a
Linux beaglebone 3.8.13-bone47 #1 SMP Fri Apr 11 01:36:09 UTC 2014 armv7l GNU/Linux
```
```
$ cat /etc/6lbr/6lbr.conf
#This file contains a default configuration for BeagleBone platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
CREATE_BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=3 #INFO and above only
```
```
$ /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd

WSN network prefix : fd00::
WSN network prefix length : 64
WSN IP address : fd00::100
WSN accept RA : True
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : fd00::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 247
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

RPL DAO ACK : True
RPL DAO ACK local repair : False
DIO triggers route refresh : False
MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay enabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : False

NAT 64 Enabled : False
NAT 64 DHCP Enabled : False
NAT 64 static port mapping Enabled : False
NAT 64 RFC-6052 prefix : False
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80
```
SensorTag Setup
---
The SensorTags are flashed with the cc26xx-web-demo built from source with `BOARD=sensortag/cc2650` .

I disabled deep sleep mode in an attempt to fix the problem. It didn't work. From `cpu/cc26xx-cc13xx/lpm.h`:
```
//#define LPM_MODE_MAX_SUPPORTED LPM_MODE_DEEP_SLEEP
#define LPM_MODE_MAX_SUPPORTED LPM_MODE_SLEEP
```

I disabled features of the demo that I don't need. From `examples/cc26xx/cc26xx-web-demo/project-conf.h`:
```
/* Enable/Disable Components of this Demo */
#define CC26XX_WEB_DEMO_CONF_MQTT_CLIENT      0
#define CC26XX_WEB_DEMO_CONF_6LBR_CLIENT      1
#define CC26XX_WEB_DEMO_CONF_COAP_SERVER      0
#define CC26XX_WEB_DEMO_CONF_NET_UART         0
```

Hopefully this isn't something obvious that I overloooked. I couldn't find others with this exact problem. I did find one person who had issues with TI SensorTags becoming unresponsive overnight: http://contiki-developers.narkive.com/ghD3ywtV/contiki-cc26xx-nodes-dropping-from-mesh But his problem magically resolved itself,and the details are different.

I appreciate your help.

EpoxyD, 2016-12-20:
I have the same issue, with the added bonus that the sensor tags don't show a parent node.

laurentderu, 2016-12-20:
It's a known problem of the CC26xx platform, see the bug report in Contiki issue tracker https://github.com/contiki-os/contiki/issues/1858. 

Now the root cause is a bit obscure, as the delay (600sec) is similar to the periodic NUD mechanism f IPv6 NDP, on the other hand someone reported that the issue disappeared if he deactivated some sensor readings...

OSPro, 2017-01-10:
Same issue. I reference this link (https://github.com/contiki-os/contiki/issues/1858) and try all cases, but seem can not solve this issue. Does Debian os on BBB is issue? any one tried an ubuntu pc instead of BBB with debian?
Thanks.

laurentderu, 2017-01-12:
@OSPro I believe the problem is inside the Sensortag, not 6LBR itself (although the problem could be triggered by 6LBR...)

dbjorklu, 2017-01-12:
Seems to me there are more than one problem, or maybe the same problem but different symptoms. I seem to have the problem described by larskholte above. I cannot reach nodes through 6lbr after 10 minutes. In contiki-os/contiki#1858 people are seing "hangs" or resets, which I do not.
After the 10 mins I will continue to receive push messages from the nodes (CoAP observe notifications), so the nodes are running just fine, but 6lbr has lost the routes to them.
Resetting a node or resetting 6lbr solves the problem (the fact that resetting 6lbr solves the problem already shows that the nodes do not hang)

PasqualeSalierno, 2017-01-13:
Hello! I’m a young hardware tester and now i am working on X.ip5 SmartMachine® a new board based on a powerful ARM® Cortex®-M3, very low active RF, MCU current and lowpower mode current consumption provide excellent battery lifetime. The X.ip5 is a 6LoWPAN device: low-power, IP-driven nodes and large mesh network support make this X.ip5 SmartMachine® the ideal solution for Internet of Things applications, thanks to its small size, the compact integrated antenna on board and battery supply.
More information on

http://telecontrolli.com/resource/downloads/datasheet/iot-system/x-ip-transceiver-family/82-x-ip5-datasheet/file.html

laurentderu, 2017-01-20:
Update from the Contiki thread: setting UIP_CONF_SEND_NA to 1 or merging https://github.com/contiki-os/contiki/issues/2073 seems to solve the 10min pseudo-hang.

closed 2017-06-15 by laurentderu