cetic/6lbr #232: 6lbr on sensortag+Devpack and ENC28J60

tarakanov, 2016-12-23:
Hi!
Does anyone success with running 6lbr on Sensortag (cc26xx) with ENC28J60? 
I've try to move from srf06 board to Sensortag but Contiki doesn't run at all (no UART, no blinking leds). 

Contiki examples run with no errors (in /examples/cc26xx from this repo). LEDs are blinking, UART is working (via devpack), sensors are working also.

laurentderu, 2017-01-06:
Have you tried with the latest develop snapshot, there was a bug that caused the CC26xx firmware to crash on startup. Now it should work on both srf06 and Sensortag, although I tested only the srf06.

tarakanov, 2017-01-15:
Thank you for suggestion, @laurentderu. I use latest branch - develop. Also I try to mimic srf06 board and change pins in `board.h` and `examples/6lbr/platform/srf06-cc26xx/enc28j60-arch-gpio.c`. And then compile it as CC2650DK board.
There is some result - it compiles, UART is working. And even leds are blinking at boot. But it happens only first time after flashing firmware. Second time - no blinking, only first led (red) are on. It seems to hang. 
How can I debug this issue? Can I use Code Composer Studio with 6lbr-repo?
By the way, Sensortag has external flash. If we use it, can we fix memory constrains of CC2650?

laurentderu, 2017-01-16:
I never tried to compiles 6lbr with Composer Studio, however you could use the generated .elf file and use it to do remote debugging of the CC2650.

Do you see any output on the UART when you start up the CC2650 the first time ?

I see two possible issues : The selected pin conflict with another hw module and so either the ENC driver or the driver of the latter block hangs while accessing those pins. Another possibility is that the NVM code of the CC2650 port for 6LBR assumes that you have 128KB of flash and that the page located before the last page is available for storing data.
The NVM code can be configured to use the external flash, you have to use cc26xx-ext-nvm.c instead of cc26xx-int-nvm.c. Be warned that it has not be tested though.

tarakanov, 2017-01-25:
I successfully add external flash. And it is solve problem with hanging. Sensortag boots up every time after reset. And I have issue with SPI pin assigment. Now lightweight 6lbr is working. I am going to make a pull request for Sensortag platform if you don't mind

closed 2017-01-25 by tarakanov

laurentderu, 2017-01-26:
Thanks for the feedback and contributions are always welcome :)

mcuxmx, 2017-04-07:
I have the same problem when i use cc1310 on 6lbr.I found this problem related to the configuration of 

#define CETIC_6LBR_NVM_SIZE 2048
// We use the penultimate flash page as our nvm
#define CETIC_6LBR_NVM_ADDRESS (0x00020000 - (2*CETIC_6LBR_NVM_SIZE))

in file cc26xx-int-nvm.c

when (2*CETIC_6LBR_NVM_SIZE)  equal 4096 this problem occurred.

so i think it's a bug of cc13xx(cc26xx) library

laurentderu, 2017-04-11:
@mcuxmx  I think when setting the NVM on the internal flash it conflicts with latest Ccxxx-ware code or data, I should have a closer look.
The workaround right now are either move the NVM earlier in the memory, as you suggested in another issue, or use an external flash or disable NVM.