cetic/6lbr #235: Is there a maximum number of connections in the 6Lowpan Border Router?

Mecabot, 2016-12-29:
Hi everybody:

In my project i configured the next architecture:

CC2650STK(CC26xx-web-demo) <-> CC2650STK(Slip-radio) <-> Raspberry Pi2 (Router with IP64) <-> External Server on windows(Mosquitto Broker) .

My IP64´s configuration is:

laurentderu, 2017-01-06:
The maximum of concurrent NAT64 active connection is defined in ip64-addrmap.c, and by default it's 32.
The maximum number of nodes is actually limited by the number of routes, which is defined in the platform specific configuration file of 6lbr. For native, it's 256.

Mecabot, 2017-01-10:
Hi @laurentderu 

I´m going to change the default number of concurrent NAT64 active connection to 256, but what i need to reflash? Maybe the firmware on the CC2650STK(Slip-radio) or the 6lbr´s software on the RPI. 

Can you help me with my other answer #236 ? The TI E2E Community suggested to me ask in this forum ([TI E2E ](https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/p/565784/2074802#2074802))

I really appreciate any help you can provide.

laurentderu, 2017-01-12:
To change the number of NAT64 sessions, you need to recompile 6LBR (on the RPi). Currently it's not possible to change the value using a configuration parameters due to how the parameter is used in Contiki...
Tell me if it works, if so I will update the default value in 6LBR

PasqualeSalierno, 2017-01-13:
Hello! I’m a young hardware tester and now i am working on X.ip5 SmartMachine® a new board based on a powerful ARM® Cortex®-M3, very low active RF, MCU current and lowpower mode current consumption provide excellent battery lifetime. The X.ip5 is a 6LoWPAN device: low-power, IP-driven nodes and large mesh network support make this X.ip5 SmartMachine® the ideal solution for Internet of Things applications, thanks to its small size, the compact integrated antenna on board and battery supply.
More information on

http://telecontrolli.com/resource/downloads/datasheet/iot-system/x-ip-transceiver-family/82-x-ip5-datasheet/file.html

closed 2017-06-15 by laurentderu