cetic/6lbr #236: How to change the time of republishing in a MQTT message?

Mecabot, 2016-12-29:
Hi everybody:

I am sending MQTT messages with several Sensors (CC2650STK). When i debugged a Sensor (CC2650STK) in sometimes  i got the following:

_Publishing
MQTT - Got mqtt_do_publish_mqtt_event!
MQTT - Sending publish message! topic iot-2/evt/status/fmt/json topic_length 25
MQTT - Buffer space is 512 
MQTT - Encoding length 464
MQTT - Encode len digit '208' length '3'
MQTT - Encode len digit '3' length '0'
MQTT - remaining_length_bytes 2
MQTT - (write_byte) buff_size: 512 write: '30'
MQTT - (write_bytes) len: 2 write_pos: 2
MQTT - (write_byte) buff_size: 509 write: '00'
MQTT - (write_byte) buff_size: 508 write: '19'
MQTT - (write_bytes) len: 25 write_pos: 25
MQTT - (write_bytes) len: 437 write_pos: 437
MQTT - (send_out_buffer) Space used in buffer: 467
MQTT - Publish Enqueued
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
............    **THE REPUBLISHING TIME IS AROUND 120 SECONDS**
Publishing... (MQTT state=8, q=0)  
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
Publishing... (MQTT state=8, q=0)
MQTT - Disconnected by tcp event 2   
APP - MQTT Disconnect. Reason 2
Disconnected
Disconnected. Attempt 1 in 512 ticks
MQTT - Abort
Registered. Connect attempt 1
MQTT - Got mqtt_do_connect_tcp_event!
Connecting (1)  
Connecting (1)_

I would like to reduce the Republishing Time.

What should i do in order to reduce the Republishing Time?

Kind Regards.

laurentderu, 2017-01-12:
I'm sorry, this question has nothing to do with 6LBR and I don't know the answer, please use the Contiki ML.

closed 2017-01-12 by laurentderu

Mecabot, 2017-01-12:
Ohhh I´m sorry , i was wrong to publish my question here.

Thank you very much for your help @laurentderu