cetic/6lbr #237: 6lbr doesn't work on a new Raspbian

janeksz, 2017-01-06:
Hi,
After creating a new RPi image:
Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linx 
and installing 6lbr
I am not able to access 6lbr webpage
Everything works fine if I use my old uSD card with the same hardware (RPi2mB, cc2650 Launchpad as SLIP radio, CC2650 sensortags with Contiki as motes)
How to debug it?

root@raspberrypi:~# service 6lbr restart
root@raspberrypi:~# service 6lbr status
● 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr)
   Active: active (running) since Fri 2017-01-06 01:51:59 UTC; 8s ago
  Process: 10617 ExecStop=/etc/init.d/6lbr stop (code=exited, status=0/SUCCESS)
  Process: 10633 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/6lbr.service
           ├─10639 /bin/sh /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
           ├─10643 /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -...
           ├─10644 /bin/sh /usr/lib/6lbr/6lbr-watchdog 10643 60 3 /var/log/6l...
           └─10704 sleep 1

Jan 06 01:51:59 raspberrypi 6lbr[10633]: Starting 6LoWPAN Border Router:.
Jan 06 01:51:59 raspberrypi systemd[1]: Started LSB: 6LoWPAN Border Router.
root@raspberrypi:~#

All my hardware and 6lbr settings are identical to previous (working) version

my /var/log/6lbr.log is being continuously updated, but i am unable to figure out what is wrong and why I can not see [bbbb::100] in my browser

Friday 6 January 02:03:22 UTC 2017 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t wlan0 -f -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip -n /etc/6lbr/node_config.conf -m /usr/lib/6lbr/plugins   
Contiki-develop-20160421-6-gb1df3d1 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2017-01-06 2:03:23.282507: NOTICE: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20160421-6-gb1df3d1)
2017-01-06 2:03:23.283637: INFO: ETH: 6LBR watchdog started (interval: 60)
2017-01-06 2:03:23.287660: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
2017-01-06 2:03:23.290524: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/lwm2m.so
2017-01-06 2:03:23.327259: INFO: PLUGIN: Initialising lwm2m-client
2017-01-06 2:03:23.368001: INFO: NVM: NVM Magic : 2009
2017-01-06 2:03:23.368046: INFO: NVM: NVM Version : 5
2017-01-06 2:03:23.368106: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
wlan0     Link encap:Ethernet  HWaddr 00:c1:41:06:0b:97  
          inet addr:172.19.124.137  Bcast:172.19.125.255  Mask:255.255.254.0
          inet6 addr: fe80::7863:b74b:b569:9336/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:25 errors:0 dropped:0 overruns:0 frame:0
          TX packets:45 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6623 (6.4 KiB)  TX bytes:6659 (6.5 KiB)

2017-01-06 2:03:24.728144: PACKET: TAP: read: 165
2017-01-06 2:03:24.729030: PACKET: TAP: read: 145
2017-01-06 2:03:25.043883: PACKET: PF: bridge_output: Sending packet to Broadcast
2017-01-06 2:03:25.044488: PACKET: PF: wireless_output: sending packet
2017-01-06 2:03:25.045212: PACKET: PF: eth_output: broadcast
2017-01-06 2:03:25.045749: PACKET: PF: eth_output: Sending packet to Ethernet
2017-01-06 2:03:25.046380: PACKET: ETH: write: 78
2017-01-06 2:03:25.046922: DUMP: ETH: 
         3333ffb5 ee8500c1 41060b97 86dd6000 
         00000018 3aff0000 00000000 00000000 
         00000000 0000ff02 00000000 00000000 
         0001ffb5 ee858700 9d750000 0000aaaa 
         00000000 00000212 4b0007b5 ee85

janeksz, 2017-01-06:
through experimentation found the problem, but I am unable to solve it:
when I change in my /etc/6lbr/6lbr.conf
from 
DEV_ETH=wlan0
to
DEV_ETH=eth0
it works,
but I need to work with wifi (wlan0) like in a previous system
What do I need to change in my RPi system to make it work?

janeksz, 2017-01-08:
following this link https://github.com/cetic/6lbr/issues/163 and this https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration I am trying to setup "the routed interface mode"
my config is:

root@raspberrypi:~# cat /etc/6lbr/6lbr.conf
#This file contains a default configuration for Raspberry PI platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=0
#DEV_BRIDGE=br0
#CREATE_BRIDGE=0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

NODE_CONFIG=/etc/6lbr/node_config.conf

LOG_LEVEL=3 #INFO and above only
root@raspberrypi:~#

and I added:
net.ipv4.ip_forward = 1
net.ipv6.conf.all.forwarding=1
into my /etc/sysctl.d

but still no luck (don't see 6lbr webpage [bbbb::100]

florian-popup, 2017-01-09:
Well, I just made a set up with a pi and rasbian.
Did you start the 6lbr service on the server ? sudo service 6lbr start

If it is not working, here is my conf

Interestingly, my ethernet connection is not named eth0 but enxb827eb15ec3a
1st do a ifconfig and check the name of your ethernet connection and correct the file with it
2nd, update your /etc/network/interfaces file

**************************************************************
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
auto enxb827eb15ec3a
iface enxb827eb15ec3a inet dhcp
# Example to keep MAC address between reboots
#hwaddress ether DE:AD:BE:EF:CA:FE

# The secondary network interface
#auto eth1
#iface eth1 inet dhcp

auto br0
iface br0 inet dhcp
    bridge_ports enxb827eb15ec3a
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show enxb827eb15ec3a | grep ether $

# WiFi Example
auto wlan0
iface wlan0 inet dhcp
#    wpa-ssid "essid"
#    wpa-psk  "password"
******************************************************************************************
make a sudo service network-manager stop & sudo service network-manager start
in order to take the config into consideration

and then, configure the 6lbr router in /etc/6lbr/6lbr.conf
*****************************************************************************
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=enxb827eb15ec3a
RAW_ETH_FCS=1

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=3 #INFO and above only
***************************************************************************************
YOU HAVE TO SET DEV_RADIO=/dev/ttyACM0 accordingly to your slip device listed in ls /dev

DO NOT FORGET TO START THE 6LBR PROCESS AT EACH STARTUP
sudo service 6lbr start
Then, to enable internet, sudo sysctl -w net.ipv6.conf.enxb827eb15ec3a.accept_ra=1

janeksz, 2017-01-09:
@florian-popup ,
Thanks, appreciate it, however it doesn't solve my problem. 
My system is running well if I use ethernet interface.
**I have a problem with wifi.** 
As your 6lbr.conf suggest, **you don't use wifi**

florian-popup, 2017-01-10:
@janeksz 
Oh, yeah, sorry.
I just made a setup bridging on the wifi interface and I can access the [bbbb:100] from it.
I m not an expert, so there is things that I don't master but here is how I made it

In the /etc/network/interfaces, I used 
auto wlan0
iface wlan0 inet dhcp 
    wpa-ssid "YOUR_SSID"
    wpa_psk "YOUR_PASSWORD"

When you modify  /etc/network/interfaces, the network manager won't manage the devices anymore, so you have to edit /etc/NetworkManager/NetworkManager.conf
and change the line managed=false for managed=true

Change the interfaces and 6lbr.conf files to bridge over wlan0 and restart your computer.

You should be connected to the wifi directly at startup.

I suppose that if ipv6 is disabled on your router, issues could happened due to the wifi interface being configured in dhcp mode.
I have to test the stability of the bridge because all the interfaces are managed by the network manager now

janeksz, 2017-01-11:
@florian-popup 
Thanks again. 
I have tried your suggestions (bridge over wlan0), but still no luck.
I don't have NetworkManager on my Raspberry Pi as it doesn't come with Raspbian OS
Do you think I should install it? Would it make a change?

florian-popup, 2017-01-11:
Well, I have exactly the same set up as you do : a rpi 3 with raspbian with
a cc2650 as slip radio and 2 cc2650 as mote.

Can you use the UI to connect to the wifi ? If so, you should have
NetworkManager already.
Type sudo service network-manager status

Be sure that you try to connect to the ip in /var/log/6lbr.log (just to be
sure)

Can set up the wifi configuration for the 6lbr and do :
sudo service 6lbr stop
sudo rm /var/log/6lbr.*
sudo service 6lbr start

wait for 20-30 sec then
sudo service 6lbr stop

then check your /var/log/6lbr.err
it should have only one line in it. If you see that the process keep
restarting, check your /var/log/6lbr.log and tell me if you have an error
line in it saying that wlan0 can't bridge to br0. If so I can tell you how
to manage to avoid it

janeksz, 2017-01-11:
@florian-popup 
I understand, that you don't have it working using wlan0, correct?
It doesn't work for me on both RPi2 (using USB wifi dongle) and RPi3 (using built-in wifi) both using latest Raspbian and latest 6lbr.
Both have network connection working and I am using them in headless mode.
Thanks for looking at this issue and please, let me know if there is any new development.
I will try to contact Laurent by email, to see his view.

laurentderu, 2017-01-12:
Bridging to a WiFi interface is complex and depends on several factors (and the old faq ins't really helpful about that, I will update it)

To have WiFi bridging you need two thinks :
* The WiFi chipset must support AP mode (or promiscuous mode)
* The WiFi interface must be configured as an Access Point (so the host must becomes a WiFi Router)

The confusion comes from some WiFI chipset and drivers that in the past allowed to bridge the WiFi interface even not in AP mode (which seems to be forbidden by the WiFi standard...)

If you don't want to make your host as a WiFi AP there are two solutions :
* Use routing between the BR tap Interface and the WiFi interface, this can be done by enable ip forwarding in the host. The BR should be configured as SmartBridge with a manual configuration. An configuration example is available here : https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration#routing-mode
* Use ProxyARP to create a fake bridge between the BR Tap Interface and the WiFi bridge. This however will only work for IPv4. For IPv6 ou will need proxy_ndp or npdd. See https://wiki.debian.org/BridgeNetworkConnectionsProxyArp for how to setup proxy arp. And here to setup ndppd https://github.com/DanielAdolfsson/ndppd This solution is more flexible but also more complex to setup and I actually never tried it !

Tell me how it works for you and I will update the doc accordingly and if it does not work I'll find more information.

PasqualeSalierno, 2017-01-13:
Hello! I’m a young hardware tester and now i am working on X.ip5 SmartMachine® a new board based on a powerful ARM® Cortex®-M3, very low active RF, MCU current and lowpower mode current consumption provide excellent battery lifetime. The X.ip5 is a 6LoWPAN device: low-power, IP-driven nodes and large mesh network support make this X.ip5 SmartMachine® the ideal solution for Internet of Things applications, thanks to its small size, the compact integrated antenna on board and battery supply.
More information on

http://telecontrolli.com/resource/downloads/datasheet/iot-system/x-ip-transceiver-family/82-x-ip5-datasheet/file.html

kameleon070, 2017-01-15:
Czesc Janek sorry za offtop, czy moglby sie z Toba skontaktowac, mam kilka pytan co do 6lbr?

janeksz, 2017-01-17:
@laurentderu Thanks for helping, much appreciate it.

I"m trying to use solution 1 as suggested by you.
My /etc/6lbr/6lbr.conf  is:

#######################################
_#This file contains a default configuration for Raspberry PI platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

#MODE=ROUTER
MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=0
DEV_ETH=wlan0
DEV_BRIDGE=br0
DEV_TAP=tap0

RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

NODE_CONFIG=/etc/6lbr/node_config.conf

LOG_LEVEL=3 #INFO and above only_
##################################################

my /etc/network/interfaces is:

#-----------------------
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

iface wlan0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
    bridge_ports wlan0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show wlan0 | grep ether | awk '{print $2}'`

allow-hotplug wlan0
iface wlan0 inet manual
    wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf

iface default inet dhcp

#-----------------------

After rebooting and starting 6lbr service, my ifconfig is:
 
root@raspberrypi:~# ifconfig
br0       Link encap:Ethernet  HWaddr b8:27:eb:54:9c:2b
          inet addr:169.254.162.190  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::1d:4dff:fe71:34a7/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
[... 80 lines cut ...]
         0115000f 00004011 79ca0000 0000ffff 
         ffff0044 00430101 ae880101 06000100 
         00000000 80000000 00000000 00000000 
         00000000 00000200 07b5cb00 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00000000 00000000 00000000 
         00000000 00006382 53633501 01370301 
         0306ff
2017-01-17 22:30:01.906529: PACKET: TAP: write: 291
2017-01-17 22:30:05.203519: PACKET: TAP: read: 382
2017-01-17 22:30:05.282624: PACKET: TAP: read: 382
2017-01-17 22:30:13.248182: PACKET: TAP: read: 342
2017-01-17 22:30:20.095790: PACKET: TAP: read: 342

Any conclusions?

laurentderu, 2017-01-20:
@janeksz If you intend to use the routing interfacing, with your configuration you are still bridging wlan0 and tap0, which does not work. You should have in your 6lbr.conf file :

    BRIDGE=0
    RAW_ETH=0

In your /etc/network/interfaces you should have a normal interface for wlan0, not a bridge.

If you have :: in 6lbr.ip, that means the BR is still waiting for the network configuration. You must instead configure it manually using the nvm_tool :

   nvm_tool --update --wsn-accept-ra 0 /etc/6lbr/nvm.dat

Or, if you are using 1.5.x, remove the /etc/6lbr/nvm.dat file to reset to default configuration.

janeksz, 2017-01-20:
@laurentderu, thanks but I am still confused, so will need your advice step by step.
I will try again on Monday, as I'm away for weekend.
My main confusion is to understand the difference between modes ( many of them) and interface configurations (3 types) and how they are related. 
Anyhow following your previous post: 
   

> Use routing between the BR tap Interface and the WiFi interface, this can be done by enable ip forwarding in the host. The BR should be configured as SmartBridge with a manual configuration. An [...]

my /etc/6lbr/6lbr.conf will be:
#######################################
_#This file contains a default configuration for Raspberry PI platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

#MODE=ROUTER
MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=0
CREATE_BRIDGE=0
DEV_ETH=wlan0
DEV_BRIDGE=br0
DEV_TAP=tap0

RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

NODE_CONFIG=/etc/6lbr/node_config.conf

LOG_LEVEL=3 #INFO and above only_
##################################################

Is this correct?

janeksz, 2017-01-23:
@laurentderu   
and my /etc/network/interfaces is:
//------------------------------------
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
//-------------------------------------

root@raspberrypi:~# service 6lbr status
● 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr)
   Active: active (running) since Sun 2017-01-22 23:55:25 UTC; 2s ago
  Process: 888 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/6lbr.service
           ├─894 /bin/sh /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
           ├─898 /usr/lib/6lbr/bin/cetic_6lbr_smart_bridge -c /etc/6lbr/nvm.d...
           ├─899 /bin/sh /usr/lib/6lbr/6lbr-watchdog 898 60 3 /var/log/6lbr.t...
           └─982 sleep 1

Jan 22 23:55:24 raspberrypi 6lbr[888]: Starting 6LoWPAN Border Router:.
Jan 22 23:55:25 raspberrypi systemd[1]: Started LSB: 6LoWPAN Border Router.
//----------------------------------
root@raspberrypi:~# ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:01:c9:7e
          inet6 addr: fe80::8943:5157:2336:12a7/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:240 (240.0 B)  TX bytes:240 (240.0 B)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          inet addr:169.254.77.32  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::ce9d:13fd:1f4:8248/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1455 (1.4 KiB)  TX bytes:10147 (9.9 KiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:54:9c:2b
          inet addr:172.19.114.51  Bcast:172.19.115.255  Mask:255.255.254.0
          inet6 addr: fe80::880d:43d2:b0ee:b09f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:327 errors:0 dropped:232 overruns:0 frame:0
          TX packets:143 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:72747 (71.0 KiB)  TX bytes:26358 (25.7 KiB)

root@raspberrypi:~#
//-----------------------------------------
/var/log/6lbr.ip is: aaaa::212:4b00:7b5:cb00 

**looks close, but no cigar (unable to access [bbbb::100] from web browser)**

Attached is the log file (renamed from6lbr.log to 6lbr.txt)
[6lbr.txt](https://github.com/cetic/6lbr/files/722354/6lbr.txt)

janeksz, 2017-01-23:
@laurentderu 
Thanks for helping
Additional info - my nvm.dat is:
root@raspberrypi:~# cat /var/log/6lbr.ip
aaaa::212:4b00:7b5:cb00
root@raspberrypi:~# /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : False
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : fd00::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 63
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 1

RPL DAO ACK : False
RPL DAO ACK local repair : False
DIO triggers route refresh : True
MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay enabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : False

NAT 64 Enabled : True
NAT 64 DHCP Enabled : True
NAT 64 static port mapping Enabled : True
NAT 64 RFC-6052 prefix : True
NAT 64 address : 192.168.1.55
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.19.114.1

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80

root@raspberrypi:~#

florian-popup, 2017-01-23:
@janeksz 
As far as I m understanding, you have to play with the routing table in order to redirect the connection on your wifi to your tap0.

In order to do this, you have to configure your 6lbr with
MODE=ROUTER
BRIDGE = 0 #You want to redirect the connection yourself
DEV_ETH=wlan0
RAW_ETH = 0

then in your interface file, you should not define br0

Once you restarted 6lbr, take a look at the ipv6 routing table 
sudo route -6 -n

then add a route in local to the server to the 6lbr table
sudo route -A inet6 add bbbb::/64 gw bbbb::100 dev tap0
(if your server address is bbbb::100)

add a route to the wifi interface to the local loopback (which will be redirected according to the previous route) to the 6lbr table
sudo route -A inet6 add bbbb::/64 dev wlan0

add a route to your devices network from the wifi interface
sudo route -A inet6 add fd00::/64 gw bbbb::100 dev wlan0
if your wifi node network is fd00::/64

You should be able to reach 6lbr server from it's own hosting device and have internet.
If your network is already configured to ipv6, I guess that you have to add a redirection from your local network address of your wlan interface to your 6lbr server address.

If you use ip64 tunneling, you have to configure the ip4 routing table.
cf route -n

I tried to redirect the local network address of my wlan interface to my 6lbr server ipv4 address but it didn't work.
I also tried to match the 6lbr ipv4 address to my local network address but it didn't work either.
Any guess @laurentderu ?

janeksz, 2017-01-24:
@florian-popup , thanks for sharing
//------------------------
my /etc/6lbr/6lbr.conf is same as yours
root@raspberrypi:~# service 6lbr status
● 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr)
   Active: active (running) since Tue 2017-01-24 00:00:58 UTC; 5min ago
  Process: 2836 ExecStop=/etc/init.d/6lbr stop (code=exited, status=0/SUCCESS)
  Process: 2848 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/6lbr.service
           ├─2856 /bin/sh /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
           ├─2864 /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o...
           ├─2865 /bin/sh /usr/lib/6lbr/6lbr-watchdog 2864 60 3 /var/log/6lbr...
           └─4911 sleep 1
Jan 24 00:00:58 raspberrypi 6lbr[2848]: Starting 6LoWPAN Border Router:.
Jan 24 00:00:58 raspberrypi systemd[1]: Started LSB: 6LoWPAN Border Router.
//--------------------
root@raspberrypi:~# sudo route -6 -n
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
bbbb::/64                      ::                         U    205 0     0 tap0
fe80::/64                      ::                         U    256 0     0 eth0
fe80::/64                      ::                         U    256 0     0 wlan0
fe80::/64                      ::                         U    256 0     0 tap0
::/0                           ::                         !n   -1  1    17 lo
::1/128                        ::                         Un   0   3     3 lo
bbbb::/128                     ::                         Un   0   1     0 lo
bbbb::846b:e981:733d:e263/128  ::                         Un   0   1     0 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::880d:43d2:b0ee:b09f/128  ::                         Un   0   1     0 lo
fe80::8dc8:28b2:6a4b:9633/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 eth0
ff00::/8                       ::                         U    256 3    15 wlan0
ff00::/8                       ::                         U    256 4    30 tap0
::/0                           ::                         !n   -1  1    17 lo
root@raspberrypi:~#
//---------------------------------------------
root@raspberrypi:~# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         **172.19.114.1**    0.0.0.0         UG    303    0        0 wlan0
169.254.0.0     0.0.0.0         255.255.0.0     U     205    0        0 tap0
172.19.114.0    0.0.0.0         255.255.254.0   U     303    0        0 wlan0
root@raspberrypi:~#
//--------------------------------------

Next I executed steps 1,2,3 as per your post above (same addresses as yours)
root@raspberrypi:~# sudo route -A inet6 add bbbb::/64 gw bbbb::100 dev tap0
root@raspberrypi:~# sudo route -A inet6 add bbbb::/64 dev wlan0
root@raspberrypi:~# sudo route -A inet6 add fd00::/64 gw bbbb::100 dev wlan0
root@raspberrypi:~#
I have tried here on Windows client  IPv6 network, theoretically it should work now, but it doesn't
root@raspberrypi:~# cat /var/log/6lbr.ip
bbbb::100
root@raspberrypi:~# cat /var/log/6lbr.ip4
0.0.0.0
root@raspberrypi:~#
from web browser - The server at bbbb::100 is taking too long to respond.
//------------------------------------------

Moving into IPV4 network. 
I am not getting IPv4 address from 6lbr
root@raspberrypi:~# cat /var/log/6lbr.ip4
0.0.0.0
root@raspberrypi:~#

root@raspberrypi:~# /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : False
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : fd00::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 75
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 1

RPL DAO ACK : False
RPL DAO ACK local repair : False
DIO triggers route refresh : True
MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay enabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : False

NAT 64 Enabled : True
NAT 64 DHCP Enabled : True
NAT 64 static port mapping Enabled : True
NAT 64 RFC-6052 prefix : False
NAT 64 address : 192.168.1.55
NAT 64 netmask : 255.255.255.0
**NAT 64 gateway : 172.19.114.1**

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80

root@raspberrypi:~# /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat

//----------------------
I want to use IP64. I have modified /etc/6lbr/nvm.dat for NAT64 DHCP as above, but I am unable to obtain an IPv4 address from 6lbr, so I am stack here ...

janeksz, 2017-01-24:
@florian-popup @laurentderu 
On my working 6lbr version I checked IP64 address here:
[image]
even if it is not shown here:
root@raspberrypi:~# cat /var/log/6lbr.ip4
0.0.0.0
How can I read valid IP64 address from the command line?

florian-popup, 2017-01-24:
@janeksz 
You don't get any ip address because you have NAT 64 DHCP Enabled
DHCP is a protocol that gives dynamically the ip addresses
Desactivate it and you will get your ip address defined as 192.168.1.55 if there is no conflict (not already given by the dhcp router to another device)

florian-popup, 2017-01-24:
@janeksz 

Just after you start 6lbr, check if you can reach [bbbb::100] from your rpi.
If you don't, there is an issue with the routing table. You have to delete all routing rules related to bbbb and wlan at the same time
sudo route -A inet6 del bbbb::/64 dev wlan0
sudo route -A inet6 del bbbb::/64 gw bbbb::100 dev wlan0

Then follow all the steps as I descibed.

EDIT !!!!!!!!!!!!!!!! YOU HAVE TO ENABLE THE POSTROUTING
sudo iptables -t nat -A POSTROUTING -j MASQUERADE

janeksz, 2017-01-24:
@florian-popup, thanks again 
re DHCP, I actually want a dynamic address and in my old system it is created and both IPv6 and IP64 work, but there is still a question how to read it from the command line (not GUI). In university environment IP addresses change very frequently and it is hard to have a static IP

janeksz, 2017-01-24:
@florian-popup, thanks again. I will try your further suggestion, when back at uni.

florian-popup, 2017-01-24:
@janeksz 
As Laurent suggested, ProxyARP would be worth a look. 
It will basiclly do the same thing as changing the ip tables and allowing ip forwarding but you won't have to manipulate the ip tables nor weaken the security and you would be able to use dhcp easly, have an autoconfiguration & autostart

janeksz, 2017-01-24:
@florian-popup , thanks for your patience, that thing is killing me, I wish I knew more about networking.

After  starting 6lbr service, bbbb::100 is reachable from RPi:
root@raspberrypi:~# ping6 bbbb::100
PING bbbb::100(bbbb::100) 56 data bytes
64 bytes from bbbb::100: icmp_seq=1 ttl=64 time=8.17 ms

After executing: 
sudo route -A inet6 add bbbb::/64 gw bbbb::100 dev tap0
sudo route -A inet6 add bbbb::/64 dev wlan0
sudo route -A inet6 add fd00::/64 gw bbbb::100 dev wlan0
my routing table looks like this:
root@raspberrypi:~# route -6 -n
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
bbbb::/64                      bbbb::100                  UG   1   0     0 tap0
bbbb::/64                      ::                         U    204 1     5 tap0
fe80::/64                      ::                         U    256 0     0 eth0
fe80::/64                      ::                         U    256 0     0 wlan0
fe80::/64                      ::                         U    256 0     0 tap0
::/0                           ::                         !n   -1  1    22 lo
::1/128                        ::                         Un   0   3     3 lo
bbbb::/128                     ::                         Un   0   1     0 lo
bbbb::846b:e981:733d:e263/128  ::                         Un   0   4     5 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::4a90:102:ce61:3090/128   ::                         Un   0   1     0 lo
fe80::880d:43d2:b0ee:b09f/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 eth0
ff00::/8                       ::                         U    256 3    15 wlan0
ff00::/8                       ::                         U    256 3    28 tap0
::/0                           ::                         !n   -1  1    22 lo

On my Windows client, still no access to [bbbb::100]

Can you see anything obviously wrong? 

[image]

florian-popup, 2017-01-25:
@janeksz 
Ok, so when you use route rules, the 6lbr servers interface isn't connected directly to you local network. Thus, you can't use DHCP or access to 6lbr directly.

So you have to enable ip forwarding (in your case ipv6 forwarding) on your rpi
sysctl -w net.ipv6.conf.all.forwarding=1

Use masquerade too in sixx mode
sudo ip6tables -t nat -A POSTROUTING -o sixxs  -j MASQUERADE

DO NOT FORGET that you can't reach the bbbb::/64 network directly on a client because the local network router (ie your internet box) does either not listen to that network and cannot reach the 6lbr serveur interface.

Only your rpi knows how to reach the bbbb network via the rules in the ip tables
The way I used to resolve that problem is to create a rule in the ip table of the client saying that in order to reach the bbbb network, you have to pass through the local network interface of your pi.
On a debian  client machine, it would be
sudo route -A inet6 del bbbb::/64 gw YOUR_RPIS_WLAN_IPV6_ADDR dev wlan0

(that works for ipv4, I tested it 2 times howerver, I don't have any ipv6 network and router configured so I can't test it but the mechanisms are exactly the same)

I m trying to set up an ARP on the rpi so there will be no need of such route rules, the 6lbr interface will be seen as a part of the local network thus, enabling DHCP, direct addressing ... but this is not my priority right now

laurentderu, 2017-01-26:
@florian-popup Thank you for providing such detailed information, I will try to digest it and improve the documentation :) Maybe your configuration could be simplified by switching to SmartBridge mode and statically configure the 6LBR IPv6 address and prefix. With that the bbbb::/64 is no longer needed and you could only keep the fd00::/64 routing.

@janeksz Related to https://github.com/cetic/6lbr/issues/237#issuecomment-274698558, I looked at it and it's actually a bug. The .ip file is not updated when the IPv4 configuration is changed. I will fix that.

If you can modify the configuration of your WiFi AP and if it is sufficiently advanced, you could configure it to add a Route Information Option in the IPv6 network configuration to create the routes for the fd00::/64 (and bbbb::/64) subnets via the RPi link local wlan address. This is what 6LBR is doing to create the route to the WSN when in Router mode.

janeksz, 2017-01-26:
@florian-popup, @laurentderu , thank you both. I am away for a few days. I couldn't make it work for now, will have a look again later.

MarCast, 2017-01-27:
Hello everyone, I have already built the 6lbr many times, but since yesterday I am having problems. I'm trying to create a new system and when I click on the [bbbb :: 100] the sensor page, says connection failed. It will be a new update?

janeksz, 2017-01-29:
@MarCast, try to return to the last working version (hardware and software) and then change one thing at a time. That is, what I do, when in trouble. Good luck.

electronut, 2017-02-10:
@florian-popup thanks for your message above on wlan0.

In my case, I am trying the simple pi setup as shown in the page below:

https://github.com/cetic/6lbr/wiki/Example-1:-Hello-6LBR

Instead of connecting eth0 to another computer, I have plugged it directly into the network. (The RJ45 cable of the Pi.)

So in that case, what is the correct routing command to be applied so that I can bring up [bbbb::100] in the browser? Currently it refuses to connect.

Thanks for any help.

florian-popup, 2017-02-10:
@electronut 

If you are using ethernet, you should have no problems making it working (I mean that you don't have to make change about the ip tables as opposed as if you used wifi instead).
In your case if the server doesn't bring up, you might have an error in your configuration. Stop the 6lbr service, suppress your 6lbr logs files and bring 6lbr up. Check your 6lbr log file (/var/log/6lbr.log)
Also check the address of the 6lbr server in /var/log/6lbr.ip . If it is empty, you have a configuration error. If not, this is the address you have to connect to.

Look at the 1 st message I put in this thread, I gave some proper configuration files working with ethernet. Due to display command github, the line in bold text have a comment sign before (hashtag)
Be sure that you changed both your /etc/network/interfaces file & your  /etc/6lbr/6lbr.conf.
Update yours according to. (my eth0 interface is named enxb827eb15ec3a and I don't know why)

At this point you should be able to connect to your 6lbr server FROM YOUR PI (using a web broswer).
From this, the simpliest way to access it from another computer is to enable the 6to4 translation via the 6lbr server interface & then to configure your computer to forward all ipv6 address request to you pi ipv4 main network address (ipv4 address of your eth0 interface)

electronut, 2017-02-10:
@florian-popup thanks for the prompt reply. Here are my configurations:

    pi@raspberrypi:/etc/6lbr $ cat /var/log/6lbr.ip
    bbbb::100

    pi@raspberrypi:/etc/6lbr $ tail -10 /var/log/6lbr.ip
    bbbb::100

`pi@raspberrypi:/etc/6lbr $ tail -10 /var/log/6lbr.err
Fri 10 Feb 10:24:21 SCT 2017 : Starting 6LBR
6LBR killed
********************************************************************************
Fri 10 Feb 10:26:06 SCT 2017 : Starting 6LBR
6LBR killed
********************************************************************************
Fri 10 Feb 10:26:59 SCT 2017 : Starting 6LBR
6LBR killed
********************************************************************************
Fri 10 Feb 11:17:28 SCT 2017 : Starting 6LBR
pi@raspberrypi:/etc/6lbr $ tail -10 /var/log/6lbr.log
CoAP push started
CoAP server started
2017-02-10 11:17:41.063010: INFO: DNS: DNS proxy started
2017-02-10 11:17:41.063100: INFO: PLUGIN: Initialising dummy
2017-02-10 11:17:41.063173: INFO: DUMMY: Dummy init
2017-02-10 11:17:41.063281: INFO: PLUGIN: Initialising lwm2m-client
2017-02-10 11:17:41.063353: INFO: LWM2M: LWM2M Client init
2017-02-10 11:17:41.063474: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-02-10 11:17:41.063555: INFO: 6LBR: CETIC 6LBR Started
RD client started`

`pi@raspberrypi:/etc/6lbr $ cat /etc/6lbr/6lbr.conf
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=3 #INFO and above only`

And here are the existing routes:

`pi@raspberrypi:/etc/6lbr $ route -6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
bbbb::/64                      ::                         U    256 4  2210 eth0
fe80::/64                      ::                         U    256 0     0 wlan0
fe80::/64                      ::                         U    256 0     0 eth0
::/0                           ::                         !n   -1  1  2690 lo
::1/128                        ::                         Un   0   1     0 lo
bbbb::/128                     ::                         Un   0   1     0 lo
bbbb::101/128                  ::                         Un   0   5    75 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::d526:a818:27e6:ae07/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 4 46022 eth0
ff00::/8                       ::                         U    256 0     0 wlan0
::/0                           ::                         !n   -1  1  2690 lo`

In /etc/network/interfaces I have:

`iface eth0 inet manual`

With the above, I am still not able to reach [bbbb:100]

florian-popup, 2017-02-10:
@electronut 

In your 6lbr conf try

RAW_ETH=0
BRIDGE=1

and in your interface file put

auto br0
iface br0 inet dhcp
bridge_ports eth0
bridge_stp off
up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
post-up ip link set br0 address `ip link show eth0 | grep ether $

Restart your pi

electronut, 2017-02-10:
Thanks @florian-popup 

The cmd below seems to be missing something:

```
`ip link show eth0 | grep ether $
```

Should it instead be:

```
`ip link show eth0 | awk '/ether/ {print $2}'
```

florian-popup, 2017-02-10:
@electronut 
Well, I have no idea and can't guess without seeing your logs. clear your logs, make the crash happend and post your 6lbr.log entirely. 
It might have been better to open a new thread since the problems are different, one is configuration, the other is pi conpatibilité with wifi chipset althrough the title is confusing here

janeksz, 2017-02-20:
@florian-popup @laurentderu Unfortunately in uni labs students are using Windows computers and don't have admin rights to change routes. We need to use for now the old Raspian and old RPi2 to be able to use WiFi with 6lbr. Hopefully some solution will come soon, before the next semester. Thanks again for your support.

INemesisI, 2017-02-27:
Hello,
I have used 6lbr on a raspberry pi 3 for about 4 months now. Unfortunately since this week I am unable to access the web interface of 6lbr.
Suspecting, that I screwed something up in my configuration, I tried to setup everything again on a fresh raspbian installation. 

To not overload this issue, I put all Info into a gist:
https://gist.github.com/INemesisI/33aaea8b31c5501e03e38a3100f75367

The weird thing is: I am able to access the website the first time right after restarting my computer. After a certain time http://[bbbb::100]/ gets unreachable.
The same thing happened, when I tried to access the website from my phone!

I spend so much time on this issue now and have no idea anymore, what to do. I would be really glad, if you could help me out here.

Also I have a second question:
How do I need to setup 6lbr, if I want to be able to run it without any external network connection (no ethernet or wifi)? I would like to be able to just use it as a standalone local network.

laurentderu, 2017-03-08:
@INemesisI I looked at your logs, and I don't see anything wrong at first sight. Loosing connection after a while is often linked to issues with the bridge on the BR host. It could either be that the multicast snooping is reactivated by some other process or a network manager app could reconfigure the bridge interface.
The later could be plausible as the eth0 and tap0 interfaces have IPv4 address although they are disabled in /etc/network/interfaces

For your second question, by default 6LBR in Router mode can work in standalone mode. You don't even have to create a bridge if you plan to only reach it from the hosting platform.

CreativityH, 2017-03-10:
@MarCast @laurentderu Hi,  I am able to access the website the first time right after restarting my computer,  when I click on the [bbbb :: 100] the sensor page, says connection failed.  Cloud you tell me how you solve this problem?

electronut, 2017-03-10:
@CreativityH I had the same issue. Just get the dev branch. See issue #246

chenek, 2017-05-02:
There is a step-by-step how-to at http://sunmaysky.blogspot.tw/2017/04/running-6lbr-on-raspberry-pi-with-wifi.html which shows you run 6lbr with WiFi enable on Raspbian.

NaniteFactory, 2017-05-14:
@florian-popup @laurentderu
Thank you for sharing detailed information about 6lbr configuration with wifi.

I'm trying to connect my 6lbr to a wifi network but all the clients other than the device running 6lbr itself are not able to access [bbbb::100].

On my RPi3 (with 6lbr), I've tried followings:

```
MODE=ROUTER
RAW_ETH=0
BRIDGE=0
DEV_ETH=wlan0

sysctl -w net.ipv6.conf.all.forwarding=1
sudo iptables -t nat -A POSTROUTING -j MASQUERADE
sudo route -A inet6 add bbbb::/64 gw bbbb::100 dev tap0
sudo route -A inet6 add bbbb::/64 dev wlan0
sudo route -A inet6 add fd00::/64 gw bbbb::100 dev wlan0
```

And then on my client side RPi3, I did this thing:

`sudo route -A inet6 add bbbb::/64 gw 2003::1 dev wlan0`

[2003::1] is a global ipv6 address of my (runnnig 6lbr) RPi's wlan0.

But still, unable to ping6 bbbb::100 from the client.

there was actually some progress though.

before executing command `sudo route -A inet6 add bbbb::/64 dev wlan0`

`ping6 bbbb::100` would not even respond (got timed out),

but now it at least gives me an error saying "host unreachable"

[2003::1] is absolutely reachable. and I can also successfully ping6 from the client to bbbb::(whatever) on the tap0 interface but to bbbb::100 .

please help me. I spent so many hours on this trying all kinds of possible combinations of ip tables. xD

what would be the problem?

janeksz, 2017-11-13:
@chenek @laurentderu @florian-popup 
Can anyone advise on a solution to run 6lbr on RPi3 using WiFi?

florian-popup, 2017-11-13:
@janeksz @NaniteFactory 
Unfurtunatly I m not working on using 6lbr anymore cause I got a mcu that is not supported by contiki natively.
So I switched to RIOT. Apparently Riot got some tools from contiki (the tuntap for exemple). If I ever make this works with the 6lbr interface I ll use 6lbr as border router and investigate that wifi routing but I have been unsucessfull up to now

florian-popup, 2017-11-13:
@janeksz @NaniteFactory
Maybe adapting this script by using the wifi interface  and defining the input/output can be worth a try: https://github.com/contiki-os/contiki/commit/8bfb8562202082b1acbf9684a7eb6ddeca20ffea
As it will use ipv6 forwarding with Masquerade

janeksz, 2017-11-16:
@florian-popup @laurentderu  @NaniteFactory  thanks, I found this http://www.thingsquare.com/blog/articles/wireless-sensor-network-raspberry-pi-zero-w/ and will try to investigate how to

horopciuc, 2018-02-11:
Hey there! 
Has anyone succeeded in accessing the web page (bbbb) using the eth0 connection? I've tried various methods, but still no success. Can someone help me with the /network/interfaces and 6lbr.conf.rpi files? Also I'm unable to find the 6lbr.conf file. I've recloned the 6lbr library from github , and I've noticed that the file that previous was 6lbr.conf might have changed it's name into 6lbr.conf.rpi . 
Could someone give me a hand?
Thanks!

chenek, 2018-02-11:
@horopciuc Try to refer to http://sunmaysky.blogspot.tw/2015/08/setup-6lbr-to-run-6lowpan-with-cc2531.html?m=1

horopciuc, 2018-02-13:
Thank you @chenek , I've tried that, but still no result.  I was wondering if there's someone who managed to access the bbbb page after the recent update of 6lbr.