cetic/6lbr #239: Problem with 6lbr on Cooja

kameleon070, 2017-01-12:
Hi, 

I've tried to run this https://github.com/cetic/6lbr/wiki/COOJA-Interface. Everything is ok till i do make cooja-small, in conosle i see:  "Could not load plugin class: be.cetic.cooja/plugins.RadioLoggerHeadless" 
What is more when i' m trying to run router i see this: "/package/usr/bin/6lbr conf-templates/cooja-1-router.conf not found"

I'm running my Cooja in Contiki on VM.

Thx for help,

Kamil

laurentderu, 2017-01-12:
A few question :)

Could you tell me which version of 6LBR repository you are using ? Have you done a git submodule update --init ? Which command are you using to launch the BR ? what is the full log ?

kameleon070, 2017-01-12:
Hi, i add my repository from cetic side + adding by 'git clone --recursive https://github.com/cetic/6lbr'
Now i've updated java and have some new problems. When i've tried to sudo make cooja-small i see that it's problem to find jni.h. 

Look at this there are so many failed builds, maybe i am so dump...
I'm using virtual box, maybe this is the problem?
`user@instant-contiki:~/user/6lbr/examples/6lbr/test$ make cooja-small 
python2.7 gensimfile.py coojagen/examples/config_preset_1dag_10nodes.py
LOADING CONFIG coojagen/examples/config_preset_1dag_10nodes.py
****
coojagen/output/coojasim-preset_1dag_10nodes-12-20170112140911.csc
Done. Generated 1 simfiles
pushd ../../../tools/cooja; ant jar; popd
~/user/6lbr/tools/cooja ~/user/6lbr/examples/6lbr/test
Buildfile: /home/user/user/6lbr/tools/cooja/build.xml

init:

compile:

copy configs:

jar_cooja:

jar:

init:

compile:

jar:

init:

mspsim:

init:

compile:

jar:

coffee:

init:

compile:
    [javac] Compiling 8 source files to /home/user/user/6lbr/tools/coffee-manager/build

configs:

jar:
      [jar] Building jar: /home/user/user/6lbr/tools/coffee-manager/coffee.jar
     [copy] Copying 1 file to /home/user/user/6lbr/tools/cooja/apps/mspsim/lib

compile:

copy:

jar:

init:

compile:

jar:

init:

clean:
   [delete] Deleting directory /home/user/user/6lbr/tools/cooja/apps/serial_socket/build
[... 62 lines cut ...]
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x1f200 ( 27496 == 0x6b68 bytes)
 INFO [main] (ContikiMoteType.java:500) - mtype221.cooja: offsetting Cooja mote address space: 0x44f58000
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x44f77020 ( 456 == 0x1c8 bytes)
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x44f77200 ( 27496 == 0x6b68 bytes)
 INFO [main] (CompileContiki.java:140) - > make TARGET=cooja clean 
 INFO [main] (CompileContiki.java:140) - > make 6lbr-demo-delay.cooja TARGET=cooja 
 INFO [main] (ContikiMoteType.java:404) - Creating core communicator between Java class Lib2 and Contiki library '/home/user/user/6lbr/examples/6lbr/test/coojagen/firmwares/6lbr-demo-delay/obj_cooja/mtype883.cooja
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x2c020 ( 1206 == 0x4b6 bytes)
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x2c4e0 ( 52328 == 0xcc68 bytes)
 INFO [main] (ContikiMoteType.java:500) - mtype883.cooja: offsetting Cooja mote address space: 0x45020000
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x4504c020 ( 1206 == 0x4b6 bytes)
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x4504c4e0 ( 52328 == 0xcc68 bytes)
 INFO [main] (CompileContiki.java:140) - > make TARGET=cooja clean 
 INFO [main] (CompileContiki.java:140) - > make 6lbr-demo.cooja TARGET=cooja 
 INFO [main] (ContikiMoteType.java:404) - Creating core communicator between Java class Lib3 and Contiki library '/home/user/user/6lbr/examples/6lbr/test/coojagen/firmwares/6lbr-demo/obj_cooja/mtype472.cooja
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x2c020 ( 1206 == 0x4b6 bytes)
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x2c4e0 ( 52328 == 0xcc68 bytes)
 INFO [main] (ContikiMoteType.java:500) - mtype472.cooja: offsetting Cooja mote address space: 0x450fc000
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x45128020 ( 1206 == 0x4b6 bytes)
 INFO [main] (ContikiMoteType.java:573) - Parsed section at 0x451284e0 ( 52328 == 0xcc68 bytes)
 INFO [main] (Visualizer.java:256) - Registered external visualizer: org.contikios.cooja.plugins.skins.DGRMVisualizerSkin
 INFO [main] (ScriptRunner.java:430) - Test script deactivated
FATAL [main] (Cooja.java:3670) - Could not load plugin class: be.cetic.cooja.plugins.RadioLoggerHeadless
 INFO [AWT-EventQueue-0] (ScriptRunner.java:430) - Test script deactivated
`

kameleon070, 2017-01-13:
I had a problem with Java. :)

closed 2017-01-13 by kameleon070

laurentderu, 2017-01-13:
What was the issue actually ? I could update the documentation with your feedback

kameleon070, 2017-01-14:
It was problem with woirking on VirtualBox, now i'm using VM and everything is ok. I'm writting Engineering Thesis. And i have a few quewstions. Can i conntact you via mial(please write if you can kamil.bojar@gmial.com)? I you can;t or you don't want. Please tell mi if in cooja-small(cooja 6lbr Test) I can run full Nat64?

laurentderu, 2017-01-16:
cooja-small can run full NAT64 without problem, you just have to enable it in the 6LBR configuration page.

stnitomy, 2017-03-17:
Even I am facing same issue.
Can you help me how to solve it. 

pi@raspberrypi:~/6lbr/examples/6lbr/test $ make cooja-small
python2.7 gensimfile.py coojagen/examples/config_preset_1dag_10nodes.py
LOADING CONFIG coojagen/examples/config_preset_1dag_10nodes.py
****
coojagen/output/coojasim-preset_1dag_10nodes-12-20170317143419.csc
Done. Generated 1 simfiles
pushd ../../../tools/cooja; ant jar; popd
~/6lbr/tools/cooja ~/6lbr/examples/6lbr/test
Error: JAVA_HOME is not defined correctly.
  We cannot execute /usr/lib/jvm//bin/java
~/6lbr/examples/6lbr/test
pushd ../../../tools/cooja/apps/serial2pty; echo /usr/lib/jvm/; ant jar; popd
~/6lbr/tools/cooja/apps/serial2pty ~/6lbr/examples/6lbr/test
/usr/lib/jvm/
Error: JAVA_HOME is not defined correctly.
  We cannot execute /usr/lib/jvm//bin/java
~/6lbr/examples/6lbr/test
pushd ../../../tools/cooja/apps/radiologger-headless; ant jar; popd
~/6lbr/tools/cooja/apps/radiologger-headless ~/6lbr/examples/6lbr/test
Error: JAVA_HOME is not defined correctly.