cetic/6lbr #240: 6LBR+RPi Crashing with large number of CC1310 nodes

stringr9, 2017-01-12:
I'm experiencing an issue with 6lbr where the router crashes randomly when there are over 30 nodes in range. 

My setup is:
- One Raspberry Pi with a CC1310 based slip-radio as a BR running 6LBR 1.4.1
- 40 CC1310 based nodes running the latest Contiki commit with the `hello-world` example (only sending network maintenance data, no application data packets)

All nodes are positioned within 5 meters of the BR and connect directly to it (don't mesh through other nodes). Up to 30 nodes on the network works fine, all OK on the 6LBR dashboard and always pingable.  I've run this for a few hours without issue.

Once I start to add more nodes on the network the dashboard will crash at some random time after startup.  Once crashed, the dashboard will be unreachable, 6LBR will not be pingable and none of the nodes will be pingable.  However, `tail`ing the 6lbr.log file shows no error and packets apparently continue to be received and transmitted by the slip radio (`wireless_input` and `wireless_output` messages still continue to appear).

The crash seems to happen more quickly with more nodes.  (e.g. 35 nodes will take up to a minute minute to crash, 40 nodes will crash within the first 30 seconds)

stringr9, 2017-01-13:
After letting this run overnight it seems like 6LBR doesn't actually crash, but just locks up and becomes very slow.  The dashboard will load a page eventually around 10% of the time. 

The load average on the Pi is very low at `0.07, 0.07, 0.09` at the 6LBR process only taking around 6% of CPU.  The next highest process when I run `top` is less than 1% of the CPU.

laurentderu, 2017-01-13:
That's a behavior I have never seen until now. Could you set the log level to 7 and send it to me when you see that 6LBR stops being responsive ?

stringr9, 2017-01-13:
Sure,  I've dumped it [here](https://gist.github.com/stringr9/31f0dfb2c1a44398f1ac507454cf7d0a).

It failed around 17:33:30.   I've dumped the logs for 10 seconds before and 3 seconds after.  The error at the end is my `service 6lbr stop` after it had frozen up.  I've a a look and there's nothing out of the ordinary when it crashes, but let me know if you see something I missed.

I was wondering if it might be that using ContikiMAC duty cycling at 50kbps causes too much lag on the slip-radio, so I tested it with nullrdc as well but had the same issue.  Ping times to all nodes are 50ms +/- 3ms when there are <25 nodes.  At 30 it starts to drop some pings and has some very long response times, by the time we reach 35 nodes the whole BR will have frozen.

I thought at first it must be a buffer overflow or similar but since it doesn't completely die I'm not sure what the issue might be.

laurentderu, 2017-01-16:
Indeed, the log looks like absolutely normal... In the statistics page of 6LBR do you see many collisions or retransmissions in the CSMA tab or message timeout ?
If you set the MAC layer to None on 6LBR and activate CSAME layer on the slip-radio, does it improves something ?

stringr9, 2017-01-16:
Interesting, lots of collisions/retransmissions.  Not sure what's normal here but I feel like this isn't great (after 5 minutes with 28 nodes):
```
Send packets : 442
Received packets : 467
Not acked packets : 88
Collisions : 368
Retransmissions : 441
Dropped packets : 15
Deferred packets : 0
```
Moving CSMA to the slip radio didn't improve anything, still seeing the BR freeze up with more than 30 nodes unfortunately.

As an experiment, I replaced my 6LBR setup with a CC1310 LP running Contiki's `rpl-border-router` example and the problem went away.  I was able to get up to 40 nodes without any reduction in ping times and could probably have added more, so the problem seems to come only when using 6LBR as the BR.

laurentderu, 2017-01-18:
The number of collision is quite high, but not necessarily the root cause of the issue (the number of dropped packets is still low). What I don't understand yet is why the BR stop being responsive and yet the log show no ack timeout or burst of messages.

If you don't have issues with rpl-border-router on the CC1310, that indicates the problem is related to the slip-radio or the interaction between the slip-radio and the BR.

Is the traffic coming from the node to the external world or is it the other way around ?

stringr9, 2017-01-18:
I'm not running any application traffic on the nodes so the only packets they are transmitting should be RPL maintenance packets (other than when I ping individual nodes to test them).

I've got the slip configured for 460800 baud on the serial bus already, I doubt that's causing a bottle neck but will try it at 921600 just to rule it out.

closed 2017-06-15 by laurentderu