cetic/6lbr #241: 6LBR-test router-config

kameleon070, 2017-01-14:
Hi,

I have strange problem I set everything with instruction now i can just click "Start" in Cooja. In "cooja-1-router console" i see commend Fetching MAC address till i press start and than i see  
2017-01-14 18:08:08.962667: INFO: NODECFG: Using node_config.conf : conf-templates/node-config.conf
2017-01-14 18:08:08.962751: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.962819: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.962881: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963432: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963497: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963560: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963638: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963701: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963763: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963826: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963888: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963959: WARN: NODECFG: Syntax error in node_config.conf : 1"

Node config file:
cooja-2 0:2:0:2:0:2:0:2
cooja-3 0:3:0:3:0:3:0:3
cooja-4 0:4:0:4:0:4:0:4
cooja-5 0:5:0:5:0:5:0:5
cooja-6 0:6:0:6:0:6:0:6
cooja-7	0:7:0:7:0:7:0:7
cooja-8 0:8:0:8:0:8:0:8
cooja-9	0:9:0:9:0:9:0:9
cooja-10	0:a:0:a:0:a:0:a
cooja-11	0:b:0:b:0:b:0:b
cooja-test	0:c:0:c:0:c:0:c

FULL LOG HERE

../package/usr/bin/6lbr conf-templates/cooja-1-router.conf
********************************************************************************
Sat Jan 14 18:08:07 CET 2017 : Starting 6LBR
********************************************************************************
Sat Jan 14 18:08:07 CET 2017 : Starting 6LBR
../bin/cetic_6lbr_router -c conf-templates/test.dat -o ../package/etc/6lbr/nvm.conf -a localhost -R -t tap0  -U ../package/usr/lib/6lbr/6lbr-ifup -D ../package/usr/lib/6lbr/6lbr-ifdown -w ../package/usr/lib/6lbr/www   -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip -n conf-templates/node-config.conf -m ../package/usr/lib/6lbr/plugins   
Contiki-develop-20161222 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2017-01-14 18:08:07.461918: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-develop-20161222)
2017-01-14 18:08:07.462038: INFO: ETH: 6LBR watchdog started (interval: 60)
2017-01-14 18:08:07.462125: INFO: NVM: NVM Magic : 2009
2017-01-14 18:08:07.462168: INFO: NVM: NVM Version : 5
2017-01-14 18:08:07.462709: INFO: CONFIG: Loading configuration : ../package/etc/6lbr/nvm.conf
2017-01-14 18:08:07.462773: WARN: CONFIG: Can not open ../package/etc/6lbr/nvm.conf : No such file or directory
2017-01-14 18:08:07.462817: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2017-01-14 18:08:07.462850: INFO: MAC: Using 'csma' mac driver
2017-01-14 18:08:07.464193: INFO: SLIP: SLIP opened to localhost:60001
2017-01-14 18:08:07.464256: INFO: SCMD: Started br-cmd process
2017-01-14 18:08:07.464303: INFO: BR-RDC: Reset SLIP Radio
2017-01-14 18:08:07.464337: INFO: BR-RDC: Fetching MAC address
2017-01-14 18:08:07.465113: ERROR: SLIP: Unknown command 3f3f (3)
2017-01-14 18:08:07.465200: INFO: BR-RDC: Got MAC: 00:01:00:01:00:01:00:01
2017-01-14 18:08:08.864573: INFO: ETH: TAP Ethernet interface init
2017-01-14 18:08:08.868291: INFO: TAP: opened device /dev/tap0
2017-01-14 18:08:08.868404: INFO: TAP: Running 6lbr-ifup script '../package/usr/lib/6lbr/6lbr-ifup'
6lbr-ifup: Create bridge br0
6lbr-ifup: Attach device tap0 to bridge br0
br0       Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e  
          inet6 addr: fe80::14b9:abff:fee0:6806/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:90 (90.0 B)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

2017-01-14 18:08:08.962192: INFO: ETH: Eth MAC address : 02:01:00:01:00:01
2017-01-14 18:08:08.962290: INFO: LLSEC: Using 802.15.4 framer
2017-01-14 18:08:08.962354: INFO: LLSEC: Using 'nullsec' llsec driver
2017-01-14 18:08:08.962462: INFO: HTTP: Starting webserver on port 80
2017-01-14 18:08:08.962597: INFO: NODECFG: Node Config init
2017-01-14 18:08:08.962667: INFO: NODECFG: Using node_config.conf : conf-templates/node-config.conf
2017-01-14 18:08:08.962751: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.962819: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.962881: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963432: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963497: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963560: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963638: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963701: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963763: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963826: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963888: WARN: NODECFG: Invalid MAC address : 
2017-01-14 18:08:08.963959: WARN: NODECFG: Syntax error in node_config.conf : 1
2017-01-14 18:08:08.964029: INFO: 6LBR: Tentative local IPv6 address fe80::201:1:1:1
2017-01-14 18:08:08.964244: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::201:1:1:1
2017-01-14 18:08:08.964446: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2017-01-14 18:08:08.964556: INFO: 6LBR: RA Daemon enabled
2017-01-14 18:08:08.964619: INFO: 6LBR: Checking addresses duplication
2017-01-14 18:08:09.470584: INFO: SLIP: cooja_cmd: setting channel: 26
cooja_cmd: setting pan-id: abcd
2017-01-14 18:08:20.965862: INFO: 6LBR: Configured as DODAG Root ::
2017-01-14 18:08:20.965893: INFO: 6LBR: Starting IP64
2017-01-14 18:08:20.965911: INFO: 6LBR: IPv4 address : 192.168.52.129
2017-01-14 18:08:20.965953: INFO: 6LBR: Starting as RPL ROUTER
2017-01-14 18:08:20.966275: INFO: CONFIG: Loading configuration : ../package/etc/6lbr/nvm.conf
2017-01-14 18:08:20.966295: WARN: CONFIG: Can not open ../package/etc/6lbr/nvm.conf : No such file or directory
2017-01-14 18:08:20.966305: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2017-01-14 18:08:20.966357: INFO: DNS: DNS proxy started
2017-01-14 18:08:20.966364: INFO: CONFIG: Loading configuration : ../package/etc/6lbr/nvm.conf
2017-01-14 18:08:20.966372: WARN: CONFIG: Can not open ../package/etc/6lbr/nvm.conf : No such file or directory
2017-01-14 18:08:20.966378: INFO: 6LBR: CETIC 6LBR Started
RD client started

What should i change?

kameleon070, 2017-01-16:
No response...

laurentderu, 2017-01-16:
The node-config.conf file of the test environment still has the old syntax, and so 6LBR complains at startup, but it does not cause any other problem. Thanks for reporting I will correct it

MrMikeyMouse, 2017-01-16:
ok i see from another topic:

[0:2:0:2:0:2:0:2]
 name =cooja-2
[0:3:0:3:0:3:0:3] 
name=cooja-3
[0:4:0:4:0:4:0:4]
 name=cooja-4
[0:5:0:5:0:5:0:5]
 name=cooja-5
[0:6:0:6:0:6:0:6] 
name=cooja-6
[0:7:0:7:0:7:0:7] 
name=cooja-7
[0:8:0:8:0:8:0:8] 
name=cooja-8 
[0:9:0:9:0:9:0:9] 
name=cooja-9
[0:a:0:a:0:a:0:a] 
name=cooja-10
[0:b:0:b:0:b:0:b] 
name=cooja-11
[0:c:0:c:0:c:0:c]
 name=cooja-test

It should be something like this. It's from issue #243 and it works :D but i still have problem with nat64

kameleon070, 2017-01-16:
ok i see from another topic:

[0:2:0:2:0:2:0:2]
name =cooja-2
[0:3:0:3:0:3:0:3]
name=cooja-3
[0:4:0:4:0:4:0:4]
name=cooja-4
[0:5:0:5:0:5:0:5]
name=cooja-5
[0:6:0:6:0:6:0:6]
name=cooja-6
[0:7:0:7:0:7:0:7]
name=cooja-7
[0:8:0:8:0:8:0:8]
name=cooja-8
[0:9:0:9:0:9:0:9]
name=cooja-9
[0🅰️0🅰️0🅰️0:a]
name=cooja-10
[0🅱️0🅱️0🅱️0:b]
name=cooja-11
[0:c:0:c:0:c:0:c]
name=cooja-test

It should be something like this. It's from issue #243 and it works :D but i still have problem with nat64

laurentderu, 2017-01-20:
File corrected

closed 2017-01-20 by laurentderu